raspberrypi/firmware #1352: Pi4 wont boot with USB powered hub

tyjtyj, 2020-03-19:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Powered hub backpower pi4 resulting reboot hug on black screen. 
Firmware 00013701 - works fine
Firmware 000137ad - hung on power on if usb 3.0 port is back powered

**To reproduce**
Plug the usb hub to external power. Issue reboot on pi 4

**Expected behaviour**
should reboot normally as per firmware 00013701

**Actual behaviour**
Hung with black screen. Red light on. No green light activity

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi4B
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-05-14
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 541db076832e228b0098065acd6244f1f2d2fc35, stage3

* Which firmware version (`vcgencmd version`)?
Mar 10 2020 15:57:18
Copyright (c) 2012 Broadcom
version 74d7f53b091e465285f38fb96896a8fc328e4c47 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux hassbian 4.19.108-v7l+ #1298 SMP Fri Mar 6 18:09:54 GMT 2020 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
none 

**Additional context**
Issue has been discussed here but no solution provided.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250990&sid=30a4347285481ebef7b11044721f859b&start=25

tyjtyj, 2020-03-19:
I flashed version 000137ad again and the issue went away. 

Step taken flash 013701 and flash 0137ad again...

tyjtyj, 2020-04-02:
Reopen the case as the pi 4 boot after the flashing 0137ad

wolfmanjm, 2021-01-15:
FWIW This is still an issue with 000138a1 which is the current version.
The whole discussion regarding backpowered Hubs is not an issue with the one I am using as I have tested the vusb pin and there is no power coming out of it. I strongly believe this is an issue with the firmware, although I have not downgraded to the (now very old version) to see if that works.
To reiterate RPI4 does not boot when the powered USB is plugged into a USB3 port and has power on.
It neither reboots nor boots from a power down state.

wolfmanjm, 2021-01-15:
I believe this maybe the reason...  https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250990#p1532221

tyjtyj, 2021-01-17:
> FWIW This is still an issue with 000138a1 which is the current version.
> The whole discussion regarding backpowered Hubs is not an issue with the one I am using as I have tested the vusb pin and [...]

Hi, My issue when away after Sept 2020 update. Now i can reboot with USB3 port backpowering. However, rpi does not really shutdown on halt command. It does not really matter to me anymore. I going to close this issue. 

Please open a new case if thi still affecting you with different newer firmware version.

closed 2021-01-17 by tyjtyj