raspberrypi/firmware #1354: Raspberry Pi 4 WiFi (wlan0) interfaces won't show up

ofirc, 2020-03-20:
Hi,
I recently purchased a ["CanaKit Raspberry Pi 4 4GB Starter Kit - 4GB RAM"](https://www.amazon.com/gp/product/B07V5JTMV9/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1) and my WiFi (wlan0 interface) won't show up.

I'm using Raspbian GNU/Linux 10 (buster) with the following kernel:
`Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux`

I started with NOOBS and the Raspbian, then tried to move to a newer OS using the Raspberry Pi Imager, but there is no wlan0 interface showing in ifconfig:
`pi@raspberrypi:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.100.102.10  netmask 255.255.255.0  broadcast 10.100.102.255
        inet6 fe80::9c73:5c79:caa4:1824  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:23:e1:97  txqueuelen 1000  (Ethernet)
        RX packets 234  bytes 29425 (28.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 181  bytes 29795 (29.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@raspberrypi:~ $`

These are the messages from dmesg after plain reboot:
pi@raspberrypi:~ $ dmesg | grep brcm
[    0.272456] brcm-pcie fd500000.pcie: dmabounce: initialised - 32768 kB, threshold 0x00000000c0000000
[    0.272478] brcm-pcie fd500000.pcie: could not get clock
[    0.272541] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.272579] brcm-pcie fd500000.pcie:   MEM 0x600000000..0x603ffffff -> 0xf8000000
[    0.338740] brcm-pcie fd500000.pcie: link up, 5.0 Gbps x1 (!SSC)
[    0.339014] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.473239] brcmstb_thermal fd5d2200.thermal: registered AVS TMON of-sensor driver
[    3.829136] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.841332] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.841829] usbcore: registered new interface driver brcmfmac
[    4.094279] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.112275] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    4.249668] brcmfmac: power management disabled
pi@raspberrypi:~ $`

Any idea?

Thanks,
Ofir

jensstark, 2020-06-09:
Enter

`lsmod`

Do you see loaded modules?
If not:

Get kernel version using 

`uname -a`

then look for a matching directory in /lib/modules (using ls)

`ls /lib/modules`

If there are no modules loaded, you probably have a wrong kernel version.

pelwell, 2020-06-09:
You can combine that into a single command (and save typing) with:
```
ls /lib/modules/`uname -r`
```
Watch out for the backticks - `  - they aren't apostrophes - '.

geargoose, 2022-07-02:
I've got the same issue.