raspberrypi/firmware #1355: More information on SPI and NAND boot modes

LonerDan, 2020-03-21:
Hello,
in the documentation of the GPIO boot mode, there is a mention of SPI and NAND boot modes with a note that there is no Linux support for them at the time. 
[image]
Since those boot modes especially intrigue me for my use case, I wanted to ask for more information. Is the Linux support being worked on? Alternatively, could this boot mode be used to load next stage bootloader (e.g. u-boot) that would then boot the full system (either from the same source or from sd card)?

MDr164, 2021-12-10:
I think it has somewhat been discussed here https://github.com/raspberrypi/firmware/issues/874 as well but there is no further clarification to be found somewhere. The documentation still says those two modes are somewhat supported but not how to actually make use of them. The documentation also says those are being cycled through by the first stage bootloader but that's about it. The linked issue says those modes don't work while docs says otherwise but not how. Still kind of confusing.

pelwell, 2021-12-10:
Documenting a feature that we have no plan to use (and haven't spent time finding the pitfalls of) isn't a good use of resources. Users would be forgiven for thinking that the feature was both tested and supported, when the opposite is true.

closed 2021-12-10 by pelwell

MDr164, 2021-12-10:
Yes that makes total sense, but please then remove the lines in the documentation on raspberrypi.com, stating it works then OR add a disclaimer that it technically works when looking at the SoC side of things but that it's not implemented (and probably won't be). As of now it just unnecessarily confuses users reading that.

JamesH65, 2021-12-10:
Can you link to the page in question?

pelwell, 2021-12-10:
Confuses? Not the most obvious word for that sentence. Teases, perhaps, although that's not the intention. By documenting what the bit does (if not how to use it) the hope is that it saves curiosity from killing the Pi.

Won't fix.

MDr164, 2021-12-10:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot-sequence is the page in question and the line that "confused" me in this case would be "NAND boot and SPI boot modes do work, although they do not yet have full GPU support." which was also mentioned in the linked issue. I think the fact that it explicitly says NAND boot and SPI boot modes do work should also mention what works in that context means, like the SoC support is there but the firmware has no support for that.