raspberrypi/firmware #1356: tvservice HDMI dropout in dual screen mode

Glenn-1990, 2020-03-21:
**Describe the bug**
HDMI0 dropout when changing the display mode of HDMI1 with tvservice -e.

**To reproduce**
1) Connect 2 HDMI displays to your Raspberry Pi 4
2) Bootup
3) Change display mode of HDMI1 (2nd), in my case "tvservice -v 7 -e 'CEA 4'"
```
status before change:
state 0xa [HDMI CUSTOM RGB full unknown AR], 1360x768 @ 60.00Hz, progressive
status after change:
state 0xa [HDMI CEA (4) RGB lim 16:9], 1280x720 @ 60.00Hz, progressive
```
4) The other HDMI output (HDMI0) drops out completely and does not recover, HDMI1 shows a corrupted image.

**Expected behaviour**
HDMI0 should not be affected when changing the displaymode of HDMI1.

**Actual behaviour**
HDMI0 dropout when changing the display mode of HDMI1.

**Logs**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2

Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
Revision        : a03111
Serial          : 100000004de54232
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

config.txt
----------

arm_freq=1500
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=500
core_freq_min=200
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
[... 395 lines cut ...]
[    9.434652] EXT4-fs (mmcblk0p2): resizing filesystem from 385024 to 3845120 blocks
[   12.267745] bcmgenet: Skipping UMAC reset
[   12.368746] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[   12.369105] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   13.448558] bcmgenet fd580000.genet eth0: Link is Down
[   16.348157] Bluetooth: Core ver 2.22
[   16.348254] NET: Registered protocol family 31
[   16.348265] Bluetooth: HCI device and connection manager initialized
[   16.351258] Bluetooth: HCI socket layer initialized
[   16.351280] Bluetooth: L2CAP socket layer initialized
[   16.351349] Bluetooth: SCO socket layer initialized
[   17.161849] Bluetooth: HCI UART driver ver 2.3
[   17.161864] Bluetooth: HCI UART protocol H4 registered
[   17.161963] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.163831] Bluetooth: HCI UART protocol Broadcom registered
[   24.822061] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   24.822071] Bluetooth: BNEP filters: protocol multicast
[   24.822089] Bluetooth: BNEP socket layer initialized
[   25.681866] EXT4-fs (mmcblk0p2): resized to 3145728 blocks
[   28.207632] EXT4-fs (mmcblk0p2): resized filesystem to 3845120
[   50.266819] logitech-hidpp-device 0003:046D:404D.0004: HID++ 4.1 device connected.
[  130.716477] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  130.716500] brcmfmac: power management disabled
[  201.130081] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  202.580006] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead

6by9, 2020-03-21:
If using vc4-fkms-v3d then you should not be using tvservice to change resolution as you're doing it behind the kernel's back. Use KMS or xrandr calls.

Glenn-1990, 2020-03-21:
@6by9 I'm running Buster Lite (no X), to be sure I disabled 'vc4-fkms-v3d' and the problem persist.
Without X and the fkms driver I should use tvservice or am I wrong?

pelwell, 2020-03-21:
1. Is your intention to be able to change the display mode of either display without affecting the other?
2. If you use the config.txt settings to force the display settings in advance, does that work?
3. Does reversing the roles make a difference?

Glenn-1990, 2020-03-21:
> 1. Is your intention to be able to change the display mode of either display without affecting the other?
> 2. If you use the config.txt settings to force the display settings in advance, does that work?
> 3. Does reversing the roles make a difference?

@pelwell 
1. -> Correct
2. -> This does indeed work, but it still fails when I change it afterwards with tvservice.
3. -> Yes, I can change the display mode of HDMI0 without affecting HDMI1.

What I also discovered is that I can recover HDMI0 (after changing HDMI1 mode) by changing it's display mode (or toggling HDMI0 off/on). HDMI1 stays corrupted in this case.