raspberrypi/firmware #1359: raspberry pi 4 wifi channel list for China (CN) on 5GHz band is wrong

ricebucket, 2020-03-29:
I cannot connect to some 5GHz wifi APs because the default 5GHz channel list for China is wrong:

(sections of the printout below omitted for brevity)
```
$ iw list
Wiphy phy0
...
	Band 2:
...
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (disabled)
			* 5190 MHz [38] (disabled)
			* 5200 MHz [40] (disabled)
			* 5210 MHz [42] (disabled)
			* 5220 MHz [44] (disabled)
			* 5230 MHz [46] (disabled)
			* 5240 MHz [48] (disabled)
			* 5260 MHz [52] (disabled)
			* 5280 MHz [56] (disabled)
			* 5300 MHz [60] (disabled)
			* 5320 MHz [64] (disabled)
			* 5500 MHz [100] (disabled)
			* 5520 MHz [104] (disabled)
			* 5540 MHz [108] (disabled)
			* 5560 MHz [112] (disabled)
			* 5580 MHz [116] (disabled)
			* 5600 MHz [120] (disabled)
			* 5620 MHz [124] (disabled)
			* 5640 MHz [128] (disabled)
			* 5660 MHz [132] (disabled)
			* 5680 MHz [136] (disabled)
			* 5700 MHz [140] (disabled)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (20.0 dBm)
			* 5765 MHz [153] (20.0 dBm)
			* 5785 MHz [157] (20.0 dBm)
			* 5805 MHz [161] (20.0 dBm)
			* 5825 MHz [165] (20.0 dBm)
```

Note that only channels 149 and above are enabled. But this does not match the actual regulations:
https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_or_5.8_GHz_(802.11a/h/j/n/ac/ax)

In fact, the reg db actually shows the correct frequencies allowed:
```

$ iw reg get 
global
country CN: DFS-FCC
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 59400 @ 2160), (N/A, 28), (N/A)
	(59400 - 63720 @ 2160), (N/A, 44), (N/A)
	(63720 - 65880 @ 2160), (N/A, 28), (N/A)
```

For example, channel 36 (5180MHz) is disabled even though both the reg db and the wikipedia article says that it should be allowed.

I tried changing the country code with `sudo reg set US`, but somehow this just puts the state into a random country 98:

```
$ sudo  iw reg set US
$ iw reg get 
global
country 98: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 25), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2482 @ 8), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 59400 @ 2160), (N/A, 0), (N/A)
	(59400 - 63720 @ 2160), (N/A, 0), (N/A)
```

And I cannot get this new country code to stick on reboot.
Is the CN country code hard-wired on the hardware sold in China?

Why is the channel list using an incorrect list of channels that is different from the channels specified in the reg db?

danywheeler, 2020-09-25:
any updates? it's been a while, I'm having the same issue.

zaoying, 2020-11-08:
I just bought a Redmi AX6 router. It says "the 160Mhz bandwith only support 36 ~ 64 channel". 
It means I have to choose in between of two options: high bandwith or compatibility with raspberry pi.

zaoying, 2020-11-12:
by changing country to US `sudo iw set reg US` ，raspberry pi 4b has successfully connected with 5G wifi.

zaoying, 2020-12-02:
Maybe the raspberry pi team isn't responsible for this issue, it just reffer to  the repos (hasn't updated for years) list below :

Central Regulatory Domain Agent (CRDA)
https://github.com/mcgrof/crda

wireless-regdb
https://github.com/robertfoss/wireless-regdb

pelwell, 2020-12-02:
There is an updated clm_blob file that might make a difference - try updating the firmware package:
```
$ sudo apt update
$ sudo apt install firmware-brcm80211
```

fantasticmao, 2021-01-20:
I have the same problem

pelwell, 2021-01-20:
Did you run the commands I posted above to update the firmware?

zaoying, 2021-01-22:
I am running ubuntu on the raspberry pi 4b, not the offical raspberry pi os.
```sh
ubuntu@ubuntu:~$ sudo apt install firmware-brcm80211

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package firmware-brcm80211
```

pelwell, 2021-01-22:
The firmware files are available from this repo. It should just be a matter of copying them to `/lib/firmware/brcm`, but I can't give instructions for other distros.

SidneySun, 2021-01-30:
I have run the commands posted above to update the firmware. But I have the same problem.

juergh, 2021-02-08:
For Ubuntu, the clm_blob is part of package linux-firmware-raspi2. Also the git repo for the regulatory DB is https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/.

Please report Ubuntu issues at https://bugs.launchpad.net and not here.

Undefined-User, 2021-02-18:
OK, thank you all for the information provided. I finally solve this problem(but not completely), regulatory database in kernel is correct, but Broadcom Wi-Fi driver using their own regulatory database file ends with `clm_blob` (save in `/lib/firmware/brcm/`). I found a regulatory database update at [here](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=292113), binary file also provided here→([brcmfmac43455-sdio.clm_blob.zip](https://github.com/raspberrypi/firmware/files/6000475/brcmfmac43455-sdio.clm_blob.zip)).

 I reboot RPi4B, switching Wi-Fi country setting in RPi4B to another and switch back to `CN`, then I run command `iw list`, I found some new band is available but some band still missing.

I'm tired in find solution, can anyone give feedback in Raspberry Pi Forums? Or tell Broadcom should let their Wi-Fi driver read standard regulatory database in kernel? Feedback link is [here](https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=291824).

Konano, 2021-03-14:
> OK, thank you all for the information provided. I finally solve this problem(but not completely), regulatory database in kernel is correct, but Broadcom Wi-Fi driver using their own regulatory [...]

The regulatory database update link is wrong, the right one is [here](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=291609)

Konano, 2021-03-14:
> OK, thank you all for the information provided. I finally solve this problem(but not completely), regulatory database in kernel is correct, but Broadcom Wi-Fi driver using their own regulatory [...]

This solution is work, but some bands(52, 56, 60, 64) are still missing.

Undefined-User, 2021-04-02:
> The regulatory database update link is wrong, the right one is [here](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=291609)

Thx, download link is correct now.

Undefined-User, 2021-04-02:
> > OK, thank you all for the information provided. I finally solve this problem(but not completely), regulatory database in kernel is correct, but Broadcom Wi-Fi driver using their own regulatory database file ends with `clm_blob` (save in `/lib/firmware/brcm/`). I found a regulatory database update at [here](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=292113), binary file also provided here→([brcmfmac43455-sdio.clm_blob.zip](https://github.com/raspberrypi/firmware/files/6000475/brcmfmac43455-sdio.clm_blob.zip)).
> > I reboot RPi4B, switching Wi-Fi country setting in RPi4B to another and switch back to `CN`, then I run command `iw list`, I found some new band is available but some band still missing.
> > I'm tired in find solution, can anyone give feedback in Raspberry Pi Forums? Or tell Broadcom should let their Wi-Fi driver read standard regulatory database in kernel? Feedback link is [here](https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=291824).
> 
> This solution is work, but some channels(52, 56, 60, 64) are still missing.

Yes, if no one can fix this error by themselves, the best solution is report bug to Broadcom.

ss8688, 2021-08-15:
Same problem

Chetan2312, 2023-07-15:
Facing same issues on raspberry pi 4b+ model. No issues with raspberry pi 3b+.

pm4rcin, 2023-08-03:
@pelwell sorry to bother you but could you tell how it was resolved? Is it on official linux-firmware tree or do we have to download it from somewhere because I'm on fedora iot right now and I'm lost reading all the threads regarding that issue. I have reg set to PL (Europe) confirmed by `iw list` output but even if I try to change to US or China for testing it still lists 5GHz channels above 48 up to 144 as disabled.

pelwell, 2023-08-03:
What does `dmesg | grep brcmfmac` return?

pm4rcin, 2023-08-03:
> What does `dmesg | grep brcmfmac` return?
```bash
dmesg | grep brcmfmac
[   20.632828] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   20.642945] usbcore: registered new interface driver brcmfmac
[   20.655909] brcmfmac mmc0:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.Raspberry Pi Foundation-Raspberry Pi 4 Model B.bin failed with error -2
[   21.093466] brcmfmac_wcc: brcmf_wcc_attach: executing
[   21.114178] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   21.117221] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
```

pelwell, 2023-08-03:
And `uname -a`?

pm4rcin, 2023-08-03:
> And `uname -a`?

`Linux Pi4 6.4.6-200.fc38.aarch64 #1 SMP PREEMPT_DYNAMIC Mon Jul 24 20:49:23 UTC 2023 aarch64 GNU/Linux`

pelwell, 2023-08-04:
Your OS is running almost the latest released kernel (6.4.6), but with the upstream firmware (`Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2`). We're on `7.45.241 (1a2f2fa CY) CRC: 349757d5 Date: Mon 2021-11-01 00:40:07 PDT Ucode Ver: 1043.2164 FWID 01-5209c09` - you can download it and the associated clm_blob from the links below. You're going to have to figure out where to put them, but it's probably `/lib/firmware/cypress`, with symbolic links from /lib/firmware/brcm`.

This line in your kernel log surprises me:
```
Direct firmware load for brcm/brcmfmac43455-sdio.Raspberry Pi Foundation-Raspberry Pi 4 Model B.bin failed with error -2
```
The missing file is not surprising (the log doesn't show the subsequent successful load of the fallback file), but the file name is very strange. The brcmfmac driver takes the common file stem `brcm/brcmfmac43455-sdio` and adds the name of the current device, which it gets from Device Tree. In our Linux tree that value is `raspberrypi,4-model-b`.

What does `strings /proc/device-tree/compatible` show for you?

Download links:
https://github.com/RPi-Distro/firmware-nonfree/blob/bullseye/debian/config/brcm80211/cypress/cyfmac43455-sdio-standard.bin
https://github.com/RPi-Distro/firmware-nonfree/blob/bullseye/debian/config/brcm80211/cypress/cyfmac43455-sdio.clm_blob

pm4rcin, 2023-08-04:
I don't have `device-tree` entry in `/proc`. Maybe it's because I use UEFI from here which has older firmware? https://github.com/pftf/RPi4

pelwell, 2023-08-04:
OK, try `strings /sys/firmware/devicetree/base/compatible` instead.

pelwell, 2023-08-04:
You're just going to have to guess which files to overwrite, but I suggest brcm/brcmfmac43455-sdio.bin and  brcm/brcmfmac43455-sdio.clm_blob (or whatever files they point to if they are symbolic links).

pm4rcin, 2023-08-04:
@pelwell can I load the new firmware from live system or do I have to reboot?

pelwell, 2023-08-04:
You _might_ be able to do something clever with unbinding the SDIO device to force a reload, but rebooting is much, much easier.