raspberrypi/firmware #1363: Question re GPIO 14 (TXD) at startup

bit-bash, 2020-04-05:
I've noticed that GPIO 14 (UART TxD) gets driven high for a bit at startup on a PI 3B+.

I've connected an LED to J8 pin-8, and I notice it lights up at about 2.5 seconds after power on - the LED goes out about 3.5 seconds later.  This "drive" is stronger than just a pullup.

I have both:
   dtoverlay=disable-bt
   enable_uart=0
in my config.txt file.  Also BOOT_UART=0 is set in the bootcode.bin binary file.

If I power up without an SDCARD present, GPIO 14 does not get driven high.

So, I'm wondering, is bootcode.bin enabling the UART by default at startup, then waiting for start.elf to process config.txt to honor the "enable_uart=0" line?  If this is true, is there a way to get bootcode.bin to not ever enable the UART on GPIO pins 14 and 15?

Thanks!

pelwell, 2020-04-06:
Reading the code, UART initialisation (including configuration of the GPIOs) happens for any of the following reasons:
* The `BOOT_UART=0` string has been edited.
* A file called `UART` exists in the root of the SD card.
* config.txt includes the setting `uart_2ndstage=1`.

If none of those conditions are true then the GPIOs should be unchanged by the time start.elf begins. If you delete/rename start.elf, or add `start_file=dummy` to config.txt, do you still see the pin driven high?

6by9, 2020-04-06:
dt-blob.dts config for GPIO 14 is for uart TX though, so it will be mapped to the UART whether the UART is initialised or not.
https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts#L1333

pelwell, 2020-04-06:
Not from bootcode.bin though, hence the question.

bit-bash, 2020-04-06:
Confirmed the following:
   BOOT_UART=0 has not been edited or changed
   UART file does not exist
   config.txt does not contain uart_2ndstage=1

But, my assumption that bootcode.bin was driving GPIO 14 is incorrect.  I started the platform with start_file=dummy and GPIO 14 was NOT driven (of course the system didn't fully boot).

I then removed the start_file=dummy line but also removed all the kernel*.img files so it couldn't boot the kernel.  This time GPIO 14 WAS driven, but remained driven while the system did what ever it does when it can't find a kernel to boot.  Not sure if it's start.elf or the kernel that actually stops the GPIO 14 pin from being driven (which happens about 6 seconds after power up) - maybe with the missing kernel image file, start.elf didn't finish processing config.txt so it didn't handle my enable_uart=0 line.

So it seems that something in start.elf is driving the GPIO 14 line - at least for a few seconds.  I'm not loading any overlays other than these two:
   dtoverlay=disable-bt
   dtoverlay=disable-wifi

Thanks!

pelwell, 2020-04-06:
Thanks - that points the finger at the dt-blob GPIO configuration that @6by9 was referring to. You can confirm this by building and installing a modified version of the file, as described here: https://www.raspberrypi.org/documentation/configuration/pin-configuration.md

It is a bit anachronistic that the UARTs are still configured from the dt-blob, and changing that is theoretically just a matter of patching that file and releasing a new firmware. However, I suspect that a number of bare metal projects have come to rely on the UART pins being configured from boot, so I'm reluctant to change it now without some kind of simple configuration change to re-enable the mapping.

pelwell, 2020-04-06:
Of course, you have the option to use a modified version of dt-blob.bin - the external file overrides the one built into start.elf. It does mean that you will miss out on changes to the upstream version unless you monitor the firmware repo for changes, but they will be rare and usually confined to the addition of support for new hardware.

bit-bash, 2020-04-06:
I think this is what I need!  I am assuming that the "built-in" dt-blob.bin in start.elf would be the same that is currently in firmware/extra (after being compiled of course).

So if get the dt-blob.dts, modify it for GPIO 14, compile it, then place it in /boot/dt-blob.bin this means that start.elf will use it and not its built-in flavor?

If this works out, I wouldn't require any changes to the released firmware - I can do this "hack" locally.

Thanks!

pelwell, 2020-04-06:
Yes - that's correct. Be careful to find the right pins section - `pins_3bplus` - remove `pin@p14` and `pin@p15`, compile it with dtc and copy to /boot/dt-blob.bin, and you _should_ find that pin stays low.

bit-bash, 2020-04-06:
Yep, I found the pins_3bplus section - seems an easy-peasy fix.  :)

Just for reference, I was interested to determine if I could have full access to all 28 GPIO pins on the RPI 3B+ without any other interference.  I bread-boarded 28 LEDs to each GPIO pin and validated that I could control each one separately (cycling them looks cool in the dark).

This is how I noticed the GPIO 14 pin being driven at boot time.  I also noticed the hardwired GPIO 2/3 pullups for I2C but those aren't an issue as they are not "driven" high.

Thanks!

bit-bash, 2020-04-06:
Just as a sanity check, I extracted the built-in dtb from start.elf, converted it to dts, and then compared it to firmware/extra/dt-blob.dts.  The only difference was that the **pins_4b** section is not included in start.elf - but this makes sense as that section would only be needed for start4*.elf files.  If I'm not going to use a PI4 then I guess I could remove it from my dt-blob.bin to reduce its size.

I did notice however, that if I extract the built-in dtb from start4.elf and compare it to firmware/extra/dt-blob.dts, the section **pins_fpga** is not included in the dt-blobs.dts file.  If I was to make a dt-blob.bin file for the RPI4, would the pins_fpga section need to be added for it to work?

Thanks!

pelwell, 2020-04-06:
The published source in the firmware repo covers all devices and is the only one you will need. There is occasionally a short lag between releasing a new device and updating the repo, but there shouldn't be. With the move to building all firmware from a common branch I wouldn't be surprised if there is some tidying up of the built-in dt-blob.

bit-bash, 2020-04-06:
Sounds good!  Got my dt-blob.bin working and GPIO 14 is no longer driven high.
Happy camper!!!

closed 2020-04-06 by bit-bash