raspberrypi/firmware #1364: Kernel 5.4 Module7.symvers, Module7l.symvers faulty

MrEngman, 2020-04-06:
With the appearance of the 5.4 kernel I though it a good idea to check the various wifi drivers I compile compiled alright.

Some failed and I will need to update them for kernel 5.4 but I found other drivers compiled OK for kernel versions 5.4.29+ and 5.4.29-v8+ but **the compiles failed for kernel versions 5.4.29-v7+ and 5.4.29-v7l+** with the error
```   Building modules, stage 2.
  MODPOST 1 modules
ERROR: "__stack_chk_guard" [/home/pi/wifi-src20/rtl8188eus/8188eu.ko] undefined!
scripts/Makefile.modpost:93: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1607: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/home/pi/src/linux'
Makefile:2047: recipe for target 'modules' failed
make: *** [modules] Error 2
```
This is for the 8188eu driver but others compiled OK or failed the same way. 

I compile the drivers on a Pi 3B+ and I prepare the kernel to compile the drivers using commands 
```
make mrproper
make bcmXXXX_defconfig
make modules_prepare
```
using the relevant bcmXXXX_defconfig file for the particular kernel version, +, -v7+ or -v7l+, or 
```
make mrproper
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- modules_prepare
```
for the -v8+ kernel and then copy the relevant Module.symvers file to the kernel source directory and compile the drivers using
```
make clean
make ARCH=arm -j4
```
for kernels +, -v7+ and -v7l+ and
```
make clean
make  ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -j4
```
for kernel -v8+.

After a load of hunting around in the various sources for the kernel and drivers I found if instead of preparing the kernel using the commands above and copying the Module.symvers file from the github repo to the kernel directory but I fully compiled the kernel version using commands
```
make mrproper
make bcmXXXX_defconfig
make ARCH=arm -j4
```
where the Module.symvers file is created by the compiler the driver compiles that previously failed now compiled OK. Checking Module.symvers in the linux source directory now showed it contained the value **__stack_chk_guard** which was not found in the Module7.symvers and Module7l.symvers files that were used to prepare the kernel using the original set of commands to prepare the kernel for compiling the drivers.

So it appears for some reason there is an issue with the Module7.symvers and Module7l.symvers files in the hexxeh/rpi-update and raspberry/firmware github repos.

Originally raised the issue in the forum at [https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&start=25#p1637260](url) and detailed the fix at [https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&start=50#p1638441](url)

MrEngman

zachberger, 2020-04-08:
I am unable to install the `wireguard-dkms` package with the same error

pprindeville, 2020-11-18:
Um... and if you kbuild with `USER_EXTRA_CFLAGS=-fno-stack-protector` that doesn't fix it?