raspberrypi/firmware #1365: RPI 3B not requestiong PXE options

whinis, 2020-04-07:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
I have been trying to setup network booting on a RPI 3B and following a few guides todo so however I could not get it to boot without the SD card. I then starting doing dhcp dump and noticed that it was not even requesting the options from the dhcp server. Dump below

**System**
https://pastebin.com/EJ6mZkXW

**Logs**

https://pastebin.com/QNL6gEQz

pelwell, 2020-04-07:
The 3B+ was the first Pi to support network booting without an SD card - the 3B can't.

pelwell, 2020-04-07:
Correction - it can, but it's limited in what it supports.

whinis, 2020-04-07:
Right, I know its limited but I cannot even get it to requests the options which from my understanding it should. Even the 2 should support it with an SD card installed.

whinis, 2020-04-07:
Also, following the debugging guide at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md After setting program_usb_boot_mode and confirmed that the ethernet lights do turn on without the SD card inserted.

timg236, 2020-04-07:
Your best option would be to use a known working TFTP server configuration such as Pi Server and adjust your configuration to match. 
https://github.com/raspberrypi/piserver

pelwell, 2020-04-07:
I'd like to see a log with the SD card present and one without it. And a copy of the server configuration.

mullcom, 2020-04-07:
Hello. Sorry ror interrupting. Have any of you any links to information how
to set this up. I can start to testing. And delirer info back. Have both
pi4 and pi3B+ Only lies  on the table and do nothing.

whinis, 2020-04-07:
I am going to attempt to get piserver running on a debian VM as I don't have another pi currently. I am getting wireshark captures currently and from what I can see without the SD card it isn't even making a DHCP requests.

pelwell, 2020-04-07:
The two links above are good places to start: 
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md
and
https://github.com/raspberrypi/piserver

whinis, 2020-04-07:
Looking at the capture log while I wait for a new debian iso to download, its not making a DHCP requests, is assuming its IP, and then automatically makes a requests to the router for tftp as it didn't request the address from my router

pelwell, 2020-04-07:
The capture log above is strange in that it is requesting a specific IP address: 192.168.1.34. I don't know where it could have got that from.

whinis, 2020-04-07:
Thats an IP it had whenever the SD card was in but its not in it any longer and its been removed from the routers lease table as well. Yet it still continues to use that ip.

pelwell, 2020-04-07:
This is going to sound odd, but I don't think that packet you captured came from the Pi bootrom/bootcode - the details don't match the code I'm looking at. Are you sure there isn't another Pi running Linux using DHCP, for example?

pelwell, 2020-04-07:
And this line would seem to support that theory:
```
OPTION:  60 ( 46) Vendor class identifier   dhcpcd-8.1.2:Linux-4.19.114-v7+:armv7l:BCM2835
```
A Pi network booting would mention "PXEClient".

whinis, 2020-04-07:
So no, that capture up there was using dhcpdump in raspbian lite.  I have a capture using wireshark middle manning just the bootroom and dhcp is never called. I can post the wireshark logs for that but its a bit more messy.

pelwell, 2020-04-07:
Still this:
> I'd like to see a log with the SD card present and one without it. And a copy of the server configuration.

whinis, 2020-04-07:
How would you like to get the logs? I have the .pcapng files as well as .txt dumps from wireshark. the txt dump for the sd card one which is from power-up to shutdown is 200kb and probably a little large for pastebin. I took 2 captures of no SD card and managed to get a DHCP requests in one of them but not the other. Its chosen a new IP again.

whinis, 2020-04-07:
Also the server configuration is more difficult as I have the DHCP setup on my mikrotik and it doesn't give me a raw config file. TFTP is done through freenas

pelwell, 2020-04-07:
Just the first dozen or so packets of each trace would be fine - until it asks for a file. .txt dump is preferable.

pelwell, 2020-04-07:
Let's start with the logs then.

whinis, 2020-04-07:
https://pastebin.com/JuAniwQF <- first 21 packets of no SD card where DHCP requests was seen
https://pastebin.com/Emu6wiTq <- First 18 packets of No SD card where No DHCP requests was seen
Both of these were cutoff where I started to see repeated spanning tree requests from the router

https://pastebin.com/X7h86nWE <- first 25 packets of SD Card, stopped where first IPV6 packet was seen.

pelwell, 2020-04-07:
From what I can see, only the first trace contains any traffic to or from the Pi (MAC address b8:27:eb:...), and even then it's only the initial DHCP DISCOVER. If you filter by MAC address you can ignore those spanning tree packets and just include the relevant data.

whinis, 2020-04-07:
I will work on filtering the packets more. After plugging it 15 times I managed to get it working once but it actually completed the DHCP requests. Every other time it seems to boot, choose a random IP address and then continue and ask the gateway for files via tftp.

whinis, 2020-04-07:
https://pastebin.com/Np9wgE4f Here is a NoSD capture no STP, MNDP or CDP and only packets to or from the pi. with a reboot around packet 102. In both cases it grabs 192.168.1.254 however while that address is not in use currently it is registered in the DNS table.

whinis, 2020-04-07:
I finally got Rasbian installed in a VM and PiServer running and its not detecting my Pi. It warned me about STP running so I disabled that but it has had no effect.

pelwell, 2020-04-08:
None of your logs show any packets going back to the Pi, although they do suggest that it has received some - it gains an IP address and the address of a server, for example. I've seen nothing to suggest that the Pi isn't behaving as it should. This isn't a support area - you should go to our Forums for that, perhaps Networking and Servers: https://www.raspberrypi.org/forums/viewforum.php?f=36

closed 2020-04-08 by pelwell

whinis, 2020-04-08:
I posted on the forum but its not gaining an IP address as much as its assigning itself one. Many times that ip address is already assigned. It seems to be a bootrom issue as its not sending packets either.