raspberrypi/firmware #1367: Pi Zero Bootcode.bin wrong USB Device ID

Badger101, 2020-04-13:
**Describe the bug**
When booting a Pi Zero (Original not W) the YS-LAN26 RT8152B:
https://shop.pimoroni.com/products/three-port-usb-hub-with-ethernet-and-microb-connector
Is seen as a:
ID 0bda:8151 Realtek Semiconductor Corp. RTL8151 Adapteon Business Mobile Networks BV
Instead of:
ID 0bda:8152 Realtek Semiconductor Corp. RTL8152 Fast Ethernet Adapter

More info [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=270232&p=1641454)

**To reproduce**
Clean install of 2020-02-13-raspbian-buster, bootcode.bin on the SD Card with 'timeout' file.

**Expected behaviour**
The Ethernet adapter in the USB hub should be correctly seen as a RTL8152 Fast Ethernet Adapter, and eth0 should come up.

**Actual behaviour**
The Ethernet adapter in the USB hub is seen as a RTL8151 Adapteon Business Mobile Networks BV, and no eth0.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* PiZero
* Which OS and version
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4
* Which firmware version
Feb 12 2020 12:38:08 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)
* Which kernel version
Linux raspberrypi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
**Logs**
Please see [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=270232&p=1641454#p1640285)

**Additional context**
Correctly seen as a RTL8152 Fast Ethernet Adapter when booting from SD Card

Badger101, 2020-04-16:
Hi,

The issue has been confirmed by another forum member, please see [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=270232#p1642652).

Thanks

malvism, 2020-12-07:
> 
> 
> Hi,
> 
> The issue has been confirmed by another forum member, please see [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=270232#p1642652).
> 
> Thanks

I have the same issue as described on forum. Something is wrong using bootloader.bin as it gives the ID with -1 of device

tmstk, 2021-03-04:
Can be confirmed on a Pi 4B with a RTL8153 Gigabit Ethernet Adapter. When booted from USB disk the USB product ID is set to 8151 instead of 8153. The device gets recognized correctly by udev when plugged in later on.

Firmware: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
OS: 2021-01-11-raspios-buster-armhf-lite
Kernel: Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

tmstk, 2021-03-04:
`usb_modeswitch -v 0BDA -p 8151 --reset-usb` e.g. in rc.local fixes this issue.

ipchama, 2021-08-24:
Just to add to this in case it helps someone else because this drove me nuts.  I have 3x RJ45-to-USB dongles (TP-Link UE300) connected to a USB 3.0 hub.

To get them all working again, I first had to grab the details for them from `lsusb`:
```
...
Bus 002 Device 009: ID 2357:0600 TP-Link 
Bus 002 Device 008: ID 2357:0600 TP-Link 
Bus 002 Device 007: ID 2357:0600 TP-Link 
...
```
Then just reset each one.  You can target them with bus+device number, but I just put 3x `usb_modeswitch -v 2357 -p 0600 --reset-usb` in rc.local to fix on every boot:

```
...
Bus 002 Device 009: ID 2357:0601 TP-Link 
Bus 002 Device 008: ID 2357:0601 TP-Link 
Bus 002 Device 007: ID 2357:0601 TP-Link 
...
```
(notice the 0600 -> 0601)

Then just a loop bring up eth1-3.

Averell7, 2021-09-10:
I have the same problem, exactly as explained by tmstk on a Raspberry Pi 4 which worked previously fine with a USB-ethernet adapter. Recently the problem appeared, but I don't know the exact cause : an update ? 
The RTL8153 Gigabit Ethernet Adapter. when booted normally from the SD card, appears as 8151 instead of 8153. The device gets recognized correctly by udev when plugged in later on, but syslog shows that there is first an error and then the problem is corrected. 

`Sep 10 15:58:36 raspberrypi kernel: [  105.910371] usb 1-1.4: USB disconnect, device number 4
Sep 10 15:58:38 raspberrypi kernel: [  107.798260] usb 1-1.4: new high-speed USB device number 5 using xhci_hcd
Sep 10 15:58:38 raspberrypi kernel: [  107.929890] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
Sep 10 15:58:38 raspberrypi kernel: [  107.929912] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=6
Sep 10 15:58:38 raspberrypi kernel: [  107.929930] usb 1-1.4: Product: USB 10/100/1000 LAN
Sep 10 15:58:38 raspberrypi kernel: [  107.929948] usb 1-1.4: Manufacturer: Realtek
Sep 10 15:58:38 raspberrypi kernel: [  107.929966] usb 1-1.4: SerialNumber: 000001
Sep 10 15:58:38 raspberrypi kernel: [  108.028824] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
Sep 10 15:58:39 raspberrypi kernel: [  108.168361] r8152 1-1.4:1.0: Direct firmware load for rtl_nic/rtl8153a-4.fw failed with error -2
Sep 10 15:58:39 raspberrypi kernel: [  108.168388] r8152 1-1.4:1.0: unable to load firmware patch rtl_nic/rtl8153a-4.fw (-2)
Sep 10 15:58:39 raspberrypi kernel: [  108.209903] r8152 1-1.4:1.0 eth1: v1.12.11
Sep 10 15:58:39 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4"
Sep 10 15:58:39 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
Sep 10 15:58:39 raspberrypi kernel: [  108.264081] usbcore: registered new interface driver cdc_ether
Sep 10 15:58:39 raspberrypi kernel: [  108.266107] usbcore: registered new interface driver r8153_ecm
Sep 10 15:58:39 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4"
Sep 10 15:58:39 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
Sep 10 15:58:39 raspberrypi systemd-udevd[1731]: Using default interface naming scheme 'v240'.
Sep 10 15:58:39 raspberrypi systemd[1]: Found device RTL8153 Gigabit Ethernet Adapter.
`