raspberrypi/firmware #1368: Pi3B+ freeze with connections via eth0, no issue with wlan0

karsc96, 2020-04-13:
_This is my first time posting an issue here, please accept my apologies in advance in case something is wrong/missing etc._

Streaming audio or sending/receiving other larger amounts of data through the eth0 connection always causes the RPI to freeze completely after some seconds transferring. It only happens using wired eth0 connection. Using wireless (wlan0) connection works like a charm.

I experienced the freeze performing the following actions:
- downloading updated packages (sudo apt-get update, sudo apt-get upgrade)
- playing music via shairport-sync (raspi acting as an airplay-server) from an iPhone (receiving)
- playing music using mpd (music player daemon). MPD runs on the raspi but the music source files are located on my Synology NAS (mounted CIFS Samba Ver.3)

The behaviour while playing music is always the same: after some seconds of playing normal the raspi suddenly freezes completely (no reaction, no ssh, but getting noticeably hotter) playing the current 1-2 seconds of music in a constant loop.
Note that I am not streaming through both methods simultaneously. Since I am using one audio output (hifiberry digi Pro) assembled to the rpi.
Normal browsing, SSH is not of a problem at all.

Things I tried so far:
- Reduced ethernet speed to 100 full duplex
- switching off eee
- rpi-update (because my description pretty much was the same as here [Issue-2598](https://github.com/raspberrypi/linux/issues/2598) and here [Issue-2608](https://github.com/raspberrypi/linux/issues/2608)
- umounting the CIFS-share (streaming only via shairport-sync)
- unplugging all remaining USB-devices (mouse, touch-screen connection ILITEK) 
- setting static IPs for eth0 (with dhcp-service **enabled**)
- setting DHCP IP for eth0
- setting static IPs for eth0 (with dhcp-service **disabled**)

**System**
Raspberry Pi3B+ with mounted HifiBerry Digi Pro (no further HW-modifications)
using a decent power source (AUKRU) with measured 5.3V on the pins in normal conditions
5f884374b6ac6e155330c58caa1fb7249b8badf1
Firmware version 1329c57b34e3de9dc867b598a9760848be3b4e13
Linux raspberrypi 4.19.114-v7+ #1303 SMP Tue Apr 7 15:44:16 BST 2020 armv7l GNU/Linux

dmesg-output:
[dmesg.txt](https://github.com/raspberrypi/firmware/files/4472017/dmesg.txt)

Although wireless connection works, my target scenario is to use wired ethernet connection.

Any help is highly appreciated. My abilities might be limited since I am pretty new to this.
Thanks in advance.
karsc96

pelwell, 2020-04-14:
Something is upsetting the built-in hub that connects to the LAN7800 Ethernet controller. Have you tried changing switch ports and cables? You didn't say that you have removed the HiFiBerry sound card for testing - it's worth doing that as well, if only to rule it out as a source of interference.

karsc96, 2020-04-14:
Thank you for getting back quickly.
I did try with different cables and ports (tried different switches).
Indeed I did not test yet with the hifiberry card removed. I have done this now:
Find the dimes attached. I observe the same behaviour (music stopping) and the same error messages.

> usb usb1-port1: disabled by hub (EMI?), re-enabling...

[dmesg_wo-hifiberry.txt](https://github.com/raspberrypi/firmware/files/4477295/dmesg_wo-hifiberry.txt)

I also observe some other error messages which I don't know about and I am not sure if/how they relate:
> [    5.732128] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.733552] usbcore: registered new interface driver brcmfmac
[    6.027572] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.045369] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2

pelwell, 2020-04-14:
Those brcmfmac messages aren't errors - that's what a happy WiFi interface looks like.

I'm starting to think this might be a hardware fault. How long have you had this 3B+? And when did the problem start occurring?

karsc96, 2020-04-14:
I bought it recently (few weeks ago).

Just tried to use a smaller SD-card (8G instead of 32G) with a fresh raspbian and already during set-up when it is looking for updates (the step right after setting the wifi connection) it already freezes while downloading current package information.
This makes me also think as being HW-issue :-/

Are there any measures I can take to rule other things out? e.g. completely removing USB-drivers and re-installing them again?
Look at some different logs?

pelwell, 2020-04-16:
Nobody else is reporting this issue so, although it's not what you want to hear, unless you can borrow another 3B+ to try I think you have no choice but to return your current one (unless you can talk to your reseller and arrange some kind of swap).

karsc96, 2020-04-16:
I see. Thanks for getting back anyway.