raspberrypi/firmware #1369: poe-hat fan problem with last firmware

4suino, 2020-04-16:
With the latest firmware, the poe-hat fan on my Raspberry isn't able to be switched off in case of low temperature but run always at a very low speed.

In the boot.txt I set a very high threshold temperature but this seems to be ignored.

Raspberry Pi3 with poe-hat v2
kernel version 4.19.114-v7+

Is there someone with the same issue?

Thank you

pelwell, 2020-04-17:
There was a recent change to add more speed controls to the rpi-poe overlay. The current help information says:
```
pi@raspberrypi:~$ dtoverlay -h rpi-poe
Name:   rpi-poe

Info:   Raspberry Pi PoE HAT fan

Usage:  dtoverlay=rpi-poe,<param>[=<val>]

Params: poe_fan_temp0           Temperature (in millicelcius) at which the fan
                                turns on (default 40000)
        poe_fan_temp0_hyst      Temperature delta (in millicelcius) at which
                                the fan turns off (default 2000)
        poe_fan_temp1           Temperature (in millicelcius) at which the fan
                                speeds up (default 45000)
        poe_fan_temp1_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 2000)
        poe_fan_temp2           Temperature (in millicelcius) at which the fan
                                speeds up (default 50000)
        poe_fan_temp2_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 2000)
        poe_fan_temp3           Temperature (in millicelcius) at which the fan
                                speeds up (default 55000)
        poe_fan_temp3_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 5000)
```
It's possible that you will have to raise all of the temperature thresholds to have the effect you are after.

4suino, 2020-04-17:
The issue was also with the previous version. This update on the overlay was in the last firmware.
However I have seen it and try to update my config.txt file without any change to the fun...

Now I have this setting:

dtoverlay=rpi-poe,poe_fan_temp0=65000,poe_fan_temp0_hyst=5000,poe_fan_temp1=67000,poe_fan_temp1_hyst=2000

and I haven't set the second and third variables for the temperature.

With this configuration the fan should hold until 65 degree of temperature is reached, but at 45 degree the fun spins... very slow but spins.

pelwell, 2020-04-17:
That's a very long line - I'm almost certain that will get truncated to the 80ish character line length limit used by config.txt. Check for Device Tree errors and warnings from the firmware with:
```
$ sudo vcdbg log msg |& grep dt
```

You can split long dtoverlay lines like this:
```
dtoverlay=rpi-poe,poe_fan_temp0=65000,poe_fan_temp0_hyst=5000
dtparam=poe_fan_temp1=67000,poe_fan_temp1_hyst=2000
etc.
```

4suino, 2020-04-17:
> That's a very long line - I'm almost certain that will get truncated to the 80ish character line length limit used by config.txt. Check for Device Tree errors and warnings from the firmware with: [...]

You are right.
I corrected it in:
dtparam=poe_fan_temp0=65000,poe_fan_temp0_hyst=5000
dtparam=poe_fan_temp1=70000,poe_fan_temp1_hyst=2000

but consider that in origin was already in this format and I correct because the fan spin-up very fast so I try a different configuration hoping to correct the problem...

4suino, 2020-04-19:
I think I found the problem. I set also **poe_fan_temp2, poe_fan_temp2_hyst, poe_fan_temp3 and poe_fan_temp3_hyst** and then the fan start works correctly.
I think this should be an issue because the additional settings aren't mandatory.

pelwell, 2020-04-20:
That's what I meant when I said "It's possible that you will have to raise all of the temperature thresholds to have the effect you are after."

I'm not sure what you mean when you say they aren't mandatory? The fact that they have default values means they have an effect even when you don't specify them.

4suino, 2020-04-20:
> That's what I meant when I said "It's possible that you will have to raise all of the temperature thresholds to have the effect you are after."
> 
Ok, now is clear.

> I'm not sure what you mean when you say they aren't mandatory? The fact that they have default values means they have an effect even when you don't specify them.

Yes, you are right. I thought that the parameters for second and third threshold was optional and haven't a default value. My mistake.

I close the issue.

Thank you

closed 2020-04-20 by 4suino