raspberrypi/firmware #1370: RPI3+ not acknowledging boot-file option sent by server, always asks for bootcode.bin

inakypg, 2020-04-19:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
RPI3+ is not acknowledging boot-file option sent by the server in the DHCP offer. 

Always asks for bootcode.bin -- the server shall be able to tell the RPI which boot file they have to use, and the RPI could default to another one.

**To reproduce**
Setup a TFTP server, ask the server to offer boot-file to target named bootfile2.bin, RPI will keep asking for bootfile.bin.

**Expected behaviour**
RPI asks for file offered by server in DHCP offer first, maybe fall back to bootfile.bin if not found.

**Actual behaviour**
RPI is always asking for bootfile.bin, ignoring the name in DHCP offer.

**System**
System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3\0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4

Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
Revision	: a020d3
Serial		: 0000000076fb1f9e
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Feb 12 2020 12:38:08 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30471072 3059176  26125088  11% /
devtmpfs          469540       0    469540   0% /dev
tmpfs             474148       0    474148   0% /dev/shm
tmpfs             474148    6356    467792   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             474148       0    474148   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   53463    204632  21% /boot
tmpfs              94828       0     94828   0% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20200218
raspberrypi-sys-mods:
[... 430 lines cut ...]
[    7.190338] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    7.207968] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    7.209942] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    7.228951] bcm2835_audio soc:audio: card created with 8 channels
[    7.251404] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    7.251442] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    7.258829] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    7.258868] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    7.269959] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    7.270001] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    7.384109] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    7.571665] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.677485] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.687085] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.687870] usbcore: registered new interface driver brcmfmac
[    7.920102] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.925406] random: crng init done
[    7.925422] random: 7 urandom warning(s) missed due to ratelimiting
[    7.934560] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[   10.063182] 8021q: 802.1Q VLAN Support v1.8
[   10.299134] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   10.899086] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   10.899101] 8021q: adding VLAN 0 to HW filter on device eth0
[   10.914901] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.666877] fuse init (API version 7.27)

pelwell, 2020-04-20:
Do you realise that bootcode.bin is requested by the ROM? That means that if you hoping to be able to send a different .bin file then that would require changes to the ROM, which isn't possible for obvious reasons.

inakypg, 2020-04-20:
definitely :) that would make it for a change for newer generations of the ROM in future products

timg236, 2020-04-20:
The ROM in future generations (2711) has had all network boot related code removed. It won't be coming back :)
Fortunately, 2711 doesn't need to load bootcode.bin because that's stored on the local EEPROM so boot-file is not relevant there either.