raspberrypi/firmware #1373: Strange behaviour if sound is played on a Pi4 with composite video selected

KenT2, 2020-04-19:
**Describe the bug**
On Pi4 turning on the A/V output does not disable HDMI sound leading to slow video/audio playing with omxplayer/mplayer and high pitched whine when using mplayer

 Following reboot, even with HDMI monitor disconnected, the first attempt to play audio through hdmi either with mplayer or omxplayer causes continuing problems with video speed and high pitched whine on local output from mplayer

**To reproduce**

With a HDMI monitor connected select composite output in RP configuration
Reboot with hdmi monitor disconnected, not just unpowered.
Noted that the audio output on the taskbar was set to hdmi
play a .mp3 track with mplayer
The track plays with an added high pitched whine

Change to local audio output  using the task bar -  the audio plays slow without whine

Play a video with omxplayer
The video runs slowly.
Change to local sound  - omxplayer -o local
The video plays slowly but faster

**Expected behaviour**
If A/V output selected:
 - Audio output at boot set to loca,l not to the last used output
 - HDMI audio selection disabled on taskbar
 - omxplayer cannot use -o hdmi

**Actual behaviour**
See above

**System**
Pi4 with Buster
 Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage5

popcornmix, 2020-09-03:
Can you try with latest rpi-update firmware? There's a potential fix.

KenT2, 2020-09-27:
Works fine
Thanks

closed 2020-09-27 by KenT2