raspberrypi/firmware #1377: RPi4 2Gb MMU crash V3d

tvjon, 2020-04-22:
As per attached image.

Board has been running relatively ok for a few weeks. Attached a 4K tv earlier & several minutes later, crash.

$ uname -a
Linux b1 5.4.32-v7l+ #1308 SMP Fri Apr 17 20:10:43 BST 2020 armv7l GNU/Linux

$ vcgencmd version
Apr 15 2020 11:56:24 
Copyright (c) 2012 Broadcom
version 1103ec16df2571274600b377411499bf5aed6d0d (clean) (release) (start_x)

[image]

timg236, 2020-04-22:
V3D is a raspberrypi/linux  driver, not sure if anyone can move the bug to the correct repo

popcornmix, 2020-04-24:
I don't believe moving issues between repos is possible. Please create issue here:
https://github.com/raspberrypi/linux/issues