raspberrypi/firmware #1378: RPi3B+: Kernel Logs/Errors

pheiduck, 2020-04-25:
Kernel 4.19.97-v7+
journalctl -p err -b
[Kernellogs.txt](https://github.com/raspberrypi/firmware/files/4533517/Kernellogs.txt)

6by9, 2020-04-26:
These are all kernel messages, therefore any query should have been against the kernel repo - https://github.com/raspberrypi/linux/issues

```
Apr 23 16:09:47 raspberrypi kernel: vc_vchi_sm_init: failed to open VCHI service (-1)
Apr 23 16:09:47 raspberrypi kernel: [vc_sm_connected_init]: failed to initialize shared memory service
Apr 23 16:09:48 raspberrypi kernel: vc_sm_cma_vchi_init: failed to open VCHI service (-1)
Apr 23 16:09:48 raspberrypi kernel: [vc_sm_connected_init]: failed to initialize shared memory service
Apr 23 16:09:48 raspberrypi kernel: bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
Apr 23 16:09:48 raspberrypi kernel: bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
```
You've most likely set gpu_mem=16, and therefore those services won't be running as they require some GPU memory.
```
Apr 23 16:09:49 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Apr 23 16:09:49 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Apr 23 16:09:49 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
```
You have wifi onboard and it's loaded the appropriate firmware
```
Apr 23 16:09:53 raspberrypi kernel: lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
```
See https://github.com/raspberrypi/linux/issues/2928, but nothing serious.

Did you actually have an issue, or were you just pointing out some messages?

pheiduck, 2020-04-26:
Had open this for Kernel Development if these critical messages, but I don't have issues with these. - So the issue can closed if not necessary.

closed 2020-04-26 by pelwell