raspberrypi/firmware #1379: Résolution raspberry pi4 Lost

gracious666, 2020-04-26:
Hello 
I have a raspberry pi4 with raspbian
He works perfectly in 4k résolution 
Now and i don t know why after a reboot the higher resolution is 1024x768 and it s impossible to have the good résolution
I reinstall ans format my SD card and it s the same issue 
I try to change the config.txt and no issue
Somebody can help me ?
My TV it s Samsung 4k hdr 
My xbox one x and my switch work perfectly at the good résolution

Thx

santricity, 2020-04-29:
The same happened to me today. I can't make it full HD anymore on my full HD TV set.

gracious666, 2020-04-29:
What do you try ?
Have you a solution ?

santricity, 2020-04-29:
I haven't fixed it yet. My original hdmi settings in config.txt (worked properly until today):
hdmi_group=0
disable_overscan=1
dtoverlay=vc4-fkms-v3d
With those settings my Rpi 4 was detecting my TV resolution 1080p @60Hz. No problems with kodi that was also running with 1080p. 

After today's restart my resolution changed to 1024x768 in kodi (kodi is dsiplayed in small window). During boot process it's sets resolution to 640x480.

What I have tried so far:
1. (forced 1080p @ 60Hz by group=1 and mode=16, enabled vc4-fkms-v3d)
disable_overscan=1
hdmi_group=1
hdmi_mode=16
dtoverlay=vc4-fkms-v3d
That gives me 1080p at boot, but in the middle of the boot the screen goes black and TV says resolution is not supported. I can hear that Kodi starts up as I have a sound of library refresh at Kodi startup. 

I can ssh to the pi, tvservice -s gives me:
state 0x6 [DVI CUSTOM RGB full 16:9], 1920x1080 @ 60.00Hz, progressive

2. (forced 1080p @ 60Hz by group=1 and mode=16, disabled vc4-fkms-v3d)
disable_overscan=1
hdmi_group=1
hdmi_mode=16
#dtoverlay=vc4-fkms-v3d
Boot process goes to the end @1080p, my TV does not complain about unsupported resolution, but Kodi doesn't start up

tvservice -s:
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

uname -a:
Linux 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

I also ran rpi-update, but didn't solve this issue.

santricity, 2020-04-29:
It looks my original raspberry pi HDMI cable got broken. I had a spare one (also the original raspberry pi) and after replacing it everything came back to normal - FullHD 1080p@60Hz with vc4-fkms-v3d enabled is now displayed properly.

gracious666, 2020-04-30:
If i understand the HDMI cable is the problem ? I ll try with an another 
Thx for your answers 
I check this ans i say to you if the rpi runs properly

santricity, 2020-04-30:
Yes, it was a cable in my case. I have no idea why, I haven't touched it since I installed it half year ago. But after replacing HDMI cable everything came back to normal.

gracious666, 2020-04-30:
Same for me i never touched the cable 
I have read in another forum it s the micro hdmi thé problem , nobody say why

gracious666, 2020-05-04:
Hi
I changed the cable and now the rpi runs perfectly and i test it a long time
I have no Idea why lol
Have a nice day
thx

JamesH65, 2020-05-04:
Excellent, please close the issue if it is all fixed.

closed 2020-05-04 by gracious666