raspberrypi/firmware #1381: The enforced open drain problem appears after connecting the PCA9685

roxanaluca, 2020-04-28:
**Describe the bug**
I have connected PCA9685 and Arducam Multi Camera Adapter on Raspberry Pi. The first pin supplies 3.3V on Arducam Adapter, and at the same time the PCA9685 motor driver. The Raspberry Pi detect the PCA9685 connection via detect -y 3.
**To reproduce**
Adding a new bus 3 in /boot/config in config.txt
start_x=1
gpu_mem=128
dtparam=i2c_arm=on
dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=1,i2c_gpio_sda=16,i2c_gpio_scl=26

**Expected behaviour**
The error should not happen because I measured the voltage in Raspberry and the PCA9685. I follow the Adafruit guideline.

**Actual behaviour**
It is possible that the servo motors wouldn't work properly.

**System**

* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage5
* Which firmware version (`vcgencmd version`)?
Feb 12 2020 12:39:27
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start_x)
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

**Logs**
dmesg | grep i2c
[    3.033396] i2c /dev entries driver
[    4.707228] gpio-16 (3.i2c): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    4.707308] gpio-26 (3.i2c): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    4.707769] i2c-gpio 3.i2c: using lines 16 (SDA) and 26 (SCL)

roxanaluca, 2020-05-01:
I put pull up resistances between SDA and SCL pins and it doesn't work.
Please let me know what to do.
Many thanks

pelwell, 2020-05-01:
The Pi GPIO pins don't have an "open drain" mode - they do outputs and inputs (and the alternate functions). What you are seeing is a warning about what something sees as an invalid or incomplete configuration in Device Tree, rather than a real problem with the hardware. I'll try to understand the issue to see if we can avoid the warning, but until then I don't think there is any chance of you damaging the servo motor as a result of the "problem" that causes the message.

pelwell, 2020-05-01:
You do, however, need pull-ups on any I2C bus. Pis have built-in 1K8 pullups on the hardware I2C buses.

Is the servo motor not working for you? I didn't understand that from your original report - I thought you were suggesting a possible problem rather than an actual one.

roxanaluca, 2020-05-01:
The servo motors are working properly. I was worried about this problem.

closed 2020-05-01 by roxanaluca

pelwell, 2020-05-02:
The i2c-gpio overlay has been updated to avoid the alarming warnings. This does not change the functionality at all. The change is in our Linux kernel source repo, and will appear later in the Raspbian kernel package when it is next rebuilt.

pelwell, 2020-06-11:
The current rpi-update kernel includes the fixed overlay.

stefanklug, 2021-01-08:
Hi @pelwell, I stumbled upon this issue and I believe the solution might be dangerous for the Pi and the connected i2c devices. Please correct me if I'm wrong. 
By specifying i2c-gpio,scl-open-drain / sda-opendrain, the i2c-gpio driver requests the gpio as GPIOD_OUT_HIGH instead of GPIOD_OUT_HIGH_OPEN_DRAIN. This effectifely supresses the warning, but also disables the open drain emulation mode inside gpiolib (where a high open drain output is emulated by switching the GPIO to input mode).

The i2c-gpio driver will now actually drive the lines high instead of letting them be pulled up.
This will lead to conflicts when a i2c device tries to pull the same line low (for example to do clock stretching). The stronger device will win and the weaker might end up with defective hardware.

Best regards,
Stefan

pelwell, 2021-01-08:
You ought to be able to see contention like that on an oscilloscope - do you?

stefanklug, 2021-01-08:
@pelwell Yes, sure, attached are some measurements.
My setup is as follows:
- RPi 3B
- no i2c device attached
- 4.7k Pullup Pin 23 to 3.3V (blue line)
- 4.7k Pullup Pin 24 to 3.3V (yellow line)
- 10k Pulldown Pin 24 to GND
The 10k resistor leads to a slightly reduced line level of around 2.3V on Pin 24 (SCL)

Now I load the original i2c-gpio-overlay with default params.
Idle state (here you see the reduced SCL level):
[image]

After sending something to the bus `i2ctransfer -y -f 11 w1@0x36 0x00` (SCL is correctly beeing pulled down):
[image]

Now I load the i2c-gpio-overlay with added i2c-gpio,scl-open-drain
Idle state (here you can see, that SCL is now forced to 3.3V, my 10k Pulldown has no effect anymore)
[image]
[image]