raspberrypi/firmware #1382: MMAL_COMPONENT_DEFAULT_CAMERA_INFO reports "testc" as name for the HQ camera

imbens, 2020-05-02:
**Describe the bug**
The MMAL_COMPONENT_DEFAULT_CAMERA_INFO reports "testc" as name for the HQ camera in the MMAL_PARAMETER_CAMERA_INFO_T structure.

**To reproduce**
Query the MMAL_COMPONENT_DEFAULT_CAMERA_INFO with mmal_port_parameter_get (theCameraInfoComponent->control, &theCameraInfo.hdr);

**Expected behaviour**
I would expect a more informative name such as "OV5647" for the Camera Module v1.

**Actual behaviour**
The name was reported as "testc"

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 Model B Rev 1.1
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
* Which firmware version (`vcgencmd version`)?
Feb 12 2020 12:37:37 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start_x)
* Which kernel version (`uname -a`)?
Linux hjdevpi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**Logs**

**Additional context**

JamesH65, 2020-05-02:
@naushir @6by9 FYI

6by9, 2020-05-02:
Yes, we need to push an update to rename it.

popcornmix, 2020-05-12:
Have you tested with latest rpi-update firmware? There were some commits related to this.

imbens, 2020-05-14:
With the latest firmware, the name is reported as "imx477". Unfortunately, the latest firmware does not work well with my application. With `dtoverlay=vc4-fkms-v3d` I get a garbled screen, where pixels seem transposed in a strange way. With `dtoverlay=vc4-kms-v3d` I get the following in dmesg: ```[   47.782471] broken atomic modeset userspace detected, disabling atomic
[   59.353497] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   84.313477] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   94.553465] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[  104.793468] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
```

JamesH65, 2020-05-14:
Just a note, KMS is still under test on the Pi4, there are likely to be issues with it.  You should be using FKMS.

popcornmix, 2020-05-14:
A photo of the garbled pixels might give us some clues. Also a test application we can run that shows the problem would also be useful.
We're not aware of a bug that gives the behaviour you describe so it's unlikely to be fixed without someone providing a means to reproduce.

imbens, 2020-05-14:
[image]
[image]
The first one is a photo of the screen, the second one is a dump of what the screen should look like.

imbens, 2020-05-15:
My display is in 1920x1080 mode. When I draw a single 1920x1080 picture I get a similarly garbled result. But when I preprocess my image with the following code:
`        SurfacePtr original = PictureManager::Global ().GetPicture ("media/test/sandtest.png");
        SurfacePtr preprocessed = new Surface (1920, 1080, original->mBytesPerPixel);
        uint8_t *p = preprocessed->GetPixels ();
        for (int i8 = 0 ; i8 < 135 ; i8++) {
            for (int i7 = 0 ; i7 < 2 ; i7++) {
                for (int i6 = 0 ; i6 < 4 ; i6++) {
                    for (int i5 = 0 ; i5 < 60 ; i5++) {
                        for (int i4 = 0 ; i4 < 2 ; i4++) {
                            for (int i3 = 0 ; i3 < 2 ; i3++) {
                                for (int i2 = 0 ; i2 < 2 ; i2++) {
                                    uint8_t *q = original->GetPixels (16*i5, 16*i2 + 64*i3 + 128*i4 + 4*i6 + 32*i7 + 256*i8);
                                    for (int i1 = 0 ; i1 < 4 ; i1++) {
                                        for (int n = 0 ; n < original->mBytesPerPixel ; n++) {
                                           *p++ = *q++;
                                        }
                                    }
                                }
                            }
                        }
                    }
                }
            }
        }
`
the preprocessed image is shown almost correctly. GetPixels (row, column) returns a pointer to the pixel data of the pixel at the specified row and column. Pixels are stored, starting with the topleft pixel first.

6by9, 2020-05-15:
```vcgencmd dispmanx_list``` please when your image is on display with FKMS.

imbens, 2020-05-15:
The output of `vcgencmd dispmanx_list` is:
```
display:2 format:XRGB8888 transform:0 layer:-127 1920x1080 src:0,0,1920,1080 dst:0,0,1920,1080 cost:1156 lbm:0
```

imbens, 2020-05-18:
I created a small test program at [https://github.com/imbens/opengltest](https://github.com/imbens/opengltest). After `sudo rpi-update stable` on Raspbian Buster Lite it shows a red rectangle on a background that cycles between green and blue. After `sudo rpi-update`, the red rectangle is distorted.

imbens, 2020-05-25:
I can fix the problem by changing the line [](https://github.com/imbens/opengltest/blob/40d9de1393a4d536c41f73051d5b0688c256a754/opengltest.c#L166). When I change `mScreenWidth` to `mScreenWidth+8` (making the X window slightly larger than the screen) or when I move the window the problem disappears. When I change `mScreenHeight` to `mScreenHeight+8` the problem persists.