raspberrypi/firmware #1383: No Signal for Raspberry pi 4 and a certain screen works on rpi 3 using same sd card

joyrider3774, 2020-05-03:
Hello,

I had a No Signal from Raspberry pi 4 hdmi on this screen https://shop.pimoroni.com/products/hdmi-10-lcd-screen-kit-1024x768 while it works on the raspberry pi 3 with the same sd card. I had tried numerous things like verifying EDID info on the rpi and dumping it forcing the rpi4 to make use of it but it made no difference. I also tried force_hdmi_hotplug as well as specifying custom hdmi modes which all failed. I've spend days messing with the boot/config.Txt until i read that the config_hdmi_boost was being ignored on raspberry pi 4. and i know the screen works with my pc as well as the raspberry pi 3. The screens also shows a led being active on the lcd driver board when it seems to get a signal and this never happens with a pi4 attached.

So i had an idea and attached an hdmi switcher between the rpi 4 and the screen. This hdmi switcher is powered so i thought it might amplify the hdmi signal as well but i'm still not sure of that
and hello after days of trying i have a signal on the rpi 4 with this screen.

But using an hdmi switcher is no option for me as this is for a picade, is there any way i can amplify the signal on a rpi 4 even just for testing purposes by a recompile of something to verify if thats really the problem or can i verify this with a rpi 3 and lowering config_hdmi_boost to 1 ?

just in case here is the base 64 edid data if needed

AP///////wBBMgEAAAAAAAEAAQSiEAwAgAAAAAAAAAAAAAAhCIABAAEAAQABAAEAAQABAAEAAQAA
AAAAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAEAAA
AAAAAAAAAAAAAAAAAGw=
 
* raspberry pi 4 4gb
* os:
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

* Which firmware version (`vcgencmd version`)?
Apr 27 2020 14:33:37
Copyright (c) 2012 Broadcom
version 3a8f1793b758d6fb7f375edaa260e069ecd34c88 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux retropie 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

popcornmix, 2020-05-12:
There is no hdmi boost on Pi4. A low quality cable may be the cause of the issue. Do you have a different one you can try?

Squuiid, 2020-05-18:
@popcornmix please don't take this the wrong way, but 'no signal' seems to be quite a pervasive issue with the Pi 4 and while I'm sure in some, relatively few by comparison, cases it is HDMI cable related not all are. I too have problems with HDMI handshaking where the Pi 3 works fine and the 4 does not. I have tried the official Pi cable, a ton of different combinations of settings in config, expensive HDMI 2.0 cables, cheap ones. Nothing works. There needs to be a concerted overhaul of HDMI handshaking in the firmware because currently it just isn't fit for purpose and pales in comparison to the Pi 2 and 3 models.

timg236, 2020-05-18:
@squuid There will be no rewrite of the Raspberry Pi HDMI firmware because it's all moved to KMS. btw: Ranting about something is the best way to ensure engineers will ignore you whether or not your complaint is valid.

Squuiid, 2020-05-18:
Apologies. And thank you for the quick response. Where can I file a proper and full bug report to better assist with getting this issue resolved?
Or should I just wait for a new KMS driver to arrive as that will fix?

JamesH65, 2020-05-18:
> @timg236 Apologies. And thank you for the quick response. Where can I file a proper and full bug report to better assist with getting this issue resolved?
> Or should I just wait for a new KMS driver to arrive as that will fix?

It's not pervasive - it's very rare. There does seem to be a issues in some cases, but we have never been able to replicate in house. There is one specific resolution that we know does not work, native 1366x768 devices. If yours is one of those then we do have a monitor that exhibits that behaviour although it's going to be awkward to fix.

Comparing with Pi3 is not helpful as they have different HDMI hardware. 

For a bug report, this repo is the right place, will need details of the Pi model, version of software, the  type of device you are attaching to and a dump of the EDID (tvservice -d edid.bin  then base64 edid.bin, copy the result to the bug report) and any other information you think relevant.

Squuiid, 2020-05-18:
It's vanilla 1080p@60hz tbh. I have the following attached to an Epson 5020UBe projector with no issues; Humax HDR-2000T, Apple TV 4K, Sony PS4, Xbox One X, Nintendo Switch and a Pi 3B+.
The Pi 4 shows no signal however.
I'll work on getting more info to you. Thanks again.

joyrider3774, 2020-05-18:
hello just to mention: alle my cables worked on all other setups being it rpi3 pc attached, nuc attached (which also uses hdmi mini/micro can't remember exactly the name) but failed to work on rpi4 except when using a passive hdmi booster which i bought  here https://www.bol.com/nl/rnwy/account/order_details/2946756810?language=en

that device makes the screen work on the rpi4. 

But just to be sure i also bought the official rpi4 hdmi cable and with that cable i do not have to use the hdmi signal booster anymore

@Squuiid  if you happen to have a powered hdmi switcher which happens to boost the signal or if you got a passive hdmi booster like the one i bought you could try those. OR buy the official rpi4 hmdi cable as that one works for me also even though all other cables worked with rpi3

Squuiid, 2020-05-18:
@joyrider3774 Thanks. I actually have the official cable, as well as several adapters, and even an HDMI2.1 certified cable (expensive!) and still now joy.
Thanks for the suggestion however.

pelwell, 2020-05-18:
@joyrider3774 Can you explain how you managed to use the same cables on Pi 3 and Pi 4?

joyrider3774, 2020-05-18:
@pelwell i had bought an hdmi micro to hdmi convertor (which works as i tested that on my intel nuc) and the cable i used to attach the screen and the hdmi micro to hdmi convertor was the same hdmi cable i used for attaching my pi3 to the same screen. I know every part works in other setups so they are not to blame. I even tried many different hdmi cables. I also tried a direct micro hdmi to hdmi cable which also works on my intel nuc when attaced to tv or pc screen. The screen worked when using with rpi3 without hdmi boost setting. So yeah even though every part works on different setups it does not work with the pi4. So eventually i had bought those passive hdmi boosters which works with the cables that did not seem to work in the pi4 the screen got a signal then but i also had bought the official rpi4 cable and that cable works without the passive hdmi boosters.

timg236, 2020-05-18:
The KMS is under development and is making rapid progress. IIRC it's being discussed on the DRI/Kernel mailing lists but is probably only of interest to kernel developers or people who reasonable knowledge of the DRI/KMS stack.
It makes things more visible which may be helpful given the wide variety of monitors and display modes but right now it sounds like a connector / booster issue.

sde1000, 2020-08-30:
> There is one specific resolution that we know does not work, native 1366x768 devices.

Huh!  That is interesting.  I've been trying to diagnose a pi 4 network boot problem for the last couple of days: I don't get the "rainbow" screen from start4.elf.  The display is a native 1360x768 Asus VT168.  The bootloader diagnostics screen displays correctly, then there's no HDMI output after start4.elf is started.  The kernel and initrd are loaded, still with no HDMI output, and it eventually starts displaying the console once the root squashfs has been mounted and (presumably) more driver modules have been loaded.

Here's the EDID from tvservice -d: [asus-vt168-edid.gz](https://github.com/raspberrypi/firmware/files/5147546/asus-vt168-edid.gz)
Here's the log from vcdbg log msg:
```
018560.652: genet: LINK STATUS: speed: 1000 full duplex
018561.488: genet: GENET START: 128 8 8
018563.912: genet: GENET: UMAC_START 0xdca6327e 0x1ae30000
018568.999: brfs: File read: /mfs/sd/config.txt
018569.443: brfs: File read: 451 bytes
018626.162: brfs: File read: /mfs/sd/config.txt
018632.513: brfs: File read: 451 bytes
019079.399: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
019081.857: *** Restart logging
019124.050: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
019124.066: HDMI0: hdmi_pixel_encoding: 300000000
019124.078: HDMI1: hdmi_pixel_encoding: 300000000
020997.389: brfs: File read: /mfs/sd/initrd.img-5.4.51-v7l+
020997.420: Loading 'initrd.img-5.4.51-v7l+' to 0x0 size 0x8eb8a6
021011.261: initramfs loaded to 0x2e714000 (size 0x8eb8a6)
021015.717: dtb_file 'bcm2711-rpi-4-b.dtb'
021015.750: brfs: File read: 9353382 bytes
021028.263: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
021028.294: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xb96f
021058.679: brfs: File read: 47471 bytes
021061.413: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
021197.714: brfs: File read: 1423 bytes
021200.149: brfs: File read: /mfs/sd/config.txt
021201.101: brfs: File read: 451 bytes
021203.905: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
021238.603: Loaded overlay 'disable-bt'
021321.180: brfs: File read: 1073 bytes
021323.874: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
021365.039: Loaded overlay 'vc4-fkms-v3d'
021365.196: dtparam: audio=on
021490.931: brfs: File read: 1238 bytes
021493.327: brfs: File read: /mfs/sd/cmdline.txt
021493.429: Read command line from file 'cmdline.txt':
021493.465: 'console=serial0,115200 boot=live netboot=nfs nfsroot=192.168.73.30:/home/steve/till-boot/raspi-boot elevator=deadline till-boot-version=20200522170418 till-boot-config=http://192.168.73.30/till-boot/drone.yaml net.ifnames=0 ip=dhcp'
022770.025: brfs: File read: 233 bytes
024047.766: brfs: File read: /mfs/sd/kernel7l.img
024047.802: Loading 'kernel7l.img' to 0x8000 size 0x5b7fc0
024047.842: Device tree loaded to 0x2e708000 (size 0xbf31)
024051.929: genet: RX_DISCARD: 0
024051.943: genet: 
024051.967: genet:   GR64: 0x00000005 
024051.994: genet:   GR64: 0x00000018 
024052.018: genet:  GR127: 0x00000002 
024052.042: genet:  GR255: 0x00000007 
024052.057: genet: 
024052.080: genet:  GR511: 0x00000001 
024052.101: genet: GR1023: 0x00003ac0 
024052.122: genet: GR1518: 0x00000000 
024052.144: genet:  GRMGV: 0x00000000 
024052.156: genet: 
024052.178: genet: GR2047: 0x00000000 
024052.204: genet: GR4095: 0x00000000 
024052.227: genet: GR9216: 0x00003ae7 
024052.251: genet:  GRPKT: 0x00f6925a 
024052.266: genet: 
024052.290: genet:  GRBYT: 0x00000004 
024052.312: genet:  GRMCA: 0x00000001 
024052.333: genet:  GRBCA: 0x00000000 
024052.355: genet:  GRFCS: 0x00000000 
024052.367: genet: 
024052.391: genet:  GRXCF: 0x00000000 
024052.415: genet:  GRXPF: 0x00000000 
024052.439: genet:  GRXUO: 0x00000000 
[... 34 lines cut ...]
024053.159: genet:  GTFCS: 0x00000000 
024053.175: genet: 
024053.198: genet:  GTOVR: 0x00000000 
024053.220: genet:  GTDRF: 0x00000000 
024053.242: genet:  GTEDF: 0x00000000 
024053.263: genet:  GTSCL: 0x00000000 
024053.277: genet: 
024053.299: genet:  GTMCL: 0x00000000 
024053.322: genet:  GTLCL: 0x00000000 
024053.346: genet:  GTXCL: 0x00000000 
024053.370: genet:  GTFRG: 0x00000000 
024053.385: genet: 
024053.409: genet:  GTNCL: 0x00000000 
024053.431: genet:  GTJBR: 0x00000000 
024053.452: genet:  GTBYT: 0x000ebeb9 
024053.476: genet:  GTPOK: 0x00003aee 
024053.489: genet: 
024053.512: genet:   GTUC: 0x00003aee 
024053.526: genet: 
024053.546: genet: GENET STOP: 0
027862.410: vchiq_core: vchiq_init_state: slot_zero = 0xde580000, is_master = 1
027868.857: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
027876.092: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000051
037840.281: TV service:host side not connected, dropping notification 0x00000008, 0x00000004, 0x00000000
```

sde1000, 2020-11-18:
The 1360x768 Asus VT168 is now working correctly with the current firmware.

NickMortimer, 2020-12-20:
@sde1000 is there any way to get the touch screen part going?

sde1000, 2020-12-20:
The touch screen part is just connected over USB and works fine once the X server starts.

NickMortimer, 2020-12-20:
Ah I'm using QLC+ to control come lights using buster 20200830 I guess that could be the problem then

sde1000, 2020-12-20:
Assuming you're using [the pre-built QLC+ on Pi image](https://www.qlcplus.org/raspberry.html), I think you'll have to check with the maintainer. I believe the underlying library (Qt 5.11.3 directly running on OpenGL with no X server) does support touch interfaces but might need some extra configuration. I don't have any experience of this myself.

This is now very off-topic for this github issue!