raspberrypi/firmware #1384: chromium freezes randomly and causes kernel errors

fjleon1980, 2020-05-03:
I use my raspberry pi 4 to watch videos all day at hideout.tv. This is done for about 16 hours a day. Around each 2 days, chromium freezes and i have to reboot the device, not even doing killall -9 chromium-browse and restarting fixes it. Today as always i killed chromium to reboot and got a kernel error in a ssh console i had opened. I ran dmesg and here are its contents. Note that it might be related to issue https://github.com/raspberrypi/firmware/issues/1298 which i reported a long time ago.

153084.025337] vc_sm_cma_vchi_rx_ack: received response 65537, throw away...
[167073.474346] [vc_sm_cma_ioctl_alloc]: something failed - cleanup. ret -512
[167073.474374] ------------[ cut here ]------------
[167073.474406] WARNING: CPU: 3 PID: 2724 at mm/memory.c:2299 apply_to_page_range+0x324/0x330
[167073.474416] Modules linked in: cmac fuse rfcomm bnep 8021q garp stp llc hci_uart btbcm serdev bluetooth ecdh_generic cdc_acm vc4 drm_kms_helper brcmfmac brcmutil sha256_generic v3d gpu_sched snd_soc_core cfg80211 rfkill snd_compress bcm2835_codec(C) snd_pcm_dmaengine snd_bcm2835(C) drm v4l2_mem2mem raspberrypi_hwmon hwmon snd_pcm videobuf2_dma_contig snd_timer drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt fb_sys_fops bcm2835_v4l2(C) snd bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[167073.474618] CPU: 3 PID: 2724 Comm: Chrome_ChildIOT Tainted: G         C        4.19.97-v7l+ #1294
[167073.474627] Hardware name: BCM2835
[167073.474655] [<c0212e04>] (unwind_backtrace) from [<c020d5e0>] (show_stack+0x20/0x24)
[167073.474678] [<c020d5e0>] (show_stack) from [<c09b15c8>] (dump_stack+0xe0/0x124)
[167073.474699] [<c09b15c8>] (dump_stack) from [<c0222544>] (__warn+0x104/0x11c)
[167073.474717] [<c0222544>] (__warn) from [<c0222694>] (warn_slowpath_null+0x50/0x58)
[167073.474734] [<c0222694>] (warn_slowpath_null) from [<c039d3e0>] (apply_to_page_range+0x324/0x330)
[167073.474755] [<c039d3e0>] (apply_to_page_range) from [<c021940c>] (cma_allocator_free+0xa4/0xec)
[167073.474771] [<c021940c>] (cma_allocator_free) from [<c02184f8>] (__arm_dma_free+0x16c/0x17c)
[167073.474787] [<c02184f8>] (__arm_dma_free) from [<c0218578>] (arm_dma_free+0x34/0x3c)
[167073.474826] [<c0218578>] (arm_dma_free) from [<bf157560>] (vc_sm_release_resource+0x158/0x198 [vc_sm_cma])
[167073.474873] [<bf157560>] (vc_sm_release_resource [vc_sm_cma]) from [<bf1581e4>] (vc_sm_dma_buf_release+0x54/0xa8 [vc_sm_cma])
[167073.474902] [<bf1581e4>] (vc_sm_dma_buf_release [vc_sm_cma]) from [<c07316c8>] (dma_buf_release+0x60/0x164)
[167073.474927] [<c07316c8>] (dma_buf_release) from [<c03d66fc>] (__fput+0xa4/0x1e8)
[167073.474947] [<c03d66fc>] (__fput) from [<c03d68b0>] (____fput+0x18/0x1c)
[167073.474968] [<c03d68b0>] (____fput) from [<c02426dc>] (task_work_run+0xbc/0xe0)
[167073.474989] [<c02426dc>] (task_work_run) from [<c0233e74>] (get_signal+0x898/0x8cc)
[167073.475007] [<c0233e74>] (get_signal) from [<c020c73c>] (do_signal+0x84/0x474)
[167073.475021] [<c020c73c>] (do_signal) from [<c020cd38>] (do_work_pending+0xdc/0xf4)
[167073.475037] [<c020cd38>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
[167073.475046] Exception stack(0xd61fdfb0 to 0xd61fdff8)
[167073.475058] dfa0:                                     00000035 80404a5a acbfcc80 74616e2f
[167073.475072] dfc0: 059aa8e6 00033000 00000035 00000036 acbfcc80 00004999 00000001 b6f51968
[167073.475082] dfe0: b6e59154 acbfcc64 b6e45224 b5523518 20000010 00000035
[167073.475133] ---[ end trace a786da0de147a603 ]---
[167073.475170] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[167073.475192] pgd = c0d02611
[167073.475208] [00000004] *pgd=17426003, *pmd=00000000
[167073.475234] Internal error: Oops: a06 [#1] SMP ARM
[167073.475248] Modules linked in: cmac fuse rfcomm bnep 8021q garp stp llc hci_uart btbcm serdev bluetooth ecdh_generic cdc_acm vc4 drm_kms_helper brcmfmac brcmutil sha256_generic v3d gpu_sched snd_soc_core cfg80211 rfkill snd_compress bcm2835_codec(C) snd_pcm_dmaengine snd_bcm2835(C) drm v4l2_mem2mem raspberrypi_hwmon hwmon snd_pcm videobuf2_dma_contig snd_timer drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt fb_sys_fops bcm2835_v4l2(C) snd bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[167073.475396] CPU: 3 PID: 2724 Comm: Chrome_ChildIOT Tainted: G        WC        4.19.97-v7l+ #1294
[167073.475408] Hardware name: BCM2835
[167073.475439] PC is at vc_sm_release_resource+0x88/0x198 [vc_sm_cma]
[167073.475463] LR is at _cond_resched+0x30/0x50
[167073.475476] pc : [<bf157490>]    lr : [<c09c996c>]    psr: 60000013
[167073.475488] sp : d61fdd70  ip : 00000100  fp : d61fddac
[167073.475500] r10: e2000000  r9 : ef6e9800  r8 : ef2a5a10
[167073.475513] r7 : 00000000  r6 : c0a01454  r5 : bf15d300  r4 : cfa1a500
[167073.475525] r3 : 00000000  r2 : 00000000  r1 : 00000200  r0 : dc782b40
[167073.475540] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[167073.475554] Control: 30c5383d  Table: 1aa4ed40  DAC: 55555555
[167073.475568] Process Chrome_ChildIOT (pid: 2724, stack limit = 0x257a493c)
[167073.475581] Stack: (0xd61fdd70 to 0xd61fe000)
[167073.475596] dd60:                                     e2000000 00000000 00000000 bf1581d0
[167073.475611] dd80: e2000000 00000000 cfa1a500 cfa1a510 ef990310 cded4330 00000000 00000008
[167073.475628] dda0: d61fddc4 d61fddb0 bf1581e4 bf157414 d3c02180 ef6e9800 d61fddec d61fddc8
[167073.475643] ddc0: c07316c8 bf15819c d603bf00 ef6e9800 ef990310 cded4330 00000000 ef6e9800
[167073.475659] dde0: d61fde2c d61fddf0 c03d66fc c0731674 00000000 00000000 c02426c0 d603bf08
[167073.475674] de00: d7559438 00000000 d755941c d7558f00 d7559438 c10ace10 d603bf00 00000036
[167073.475689] de20: d61fde3c d61fde30 c03d68b0 c03d6664 d61fde64 d61fde40 c02426dc c03d68a4
[167073.475705] de40: d61fdfb0 c1004d88 fffffe00 d61fdec4 b5523518 b552351c d61fdebc d61fde68
[167073.475720] de60: c0233e74 c024262c 00000000 00000000 6f726843 7542656d 2e2e0066 2f2e2e2f
[167073.475735] de80: d92eb9c0 6c2f7866 d7d45b80 d61fdec4 00000000 d61fdfb0 c1004d88 fffffe00
[167073.475751] dea0: d61fdec4 b5523518 b552351c 00000036 d61fdf8c d61fdec0 c020c73c c02335e8
[167073.475766] dec0: d61fc000 00000036 d61fdf6c d61fded8 c03e98a4 bf159528 d61fdf40 d61fdf3c
[167073.475782] dee0: 00000040 acbfd4e0 d61fdf04 d61fdef8 c03f5be4 c03f52b8 d61fdf14 03090087
[167073.475797] df00: c03f5c58 c1004d88 d87ed880 00000040 acbfd4e0 c087ae5c d8ef7840 c1004d88
[167073.475813] df20: d61fdf94 d61fdf30 c087ae5c c03f5be4 059aa8e6 acbfcc80 00000035 03090087
[167073.475828] df40: d61fdf5c c0e9bad8 c020ccc0 c02011c4 d61fdfb0 c02011c4 d61fc000 03090087
[167073.475843] df60: d61fdf8c 00000085 d61fc010 c02011c4 d61fdfb0 c02011c4 d61fc000 00000036
[167073.475859] df80: d61fdfac d61fdf90 c020cd38 c020c6c4 059aa8e6 00033000 00000035 00000036
[167073.475874] dfa0: 00000000 d61fdfb0 c0201034 c020cc68 00000035 80404a5a acbfcc80 74616e2f
[167073.475890] dfc0: 059aa8e6 00033000 00000035 00000036 acbfcc80 00004999 00000001 b6f51968
[167073.475905] dfe0: b6e59154 acbfcc64 b6e45224 b5523518 20000010 00000035 00000000 00000000
[167073.475949] [<bf157490>] (vc_sm_release_resource [vc_sm_cma]) from [<bf1581e4>] (vc_sm_dma_buf_release+0x54/0xa8 [vc_sm_cma])
[167073.475983] [<bf1581e4>] (vc_sm_dma_buf_release [vc_sm_cma]) from [<c07316c8>] (dma_buf_release+0x60/0x164)
[167073.476010] [<c07316c8>] (dma_buf_release) from [<c03d66fc>] (__fput+0xa4/0x1e8)
[167073.476033] [<c03d66fc>] (__fput) from [<c03d68b0>] (____fput+0x18/0x1c)
[167073.476056] [<c03d68b0>] (____fput) from [<c02426dc>] (task_work_run+0xbc/0xe0)
[167073.476079] [<c02426dc>] (task_work_run) from [<c0233e74>] (get_signal+0x898/0x8cc)
[167073.476100] [<c0233e74>] (get_signal) from [<c020c73c>] (do_signal+0x84/0x474)
[167073.476117] [<c020c73c>] (do_signal) from [<c020cd38>] (do_work_pending+0xdc/0xf4)
[167073.476136] [<c020cd38>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
[167073.476149] Exception stack(0xd61fdfb0 to 0xd61fdff8)
[167073.476163] dfa0:                                     00000035 80404a5a acbfcc80 74616e2f
[167073.476179] dfc0: 059aa8e6 00033000 00000035 00000036 acbfcc80 00004999 00000001 b6f51968
[167073.476193] dfe0: b6e59154 acbfcc64 b6e45224 b5523518 20000010 00000035
[167073.476212] Code: e894000c e5950000 e3a0cc01 e3a01c02 (e5823004) 
[167073.476231] ---[ end trace a786da0de147a604 ]---
[167073.557295] bcm2835_audio soc:audio: failed to close VCHI service connection (status=1)

Linux 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

nixternal, 2021-05-28:
I'm getting a very similar crash, though with VLC where I am looping the same file over & over.
[kern.log](https://github.com/raspberrypi/firmware/files/6562729/kern.log)

popcornmix, 2021-06-02:
@nixternal can you create an issue here: https://github.com/RPi-Distro/vlc/issues/ with instructions to reproduce it.
Include a sample file (or a link to a publicly available file, like Big Buck Bunny)

nixternal, 2021-06-04:
Done deal, thanks!