raspberrypi/firmware #1385: pi4 doesn't support 2560x1440 resolution

illmillrig, 2020-05-10:
please add support for 2560x1440 resolution monitors.  It's one of the most common resolutions ;)

illmillrig, 2020-05-10:
seems you've had lots of problems with this resolution specifically.

https://www.tomshardware.com/news/raspberry-pi-4-wi-fi-not-working

illmillrig, 2020-05-10:
looks like this is a sensitive subject for you guys.  have you really had this much of a hassle with supporting this resolution? It's too bad.  I think a lot of people will buy 27" displays assuming it works with pi4.

illmillrig, 2020-05-10:
so this actually worked on another pi4 I had.  except it seems like the pi4 is not capable of powering two screens at this resolution.  I get constant black tearing and screen blanking. switching both displays to 1080p works fine, or running one display at 1440p is fine.  but not both.

JamesH65, 2020-05-11:
What Hz are the displays running at? You cannot run both at 60fps, that is simply too much for the HW.  You could run 2x1080p60, that is within the pixel limit.

illmillrig, 2020-05-11:
so I'm a bit confused now.

the pi4 can display 3840x2160@60 on screen0, with screen1 at 1920x1080@60. by my count, that's 10,368,000 pixels total.  dual 27" (2560x1440@60) displays are 7,372,800 combined pixels which seems well within the pixel limit, no?

illmillrig, 2020-05-11:
even the two 27" are less pixels than one 4k display. 8,294,400 vs 7,372,800

illmillrig, 2020-05-11:
how does marketing get away with advertising the pi as supporting two 4k displays?

JamesH65, 2020-05-11:
Because it works fine with 2x4kp30 displays. Or one 4kp60 and one 4kp30 IIRC. I know this because I've tried it.

The symptoms you describe are typical of exceeding some bandwidth somewhere. If it's not overall pixels it must be somewhere else. Could be the V3D struggling with something somewhere. Does it go away if you turn off the V3D? Comment out the FKMS line in config.txt. Odd that 2x4k works though.

popcornmix, 2020-05-11:
@illmillrig have you got `hdmi_enable_4kp60=1` in config.txt?
We do boost clocks for 4kp60 users and that change may also help your case.

illmillrig, 2020-05-11:
I tried to enable `hdmi_enable_4k60`, that gave me one display at 2560x1440@60, but the other display was set to 1920x1080@60.

if I comment out the v3d line, then I do not get dual displays at all - only one display, with a resolution of something strange like 1854x956 (dont quote me on those exact numbers). 

enabling v3d gives me dual displays at 2560x1440@59.95.  I'm not sure why it doesn't say 60 here, when other times it does

illmillrig, 2020-05-11:
I gotta question who has dual 4k screens and is fine running them at 30p? is that a practical use case for the pi4? I'm not able to test it, but i imagine this is barely usable also.

In my testing, with one 4k screen at 60p, the pi4 can hardly keep up playing back video, scrolling webpages, etc, etc. it's not really usable. 

so it's just not really able to do either all that well. it can't do dual qhd, and it can't actually really do 4k all that well either.

janrinze, 2020-06-01:
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
gpu_mem=256
hdmi_group=2
hdmi_mode=87
hdmi_cvt=2560 1440 60 3 0 0 1
max_framebuffer_width=2560
max_framebuffer_height=1440
hdmi_pixel_freq_limit=400000000
hdmi_group:1=2
hdmi_mode:1=87
hdmi_cvt:1=2560 1440 60 3 0 0 1
hdmi_pixel_freq_limit:1=400000000

When using this in config.txt both hdmi outputs do 2560x1440 without any problem.

cybercussion, 2023-02-10:
Thanks for this.  I'm trying to do the mini hdmi to a DVI-D but so far I see the logo on boot but once it gets to desktop it's just black.  Think these old Yamakasi monitors are picky about res/refresh.