raspberrypi/firmware #1386: rpi4 usb init failed with 32bit uboot

CXianRen, 2020-05-13:
for some reason, I have to load a new mypi4.dtb (modify form the base dtb ) with uboot just like:
fdtlod mmc 0:1 ${fdt_addr} mypi4.dtb
and
fat addr ${fdt_addr}
then
load the kernel7.img and bootz 
everthing work fine except the usb:
[    0.379602] xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -16
[    0.381474] brcmstb_thermal soc:thermal@7d5d2200: invalid resource

but if I don't load the dtb of myself, just use the addr passed from start4x, the usb is ok.
what should I do ?

timg236, 2020-05-13:
This appears to be a u-boot issue so you need to ask the u-boot maintainers

CXianRen, 2020-05-13:
thanks,  I just want to make sure the usb's device tree won't be modiefied by start4x and bindcode or other firmware

pelwell, 2020-05-13:
The firmware makes many small changes to the normal dtb before passing it to the kernel including things like the memory size. By bypassing the firmware you are missing out on these changes.

It would be better to let the firmware load your mypi4.dtb, allowing it to modify it as necessary, then let u-boot pass it on to the kernel.

CXianRen, 2020-05-13:
@pelwell  thanks, so there is no way to install 2 independent system with different config setting and different dtb?

pelwell, 2020-05-13:
You might be able to make use of the new `os_prefix` config.txt setting (see https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md), but for truly independent systems use two cards.

closed 2025-02-14 by popcornmix