raspberrypi/firmware #1387: how to display more debug log on start4.elf for PI4

figozhang, 2020-05-17:
hi @JamesH65 @pelwell

i have add "uart_2ndstage=1" in the config.txt for PI4.

Here is the log:
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
Loading recover4.elf hnd: 0x00000000
Failed to read recover4.elf error: 6
Loading recovery.elf hnd: 0x00000000
Failed to read recovery.elf error: 6
Loading start4.elf hnd: 0x000002d8
Loading fixup4.dat hnd: 0x00000594
MEM GPU: 76 ARM: 948 TOTAL: 1024
FIXUP src: 128 256 dst: 948 1024
Starting start4.elf @ 0xfec00200

but, my question is how can i get more debug log of start4.elf?
when i using PI3, it can see such log:
MESS:00:00:02.593428:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:02.608463:0: Failed to load Device Tree file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:02.614263:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:02.624294:0: brfs: File read: /mfs/sd/benos3.bin
MESS:00:00:02.628167:0: Loading 'benos3.bin' to 0x80000 size 0x5d0
MESS:00:00:02.634108:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:02.642963:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:02.649253:0: uart: Baud rate change done...
MESS:00:00:02.652682:0: uart: Baud

my question, how i can see such log on PI 4?

timg236, 2020-05-18:
This was disabled on Pi 4 because enabling the UART logging by default if the bootloader had left it enabled cause problems because it conflicted with the BT UART. Logging is initialised before config.txt is read which made it a non-trivial problem to resolve.
On Linux "vcdbg log msg" will retrieve the logs

figozhang, 2020-05-19:
hi @timg236 
i check the "vcdbg log msg" it is works.
is it possible disable the BT UART in config.txt to enable the start4.elf log? or let the log of start4.elf put into mini uart?

samueldr, 2020-06-08:
Hi,

> This was disabled on Pi 4

Can an alternative version be released that has the option turned on? Or figure out a method to configure it on/off via hex editing or something of the like for advanced users?

I can't use `vcdbg log msg` to read the log if my OS doesn't start. It kinda defeats the purpose of this log.

ptesarik, 2020-06-08:
Alternatively, could you add a `config.txt` option which makes the firmware dump the in-memory log buffer to the serial console immediately after it is parsed?

figozhang, 2020-06-09:
@ptesarik what option can dump to log buffer to serial console in config.txt?

timg236, 2020-06-12:
This is fixed on the internal dev branch and will appear in rpi-update, I'll close the bug when the code is available.
N.B. This will also require an updated bootloader EEPROM because the fix required changing the debug UART handover between bootloader and start.elf

figozhang, 2020-06-12:
that will be great, when will release it?

timg236, 2020-06-12:
Probably not until next week, I'll close this bug with details once it's released

figozhang, 2020-06-12:
Hopeful will have some guide or document about how to use it.

timg236, 2020-06-12:
It's the same as before
https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md
The doc will be updated to indicate the minimum firmware version

sde1000, 2020-08-29:
Any updates on this?  I'm trying to debug a network boot issue on a pi 4 4Gb version 1.3, and the last thing I see on the serial output is:
```
Starting start4db.elf @ 0xfec00200 partition -1
```
I'm using the bootloader from 31st July and start*.elf from `raspberrypi-bootloader 1.20200819-1`

I have the following in config.txt:
```
[pi4]
uart_2ndstage=1
start_debug=1
dtoverlay=disable-bt
```
I can see from TFTP server logs that start4.elf fetches many files, ending with `kernel7l.img`.  After this, the screen is blank ("HDMI no signal" on the monitor) and the green LED is emitting four(?) very rapid pulses about once per second, which stops if a SD card is inserted.

SatishKumarReddyPokala, 2020-09-04:
Hi,

I am facing the same issue u-boot.bin is not loaded on the console.
It just stuck at Starting start4db.elf @ 0xfec00200.
I used latest start4.elf and fixup4.dat from git repo.

Is this issue was fixed in any old version so that i can use old version only ?
Please update if anyone knows about it.

Thanks

SatishKumarReddyPokala, 2020-09-09:
Hi,

For Raspberry pi 4 U-Boot porting process is little different.
Somehow i managed to build U-Boot successfully and tested, U-Boot prompt is visible on serial terminal.
Hopefully following links will helps for this issue.

https://github.com/SatishKumarReddyPokala/RPI4/tree/master/U-Boot/Build_U-Boot
https://archlinuxarm.org/packages/armv7h/uboot-raspberrypi/files/PKGBUILD

figozhang, 2020-09-19:
@timg236 how to do that?

figozhang, 2020-09-21:
i have try the new PI FW, it works

timg236, 2020-09-21:
So long as you have Raspberry Pi OS firmware 2020-08-20 and the latest production bootloader 2020-09-03 uart_2ndstage=1 will behave as described in 
https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md

@JamesH65  I think we can close this as resolved

closed 2020-09-21 by pelwell