raspberrypi/firmware #1392: `dtoverlay=vc4-fkms-v3d` causes blank screen

MartenBE, 2020-05-21:
**Describe the bug**
When hooking up a Rasberry Pi 4B to a AOC U3277FWQ monitor with 4K@60hz enabled causes a black screen on boot. The problems go away when the `dtoverlay=vc4-fkms-v3d` is removed from the boot config. This thread describes the problem very accurately: https://www.raspberrypi.org/forums/viewtopic.php?t=255563

**To reproduce**
* Connect the Rasberry Pi 4B to a AOC U3277FWQ.
* Enable 4K@60hz 
* Boot

This behavior has been confirmed by two separate cases as described in https://www.raspberrypi.org/forums/viewtopic.php?t=255563

**Expected behaviour**
To correctly boot the raspberry and display the screen

**Actual behaviour**
The screen is blank

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Raspinfo output: https://pastebin.com/SbJ04KSJ

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

```
Rasberry Pi 4B
```

* Which OS and version (`cat /etc/rpi-issue`)?

```
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4
```

* Which firmware version (`vcgencmd version`)?

```
Feb 12 2020 12:36:21
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
```

**Logs**

Full dmesg log can be seen at https://pastebin.com/SbJ04KSJ (bottom). At the bottom I noticed the following errors:

```
[   17.768338] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[   18.847585] fuse init (API version 7.27)
[   28.008350] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[   38.248345] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:51:HDMI-A-1] flip_done timed out
[   48.488359] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[   58.728339] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[   68.968341] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[   79.208223] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:51:HDMI-A-1] flip_done timed out
[   89.448138] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[   99.688110] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[  109.928094] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[  120.168095] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:51:HDMI-A-1] flip_done timed out
[  130.408079] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[  140.648352] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[  152.808766] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[  163.049012] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:51:HDMI-A-1] flip_done timed out
[  173.289231] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:49:crtc-0] flip_done timed out
[... 367 lines cut ...]
[   391.342] (II) modeset(0): Modeline "1920x1080"x0.0  148.50  1920 2448 2492 2640  1080 1084 1089 1125 +hsync +vsync (56.2 kHz e)
[   391.342] (II) modeset(0): Modeline "1280x720"x0.0   74.25  1280 1720 1760 1980  720 725 730 750 +hsync +vsync (37.5 kHz e)
[   391.342] (II) modeset(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[   391.342] (II) modeset(0): Modeline "720x576"x0.0   27.00  720 732 796 864  576 581 586 625 -hsync -vsync (31.2 kHz e)
[   391.342] (II) modeset(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[   391.342] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[   391.342] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 664 704 832  480 489 492 520 -hsync -vsync (37.9 kHz e)
[   391.342] (II) modeset(0): Modeline "640x480"x0.0   30.24  640 704 768 864  480 483 486 525 -hsync -vsync (35.0 kHz e)
[   391.342] (II) modeset(0): Modeline "720x400"x0.0   28.32  720 738 846 900  400 412 414 449 -hsync +vsync (31.5 kHz e)
[   391.342] (II) modeset(0): Modeline "1280x1024"x0.0  135.00  1280 1296 1440 1688  1024 1025 1028 1066 +hsync +vsync (80.0 kHz e)
[   391.342] (II) modeset(0): Modeline "1024x768"x0.0   78.75  1024 1040 1136 1312  768 769 772 800 +hsync +vsync (60.0 kHz e)
[   391.342] (II) modeset(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz e)
[   391.342] (II) modeset(0): Modeline "800x600"x0.0   49.50  800 816 896 1056  600 601 604 625 +hsync +vsync (46.9 kHz e)
[   391.342] (II) modeset(0): Modeline "1280x1024"x0.0  108.00  1280 1328 1440 1688  1024 1025 1028 1066 +hsync +vsync (64.0 kHz e)
[   391.342] (II) modeset(0): Modeline "1280x960"x0.0  108.00  1280 1376 1488 1800  960 961 964 1000 +hsync +vsync (60.0 kHz e)
[   391.342] (II) modeset(0): Modeline "1440x900"x0.0  136.75  1440 1536 1688 1936  900 903 909 942 -hsync +vsync (70.6 kHz e)
[   391.342] (II) modeset(0): Modeline "1440x900"x0.0   88.75  1440 1488 1520 1600  900 903 909 926 +hsync -vsync (55.5 kHz e)
[   391.342] (II) modeset(0): Modeline "1680x1050"x0.0  119.00  1680 1728 1760 1840  1050 1053 1059 1080 +hsync -vsync (64.7 kHz e)
[   391.343] (--) modeset(0): HDMI max TMDS frequency 600000KHz
[   391.363] (II) modeset(0): Disabling kernel dirty updates, not required.
```

**Additional context**

None

6by9, 2020-05-21:
Your timings of 
```
[   141.976] (II) modeset(0): Modeline "3840x2160"x60.0  533.25  3840 3888 3920 4000  2160 2163 2168 2222 +hsync -vsync (133.3 kHz eP)
```
are not the standard timings for 4k60.
```
[   141.976] (II) modeset(0): Modeline "3840x2160"x60.0  594.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (135.0 kHz e)
```
are the standard ones.
The firmware will only select a standard timing unless you provide full custom timings, whereas vc4-fkms-v3d reparses the EDID and will allow selection of any mode listed therein.

You also have a weird timing for 4k30
```
[   141.977] (II) modeset(0): Modeline "3840x2160"x30.0  262.75  3840 3888 3920 4000  2160 2163 2168 2191 +hsync -vsync (65.7 kHz e)
```
the normal timing would be 
```
[   141.976] (II) modeset(0): Modeline "3840x2160"x30.0  297.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (67.5 kHz e)
```

Seeing as the mode still comes out as 3840x2160@60 I can't think of an easy way to make it select the alternative timing - most modes are selected by width, height, and refresh rate only.

Both of the odd timings come from the detailed timings portion of the EDID, so the monitor is claiming that these modes are supported there. The full decode is
```
EDID version: 1.3
Manufacturer: AOC Model 3277 Serial Number 431
Made in week 52 of 2017
Digital display
Maximum image size: 70 cm x 39 cm
Gamma: 2.20
DPMS levels: Off
RGB color display
First detailed timing is preferred timing
Display x,y Chromaticity:
  Red:   0.6845, 0.3095
  Green: 0.2597, 0.6845
  Blue:  0.1523, 0.0546
  White: 0.3134, 0.3291
Established timings supported:
  720x400@70Hz 9:5 HorFreq: 31469 Hz Clock: 28.320 MHz
  640x480@60Hz 4:3 HorFreq: 31469 Hz Clock: 25.175 MHz
  640x480@67Hz 4:3 HorFreq: 35000 Hz Clock: 30.240 MHz
  640x480@72Hz 4:3 HorFreq: 37900 Hz Clock: 31.500 MHz
  640x480@75Hz 4:3 HorFreq: 37500 Hz Clock: 31.500 MHz
  800x600@60Hz 4:3 HorFreq: 37900 Hz Clock: 40.000 MHz
  800x600@75Hz 4:3 HorFreq: 46900 Hz Clock: 49.500 MHz
  1024x768@60Hz 4:3 HorFreq: 48400 Hz Clock: 65.000 MHz
  1024x768@75Hz 4:3 HorFreq: 60000 Hz Clock: 78.750 MHz
  1280x1024@75Hz 5:4 HorFreq: 80000 Hz Clock: 135.000 MHz
Standard timings supported:
  1920x1080@60Hz 16:9
  1280x1024@60Hz 5:4 HorFreq: 64000 Hz Clock: 108.000 MHz
  1280x960@60Hz 4:3 HorFreq: 60000 Hz Clock: 108.000 MHz
  1440x900@75Hz 16:10 HorFreq: 82300 Hz Clock: 156.000 MHz
  1440x900@60Hz 16:10 HorFreq: 55500 Hz Clock: 88.750 MHz
  1680x1050@60Hz 16:10 HorFreq: 64700 Hz Clock: 119.000 MHz
  1280x720@60Hz 16:9
Detailed mode: Clock 533.250 MHz, 698 mm x 393 mm
               3840 3888 3920 4000 hborder 0
               2160 2163 2168 2222 vborder 0
               +hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 133312 Hz
Detailed mode: Clock 262.750 MHz, 698 mm x 393 mm
               3840 3888 3920 4000 hborder 0
               2160 2163 2168 2191 vborder 0
               +hsync -vsync 
               VertFreq: 29 Hz, HorFreq: 65687 Hz
Monitor name: U3277WB
Monitor ranges (GTF): 23-80Hz V, 30-160kHz H, max dotclock 600MHz
Has 1 extension blocks
[... 46 lines cut ...]
Supports YCbCr 4:4:4
Supports YCbCr 4:2:2
1 native detailed modes
Detailed mode: Clock 74.250 MHz, 698 mm x 393 mm
               1280 1390 1430 1650 hborder 0
                720  725  730  750 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 45000 Hz
Detailed mode: Clock 27.000 MHz, 698 mm x 393 mm
                720  736  798  858 hborder 0
                480  489  495  525 vborder 0
               -hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 31468 Hz
Detailed mode: Clock 277.250 MHz, 698 mm x 393 mm
               1920 1968 2000 2080 hborder 0
               2160 2163 2173 2222 vborder 0
               +hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 133293 Hz
Detailed mode: Clock 297.000 MHz, 698 mm x 393 mm
               3840 4016 4104 4400 hborder 0
               2160 2168 2178 2250 vborder 0
               +hsync -vsync 
               VertFreq: 30 Hz, HorFreq: 67500 Hz
Checksum: 0x9e (valid)
```

MartenBE, 2020-05-21:
I'm afraid this is a little bit above my understanding. I've found the following in the manual:

[image]

Is there anything I can do? Is it possible to spoof the EDID?

strategist333, 2020-10-23:
I have this exact same problem on Raspberry Pi 4B on a different monitor: BenQ PD3200U. By exact same problem, I mean I get a blank screen when enabling 4k60, which is resolved by removing `dtoverlay=vc4-fkms-v3d`. Interestingly, my EDID also shows the exact same nonstandard timings (3840 3888 3920 4000). I've never had an issue with getting 4k60 from any other sources on this monitor. Could it be that there is some nontrivial subset of the market with these nonstandard timings? Any recommendation for getting 4k60 to work without having to disable 3D accelerator?

Here is my EDID and decode
```
AP///////wAJ0SWARVQAAAIcAQOARih4LofRqFVNnyUOUFSla4CBgIHAgQCpwLMA0cABAQEBTdAA
oPBwPoAwIDUAxI8hAAAaAAAA/wBBMUowMDgxNjAxOQogAAAA/QAyTB6MPAAKICAgICAgAAAA/ABC
ZW5RIFBEMzIwMFUKAUoCAzPxUWFgX15dEB8iISAFFAQTEgMBIwkHB4MBAABsAwwAIAA4PCAAQAEC
Z9hdxAF4iAMEdAAw8nBagLBYigDEjyEAAB5WXgCgoKApUC8gNQDEjyEAABoAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGA==
```

```Extracted contents:
header:          00 ff ff ff ff ff ff 00
serial number:   09 d1 25 80 45 54 00 00 02 1c
version:         01 03
basic params:    80 46 28 78 2e
chroma info:     87 d1 a8 55 4d 9f 25 0e 50 54
established:     a5 6b 80
standard:        81 80 81 c0 81 00 a9 c0 b3 00 d1 c0 01 01 01 01
descriptor 1:    4d d0 00 a0 f0 70 3e 80 30 20 35 00 c4 8f 21 00 00 1a
descriptor 2:    00 00 00 ff 00 41 31 4a 30 30 38 31 36 30 31 39 0a 20
descriptor 3:    00 00 00 fd 00 32 4c 1e 8c 3c 00 0a 20 20 20 20 20 20
descriptor 4:    00 00 00 fc 00 42 65 6e 51 20 50 44 33 32 30 30 55 0a
extensions:      01
checksum:        4a

Manufacturer: BNQ Model 8025 Serial Number 21573
Made week 2 of 2018
EDID version: 1.3
Digital display
Maximum image size: 70 cm x 40 cm
Gamma: 2.20
DPMS levels: Off
Supported color formats: RGB 4:4:4, YCrCb 4:4:4
Default (sRGB) color space is primary color space
First detailed timing is preferred timing
Established timings supported:
  720x400@70Hz
  640x480@60Hz
  640x480@75Hz
  800x600@60Hz
  800x600@75Hz
  832x624@75Hz
  1024x768@60Hz
  1024x768@75Hz
  1280x1024@75Hz
  1152x870@75Hz
Standard timings supported:
  1280x1024@60Hz
  1280x720@60Hz
  1280x800@60Hz
  1600x900@60Hz
  1680x1050@60Hz
  1920x1080@60Hz
Detailed mode: Clock 533.250 MHz, 708 mm x 399 mm
               3840 3888 3920 4000 hborder 0
               2160 2163 2168 2222 vborder 0
               +hsync -vsync 
Serial number: A1J00816019
Monitor ranges (GTF): 50-76Hz V, 30-140kHz H, max dotclock 600MHz
Monitor name: BenQ
Has 1 extension blocks
Checksum: 0x4a (valid)

CEA extension block
Extension version: 3
47 bytes of CEA data
  Video data block
    VIC  97 3840x2160@60Hz 
    VIC  96 3840x2160@50Hz 
    VIC  95 3840x2160@30Hz 
    VIC  94 3840x2160@25Hz 
    VIC  93 3840x2160@24Hz 
    VIC  16 1920x1080@60Hz 
    VIC  31 1920x1080@50Hz 
    VIC  34 1920x1080@30Hz 
    VIC  33 1920x1080@25Hz 
    VIC  32 1920x1080@24Hz 
    VIC   5 1920x1080i@60Hz 
    VIC  20 1920x1080i@50Hz 
    VIC   4 1280x720@60Hz 
    VIC  19 1280x720@50Hz 
    VIC  18 720x576@50Hz 
    VIC   3 720x480@60Hz 
    VIC   1 640x480@60Hz 
  Audio data block
    Linear PCM, max channels 2
    Supported sample rates (kHz): 48 44.1 32
    Supported sample sizes (bits): 24 20 16
  Speaker allocation data block
    Speaker map: FL/FR
  Vendor-specific data block, OUI 000c03 (HDMI)
    Source physical address 2.0.0.0
    DC_36bit
    DC_30bit
    DC_Y444
    Maximum TMDS clock: 300MHz
    Extended HDMI video details:
      HDMI VIC 0 3840x2160@30Hz
      HDMI VIC 1 3840x2160@25Hz
  Vendor-specific data block, OUI c45dd8
Underscans PC formats by default
Basic audio support
Supports YCbCr 4:4:4
Supports YCbCr 4:2:2
1 native detailed modes
Detailed mode: Clock 297.000 MHz, 708 mm x 399 mm
               3840 4016 4104 4400 hborder 0
               2160 2168 2178 2250 vborder 0
               +hsync +vsync 
Detailed mode: Clock 241.500 MHz, 708 mm x 399 mm
               2560 2607 2639 2720 hborder 0
               1440 1443 1448 1481 vborder 0
               +hsync -vsync 
Checksum: 0x18 (valid)

EDID block does NOT conform to EDID 1.3!
	Detailed block string not properly terminated
```

vatbrain, 2020-12-13:
Experiencing what I believe to be the same problem with a Lenovo L28u-30 4k monitor.

nanoant, 2021-04-12:
Same problem here with LG 27UK650-W (27", 4K UHD) monitor, that works perfectly fine with all my other computers including Mac mini, Windows Dell Laptop and Lenovo PC running both Windows and Linux, but unfortunately does not work with my brand new Raspberry Pi 400 when enabling 4kp60 via `hdmi_enable_4kp60=1` in `config.txt`.

Commenting out `dtoverlay=vc4-fkms-v3d` makes 4kp60 work but is *NOT* a solution since we loose GPU acceleration.

Here's my EDID:
~~~
pi@raspberrypi:~ $ tvservice -d edid.dat
Written 256 bytes to edid.dat
pi@raspberrypi:~ $ base64 edid.dat 
AP///////wAebQZ3NmoHAAQcAQOAPCJ46j4xrlBHrCcMUFQhCABxQIGAgcCpwNHAgQABAQEBCOgA
MPJwWoCwWIoAWFQhAAAeBHQAMPJwWoCwWIoAWFQhAAAaAAAA/QAwPR6HPAAKICAgICAgAAAA/ABM
RyBIRFIgNEsKICAgAT8CA0FxTZAiIB8SAwQBYWBdXl8jCQcHbQMMABAAuDwgAGABAgNn2F3EAXiA
A+MPAANoGgAAAQEwPQDjBcAA4wYFAQI6gBhxOC1AWCxFAFhUIQAAHlZeAKCgoClQMCA1AFhUIQAA
GgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5g==
~~~

My timings for `"3840x2160"x60.0` in `Xorg.0.log` are OK according to what @6by9 wrote above. Still I can use only 30Hz mode for 4K.
~~~
[    10.278] (II) modeset(0): Modeline "3840x2160"x60.0  594.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (135.0 kHz eP)
[    10.278] (II) modeset(0): Modeline "3840x2160"x50.0  594.00  3840 4896 4984 5280  2160 2168 2178 2250 +hsync +vsync (112.5 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x59.9  593.41  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (134.9 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x30.0  297.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync -vsync (67.5 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x30.0  297.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (67.5 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x25.0  297.00  3840 4896 4984 5280  2160 2168 2178 2250 +hsync +vsync (56.2 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x24.0  297.00  3840 5116 5204 5500  2160 2168 2178 2250 +hsync +vsync (54.0 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x30.0  296.70  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (67.4 kHz e)
[    10.278] (II) modeset(0): Modeline "3840x2160"x24.0  296.70  3840 5116 5204 5500  2160 2168 2178 2250 +hsync +vsync (53.9 kHz e)
~~~

This is really a bummer since Pi4 and Pi400 are advertised as 4K capable, but 4kp60 does not work as advertised with such popular 4k monitor as mine.

rlkennedyreid, 2021-04-13:
Same here with an LG 27UL850 and a Pi 4 8GB model. Was trying various settings, like HDMI mode force, with HDMI Group 1, Mode 97, and obviously 4kp60 enabled. With 4kp60 I'd always get a blank screen on one HDMI port, and a 1080p screen on the other port. Disabling DT overlay sorted the problem immediately. Attached is my edid output for the monitor.
[edid.txt](https://github.com/raspberrypi/firmware/files/6304604/edid.txt)

nanoant, 2021-04-13:
> With 4kp60 I'd always get a blank screen on one HDMI port, and a 1080p screen on the other port. Disabling DT overlay sorted the problem immediately.

Same here. But I wouldn't call disabling DT overlay a solution since it disables 3D acceleration, which cripples the graphics performance. We need a real solution.

FYI this problem was described also in two threads on the forums:
https://www.raspberrypi.org/forums/viewtopic.php?t=255563 (listed in the original message of this issue)
https://www.raspberrypi.org/forums/viewtopic.php?t=259333 (another one)

@itoral @anholt Humbly asking for your help or if you know someone who can help with this apparently v3d related issue.

itoral, 2021-04-14:
If this is an FKMS issue it is not something I can help with, but I'll try to ping someone at Rpi to look into it. By the way, does the issue persist using the real KMS driver?

nanoant, 2021-04-14:
When using KMS `vc4-kms-v3d-pi4` with:
~~~
dtoverlay=vc4-kms-v3d-pi4
hdmi_enable_4kp60=1
hdmi_group=1
hdmi_mode=95
~~~
I cannot even see UHD 60Hz modes in `xrandr`
~~~
Screen 0: minimum 320 x 200, current 3840 x 2160, maximum 7680 x 7680
HDMI-1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 600mm x 340mm
   3840x2160     30.00*   30.00    25.00    24.00    29.97    23.98  
   2560x1440     59.95  
   1920x1080     60.00    50.00    59.94    30.00    24.00    29.97    23.98  
   1600x900      60.00  
   1280x1024     60.02  
   1280x800      59.91  
   1280x720      60.00    59.94  
   1024x768      60.00  
   800x600       60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       60.00    59.94  
HDMI-2 disconnected (normal left inverted right x axis y axis)
~~~

With FKMS `vc4-fkms-v3d` I can see UHD 60Hz modes:
~~~
Screen 0: minimum 320 x 200, current 3840 x 2160, maximum 7680 x 7680
HDMI-1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 600mm x 340mm
   3840x2160     60.00 +  50.00    59.94    30.00*   30.00    25.00    24.00    29.97    23.98  
   2560x1440     59.95  
   1920x1080     60.00    50.00    59.94    30.00    24.00    29.97    23.98  
   1600x900      60.00  
   1280x1024     60.02  
   1280x800      59.91  
   1280x720      60.00    59.94  
   1024x768      60.00  
   800x600       60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       60.00    59.94 
~~~
But if I try to switch to `3840x2160@60.00` I get blank screen.

@popcornmix Sorry to ping you here, but you maybe have an idea who could help with this issue?

6by9, 2021-04-14:
Patches for 4k60 in vc4-kms-v3d are still to be merged - they're on dri-devel at present and should be merged soon.
HDMI modes above 4k30 require HDMI2.0 with SCDC, scrambling support, and a load of other bells and whistles.

Forcing your EDID via config.txt
```
hdmi_edid_file:0=1
hdmi_edid_filename:0=fw1392edid.bin
hdmi_enable_4kp60=1
```
has come up fine on my Dell 4k monitor, and ```edid-decode /sys/class/drm/card0-HDMI-A-1/edid``` confirms it as ```Display Product Name: 'LG HDR 4K'```
```
pi@raspberrypi:~ $ DISPLAY=:0 xrandr
Screen 0: minimum 320 x 200, current 3840 x 2160, maximum 7680 x 7680
HDMI-1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 600mm x 340mm
   3840x2160     60.00*+  50.00    59.94    30.00    30.00    25.00    24.00    29.97    23.98  
   2560x1440     59.95  
   1920x1080     60.00    50.00    59.94    30.00    24.00    29.97    23.98  
   1600x900      60.00  
   1280x1024     60.02  
   1280x800      59.91  
   1280x720      60.00    59.94  
   1024x768      60.00  
   800x600       60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       60.00    59.94  
pi@raspberrypi:~ $ 
```

That it works without vc4-fkms-v3d is odd. The only real difference is that in loading vc4-fkms-v3d is that it will force a mode change to the DRM selected timings rather than the ones chosen by the firmware.
Do you get similar issues if you force mode switches using tvservice when vc4-fkms-v3d is not loaded? It's possible it's a corner case on the clock setup.

nanoant, 2021-04-14:
> That it works without vc4-fkms-v3d is odd.

This is exactly what this thread and my and others reports are about. If we disable `vc4-fkms-v3d` overlay then we get 4k@60Hz after boot, but without GPU/3D acceleration of course. So something in Pi4/400 can drive 4k@60Hz properly in our monitors, which brings some hope it can be resolved. 🤞 

> Do you get similar issues if you force mode switches using tvservice when vc4-fkms-v3d is not loaded?

Yes, when I disable `vc4-fkms-v3d` and do `tvservice -p` (preferred) or `tvservice -e 'CEA 97 HDMI'` then I lose HDMI signal according to my monitor. If I do `tvservice -e 'CEA 95 HDMI'` (which is 4k@30Hz) then I have a signal (of course I need to restart X display manager afterwards).

Here's `tvservice --modes=CEA` output on my Pi400 if this may help. But probably it is exactly what was in EDID I've sent before anyway.
~~~
% tvservice --modes=CEA                 
Group CEA has 16 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive 
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive 
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
  (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive 
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive 
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive 
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive 
           mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive 
           mode 34: 1920x1080 @ 30Hz 16:9, clock:74MHz progressive 
           mode 93: 3840x2160 @ 24Hz 16:9, clock:297MHz progressive 
           mode 94: 3840x2160 @ 25Hz 16:9, clock:297MHz progressive 
           mode 95: 3840x2160 @ 30Hz 16:9, clock:297MHz progressive 
           mode 96: 3840x2160 @ 50Hz 16:9, clock:594MHz progressive 
  (prefer) mode 97: 3840x2160 @ 60Hz 16:9, clock:594MHz progressive
~~~

Also, comparing timings reported by X when `vc4-fkms-v3d` is enabled to EIA/CEA-861 standard resolutions and timings I don't see anything wrong.
~~~
[    10.278] (II) modeset(0): Modeline "3840x2160"x60.0  594.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (135.0 kHz eP)
~~~
https://en.wikipedia.org/wiki/Extended_Display_Identification_Data#Structure,_version_1.4
97 | 2160p60 | 16∶9 | 1∶1 | 594 | 60 | 135.0 | 3840 | 2160 | 4400 | 2250 | 60
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

6by9, 2021-04-14:
> This is exactly what this thread and my and others reports are about. 

Actually in this case, no. You've hijacked a thread where someone had a totally non-standard 4k60 mode advertised by their monitor. 

> Yes, when I disable vc4-fkms-v3d and do tvservice -p (preferred) or tvservice -e 'CEA 97 HDMI' then I lose HDMI signal according to my monitor. If I do tvservice -e 'CEA 95 HDMI' (which is 4k@30Hz) then I have a signal (of course I need to restart X display manager afterwards).

It sounds like something in the setup isn't re-enabling scrambling correctly. HDMI2.0 requires this for any mode with a pixel clock above 340MHz, and it also alters the HDMI TMDS clock rate as well.
The source (Pi) will tell the sink (display) over the SCDC link the scrambling state when the link comes up, and the sink should adopt that.

What firmware and kernel versions are you running? Part of the HDMI handling was reworked recently. After March 10th 2021 should have that.

nanoant, 2021-04-14:
> You've hijacked a thread where someone had a totally non-standard 4k60 mode advertised by their monitor.

I am sorry to hear that. If I did it it was unintentional. Still there were other reports at https://www.raspberrypi.org/forums/viewtopic.php?t=255563 linked to this issue, and there was also @vatbrain reporting same problem before I did, and then @rlkennedyreid after my post. So this place looked like most relevant place to post to, especially when you read the subject.

> What firmware and kernel versions are you running?

~~~
% vcgencmd version
Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

% uname -a
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
~~~

> Part of the HDMI handling was reworked recently. After March 10th 2021 should have that.

I presume I'd need try pull `boot/` files from this repo, right? It isn't something that is already released at https://www.raspberrypi.org/software/operating-systems/?

6by9, 2021-04-14:
Yes, rpi-update pulls in the absolute latest firmware and kernels.
https://www.raspberrypi.org/documentation/raspbian/applications/rpi-update.md

Any investigations will always be on the absolute latest firmware & kernel.

nanoant, 2021-04-14:
> Any investigations will always be on the absolute latest firmware & kernel.

Understood. I updated to the latest firmware & kernel and still no luck with 4k@60Hz with `dtoverlay=vc4-fkms-v3d`. And it still works ok without `vc4-fkms-v3d`. Is there anything else I can try?

~~~
% vcgencmd version
Apr 12 2021 15:52:31 
Copyright (c) 2012 Broadcom
version 0227a0840ee8c94294cf3b2c4b2d42031a58f05a (clean) (release) (start)

% uname -a
Linux raspberrypi 5.10.27-v7l+ #1409 SMP Tue Apr 6 18:26:02 BST 2021 armv7l GNU/Linux
~~~

6by9, 2021-04-14:
Without vc4-fkms-v3d, disconnect and fully power cycle the monitor (ie pull the mains cable), and then reconnect to the Pi. Does it recover?

rlkennedyreid, 2021-04-14:
> Same here. But I wouldn't call disabling DT overlay a solution

Agreed, @nanoant. 'Solved' was bad wording here - I have just decided to use 30Hz until there's a fix. Disabling 3D acceleration is certainly no fix - absolutely cripples almost anything you want to do, especially in-browser.

nanoant, 2021-04-14:
> Without vc4-fkms-v3d, disconnect and fully power cycle the monitor (ie pull the mains cable), and then reconnect to the Pi. Does it recover?

Yes it does. I pulled HDMI cable out of Pi400, turned off the monitor, pulled the power cable, put the cable back, powered the monitor, plugged the HDMI cable back and I get the Pi400 screen.

nanoant, 2021-04-14:
> Agreed, @nanoant. 'Solved' was bad wording here - I have just decided to use 30Hz until there's a fix. Disabling 3D acceleration is certainly no fix - absolutely cripples almost anything you want to do, especially in-browser.

@rlkennedyreid Yeah, indeed. If you read https://www.raspberrypi.org/products/raspberry-pi-4-model-b/ you can get excited that pi4 can drive two 4k displays, but then if you read details you see it can drive only one 4k display at 60Hz with `hdmi_enable_4kp60=1`, which is a necessity if you do desktop work. Still with this knowledge I decided to give it a try and buy Pi 400 for tinkering, which is Pi4 with higher clock embedded into a keyboard. But it turns out that in my case I cannot even get it running 4k at 60Hz here.

strategist333, 2021-04-15:
I've been following this thread for a while, hoping that it would see some activity. Given how quickly multiple people chimed in, I suspect there may be a few others in a similar boat.

In case it helps any, here is another edid (from a BenQ PD3200U monitor), for which I see the exact same symptoms (boots either on 4k30 with vc4-fkms-v3d, or on 4k60 disabling vc4-fkms-v3d). To add a bit more detail, when trying to boot on 4k60 with vc4-fkms-v3d, I see the terminal-style initial boot screen (the one with the Raspberry Pi logo), followed by the color wheel screen, and then the HDMI signal cuts out.

```
pi@raspberrypi:~ $ base64 edid.dat 
AP///////wAJ0SWARVQAAAIcAQOARih4LofRqFVNnyUOUFSla4CBgIHAgQCpwLMA0cABAQEBTdAA
oPBwPoAwIDUAxI8hAAAaAAAA/wBBMUowMDgxNjAxOQogAAAA/QAyTB6MPAAKICAgICAgAAAA/ABC
ZW5RIFBEMzIwMFUKAUoCAzPxUWFgX15dEB8iISAFFAQTEgMBIwkHB4MBAABsAwwAEAA4PCAAQAEC
Z9hdxAF4iAMEdAAw8nBagLBYigDEjyEAAB5WXgCgoKApUC8gNQDEjyEAABoAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKA==
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.27-v7l+ #1409 SMP Tue Apr 6 18:26:02 BST 2021 armv7l GNU/Linux
pi@raspberrypi:~ $ /opt/vc/bin/vcgencmd version
Apr 12 2021 15:52:31 
Copyright (c) 2012 Broadcom
version 0227a0840ee8c94294cf3b2c4b2d42031a58f05a (clean) (release) (start)
```

alanbork, 2021-04-15:
folks, it would be wise to be precise about which video resolution you are talking about , as people often talk about 3840x2160-60 and 4096x2160-60 as 4k60. I've found the latest fkms to work with  3840x2160-60 pretty well, but to crash badly with 4096x2160-60 (https://github.com/raspberrypi/linux/issues/3842). not trying to hijack the thread, just pointing out the ambiguity.

strategist333, 2021-04-15:
Yes, good point. I was referring to 3840x2160-60 which does not work on Pi 4B for me on my BenQ PD3200U monitor. (The monitor works fine on this setting with other computers.)

nanoant, 2021-04-15:
Yes, my problem is also about `3840x2160 @ 60Hz` aka 4K UHD which according to https://en.wikipedia.org/wiki/4K_resolution is dominant 4K standard and AFAIK most of TVs and screens are 4K UHD. 

4096x2160 aka 4K DCI on other hand is not that popular and related to cinema industry.

Therefore I'd say ops or @MartenBE could add "4K **UHD**@60Hz" in the subject just to clarify that.

alanbork, 2021-04-15:
why not use the actual resolution? seems a lot clearer than an arbitrary TLA.

MartenBE, 2021-04-15:
I have updated the title. I have sold my Pi shortly after the bug report, so I cannot assist any further anymore.

nanoant, 2021-04-15:
> I have sold my Pi shortly after the bug report, so I cannot assist any further anymore.

@MartenBE, out of curiosity, was the 4K support the reason you've sold your Pi?

> It sounds like something in the setup isn't re-enabling scrambling correctly. HDMI2.0 requires this for any mode with a pixel clock above 340MHz, and it also alters the HDMI TMDS clock rate as well.
The source (Pi) will tell the sink (display) over the SCDC link the scrambling state when the link comes up, and the sink should adopt that.

@6by9, since you mention SCDC which seems to be DDC extension, I also tested if DDC works correctly with `ddcutil` when FKMS is enabled, and it works well on Pi400 with my monitor, I can read and set things like brightness and contrast.

Is this SCDC communication done by the firmware or linux driver? Is it something that is open-source? Because if it is I am willing to do my best and try to work on the issue.

> Patches for 4k60 in vc4-kms-v3d are still to be merged - they're on dri-devel at present and should be merged soon.

I understand KMS is complete rewrite that does no rely on DisplaymanX. Does it mean HDMI 4kp60 support is also rewritten and independent from the firmware?

> If this is an FKMS issue it is not something I can help with, but I'll try to ping someone at Rpi to look into it. By the way, does the issue persist using the real KMS driver?

@itoral Current KMS build does not include 4kp60, but as @6by9 wrote above it looks there are some patches for that on dri-devel that will be merged soon.

popcornmix, 2021-04-16:
> I understand KMS is complete rewrite that does no rely on DisplaymanX. Does it mean HDMI 4kp60 support is also rewritten and independent from the firmware?

Yes. Control of hdmi (and hvs) is purely from arm code and doesn't make use of the firmware drivers.

nanoant, 2021-04-21:
@popcornmix I have pulled latest kernel build that seems to contain https://github.com/raspberrypi/linux/pull/4302 introducing 4k @ 60Hz support for KMS driver by @mripard
~~~
% cat /boot/.firmware_revision 
449d410d961e51b3691dff8b315c9efa119255f3
~~~
I confirm that now I can see new `60.00 +  50.00    59.94` modes in `xrandr`.

Unfortunately switching to 60Hz produces now "no signal detector" on my monitor, similar to what has been already observed with FKMS. The only case when 4k @ 60Hz works is when no FKMS or KMS are enabled.

What shall we do now?

alanbork, 2021-04-21:
have you tried overclocking the pi, as in the fkms bug report?

nanoant, 2021-04-21:
> have you tried overclocking the pi, as in the fkms bug report?

If you mean `core_freq=600` and `core_freq_min=600` from https://github.com/raspberrypi/linux/issues/4271 then yes, I just tried that and no change, still no signal here on `3840x2160 @ 60.00`.

Anyway I had impression that `hdmi_enable_4kp60=1` is already raising the GPU clock, so no explicit `core_freq` should be needed once this is on.

I am really trying to give Pi400 a chance, but I am so far really disappointed. Not to mention horrible tearing when dragging windows or web content while running `3840x2160 @ 60.00`, which is another big issue.

6by9, 2021-04-21:
Please have some patience. 4k60 support has just been merged but there are likely to be some rough edges still, and it is being worked on.

Setting core_freq/core_freq_min to 600 is required for 4096x2160. It shouldn't be required for 3840x2160.

You've just said you can't get 3840x2160@60 working on full KMS. FKMS has known issues with vsync events. Wait for the rough edges to be sorted with full KMS - it shouldn't be long.

alanbork, 2021-04-21:
> You've just said you can't get ***@***.*** working on full KMS. FKMS has

what sort of bug reports will be useful for the new kms release?

alanbork, 2021-04-21:
quick update: 3840x and 4096x are working with kms. vsync (as measured by consistency of input lag) looks right.  Will do more testing but this is looking like a great step forward. 

tvservice needs an update to at least warn that it's not supported under kms as the error message ([E] No device present) isn't very helpful.

nanoant, 2021-04-21:
> Please have some patience. 4k60 support has just been merged but there are likely to be some rough edges still, and it is being worked on.

What makes me bit worried is that Rpi4 was released June 2019, so almost 2 years ago and it was advertised as 4K capable including decoding H.265 at 4K and 60Hz. `hdmi_enable_4kp60=1` was available from the beginning, so one could assume that 2 years later it is something that has been already extensively tested and should be working well.

I understand that KMS is completely new driver and 4K @ 60Hz support (i.e. scrambling) has been just added/merged. But still I would like to know how this issue here is supposed to be resolved? Are the Rpi developers working on KMS or/and FKMS aware of the problem? Do they have or need an access to the monitors that have the problem?

alanbork, 2021-04-21:
switching back to normal cpu clock, I'm not having as much success, even with 3840x, with it working for the first 10 or so seconds and then failing with

[  598.238501] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  608.478821] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out

too soon to be sure if this is just an intermittent issue or actually tied to the clock.

alanbork, 2021-04-21:
^ postscript: it's definitely tied to the cpu clock. with core_freq/core_freq_min  600, 3840x works reliably (entered and exited video mode 10 times in a row with no problems), without overclock, it fails about half the time, either immediately, or after  a few seconds.  Does this need a new issue opened?

same test with 4096x works fine (entering and exiting 10x with core_freq/core_freq_min  600).

popcornmix, 2021-04-23:
@alanbork this issue is about the fkms driver.
if you can provide something concrete that fails reliably (e.g. a script using modetest) with kms driver then please open a new issue. This issue has got muddled enough with a mixture of fkms/kms issues and 3840/4096, that it's hard to keep track of new information.

nanoant, 2021-05-03:
@popcornmix If we focus only on FKMS and 3840x2160 @ 60Hz, is there anything else we can do to move this forward?

I think most important information is that 60Hz works when neither FKMS or KMS are enabled. Can you please explain in such case which driver is driving HDMI at 60Hz? Because if it works there I think it should be possible to make FKMS and KMS work with our monitors.

popcornmix, 2021-05-04:
Both fkms and "no kms" use the same firmware side code for setting hdmi mode.
But with fkms the kernel side chooses which resolution/framerate to use which may be different.
Does `tvservice -s` report the same in the two cases?

nanoant, 2021-05-04:
@popcornmix Thanks for suggesting checking `tvservice -s`. It seems I made some good progress.

With no overlay (no FKMS or KMS) I get:
~~~
% tvservice -s
state 0xa [HDMI CEA (97) RGB lim 16:9], 3840x2160 @ 60.00Hz, progressive
~~~

With FKMS overlay enabled my X starts in 30 Hz (I set it once in Screen Configuration, now it seems to persist) and I get:
~~~
% xrandr           
Screen 0: minimum 320 x 200, current 3840 x 2160, maximum 7680 x 7680
HDMI-1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 600mm x 340mm
   3840x2160     60.00 +  50.00    59.94    30.00*   30.00    25.00    24.00    29.97    23.98

% tvservice -s
state 0xa [HDMI CUSTOM RGB full 16:9], 3840x2160 @ 30.00Hz, progressive
~~~
What is suspicious that the mode is `CUSTOM` not `CEA`, but it works well at 30Hz with my monitor.

When I change to 60Hz with xrandr (or Screen Configuration) and get blank screen (no HDMI signal according to my monitor):
~~~
% xrandr --output HDMI-1 --mode 3840x2160 --rate 60

% tvservice -s                                     
state 0xa [HDMI CUSTOM RGB lim 16:9], 3840x2160 @ 60.00Hz, progressive
~~~
`tvservice -s` reports `CUSTOM` instead of `CEA (97)`.

And there goes interesting part, if I use `tvservice -e 'CEA 97 HDMI'` to force CEA mode I get 3840x2160 @ 60.00Hz working absolutely fine with my monitor:
~~~
% tvservice -e 'CEA 97 HDMI'

% tvservice -s              
state 0xa [HDMI CEA (97) RGB lim 16:9], 3840x2160 @ 60.00Hz, progressive
~~~

Any ideas why FKMS driver is selecting some `CUSTOM` options instead of CEA?

Btw. here's output of `tvservice -m`:
~~~
% tvservice -m CEA    
Group CEA has 16 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive 
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive 
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
  (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive 
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive 
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive 
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive 
           mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive 
           mode 34: 1920x1080 @ 30Hz 16:9, clock:74MHz progressive 
           mode 93: 3840x2160 @ 24Hz 16:9, clock:297MHz progressive 
           mode 94: 3840x2160 @ 25Hz 16:9, clock:297MHz progressive 
           mode 95: 3840x2160 @ 30Hz 16:9, clock:297MHz progressive 
           mode 96: 3840x2160 @ 50Hz 16:9, clock:594MHz progressive 
  (prefer) mode 97: 3840x2160 @ 60Hz 16:9, clock:594MHz progressive
~~~

6by9, 2021-05-04:
The firmware will always choose one of the standard CEA or DMT modes. It does look at the detailed timings in the EDID, but only to see which standard mode most closely matches. tvservice will therefore always report one of the standard modes.
(The exception being using ```hdmi_group=2``` and ```hdmi_mode=87``` with ```hdmi_cvt=``` or ```hdmi_timings=``` to specify a custom timing to the exception of anything else).

KMS (whether vc4-kms-v3d or vc4-fkms-v3d) parses the full EDID and creates a full mode definition for each and every mode described therein. It passes that full timing to the driver to select a mode, therefore there has to be different handling within the firmware to be able to accept that, and that is why tvservice shows CUSTOM - it's using the FKMS specific entry.
(If the mode happens to match one of the CEA HDMI modes, then that information is also passed down so that it can be signalled in the HDMI AVI Infoframe)

So the real question is why KMS is seemingly coming up with a mode definition that is apparently incompatible with your monitor.  

Looking again at the [EDID you posted earlier](https://github.com/raspberrypi/firmware/issues/1392#issuecomment-818232370), the decode for the detailed timings includes
```
Detailed mode: Clock 533.250 MHz, 698 mm x 393 mm
               3840 3888 3920 4000 hborder 0
               2160 2163 2168 2222 vborder 0
               +hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 133312 Hz
```
That differs from  the [CEA standard mode 97](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/drm_edid.c#L1225).
```3840, 4016, 4104, 4400, 0, 2160, 2168, 2178, 2250, 0, clock 594MHz```
(And I disagree with edid-decode's frame rate. 533250000 / (4000 * 2222) = 59.997Hz VertFreq according to my calculator) 

KMS will almost certainly go for the former. The firmware will go for the latter.
```xrandr --verbose``` will list the modes that have been parsed.

I'll try to remember in the morning to hook that EDID up with our HDMI analyser to confirm that the Pi is putting out the mode described, but if so it largely means that your monitor's EDID is incorrect.

6by9, 2021-05-05:
And this is why hijacking issues is a bad idea - I'd actually picked up MartenBE's EDID with the non-standard 4k60 mode which is where this thread started. Last night's post updated to remove the erroneous information.

Tested with your EDID on the analyser on a CM4. Syncs fine, and the analyser reports VIC 97 from the AVI frame, so DRM is ending up with the same mode.

What you're doing with your tvservice and xrandr commands is just reselecting effectively the same mode.
It'd be interesting to know if from an SSH shell doing ```DISPLAY=:0 xrandr --output HDMI-1 --off``` and then ```DISPLAY=:0 xrandr --output HDMI-1 --preferred``` wakes it up happily.

nanoant, 2021-05-05:
> It'd be interesting to know if from an SSH shell doing DISPLAY=:0 xrandr --output HDMI-1 --off and then DISPLAY=:0 xrandr --output HDMI-1 --preferred wakes it up happily.

Yes it does. `--off` and `--preferred` sequence wakes up happily.

~~~
% DISPLAY=:0 xrandr --output HDMI-1 --mode 3840x2160 --rate 60  # --> no signal
% DISPLAY=:0 xrandr --output HDMI-1 --off                       # --> monitor is off
% DISPLAY=:0 xrandr --output HDMI-1 --preferred                 # --> wakes fine in 3840x2160 @ 60 Hz
% tvservice -s              
state 0xa [HDMI CUSTOM RGB lim 16:9], 3840x2160 @ 60.00Hz, progressive
~~~

6by9, 2021-05-05:
OK, so something on your particular system seems to not like the mode switching. The timing on that is involved, but is robust on most systems. I can't answer at the moment as to whether your system is an edge case, or if something else is involved.

nanoant, 2021-05-06:
It seems the problem lies in timing according to EDID provided by monitor, because CEA 97 seems to be robust for me and others reporting here and on the forums, since 60 Hz works for us when FKMS/KMS are not enabled.

An option forcing CEA timings for both FKMS (and KMS for future) would be good workaround for all of the users here I believe. Or am I demanding too much?

Also I don't understand how my system can be an edge case if my LG 27UK650-W is extremely popular model from well known brand producing computer screens. Moreover others are having same problem with other brands such as AOC, Lenovo or Benq. So this is not an isolated case.

alanbork, 2021-05-06:
I've asked in the past for a way to add CEA modes (some or even the entire set) to the list offered by DRM/KMS via config.txt, but that's been shot down.

however there's nothing preventing you from generating a modeline that exactly matches cea 97 and putting that in your xorg file or whatever.

popcornmix, 2021-05-07:
> I've asked in the past for a way to add CEA modes (some or even the entire set) to the list offered by DRM/KMS via config.txt, but that's been shot down.

it's just not an option architecturally. config.txt is parsed by firmware and not seen by kernel. The kms driver controls hdmi hardware directly without firmware support.

Custom modelines should be added using standard linux mechanisms (e.g. xorg.conf or provide a custom edid).

6by9, 2021-05-07:
> > I'll try to remember in the morning to hook that EDID up with our HDMI analyser to confirm that the Pi is putting out the mode described, but if so it largely means that your monitor's EDID is incorrect.
> > Edited to remove erroneous information
> 
> @6by9 I just did `edid-decode` on my EDID and I cannot reproduce the information you posted. Was it a reason of the removal of erroneous information? What I see in my EDID is:
> 
> ```
> Detailed mode: Clock 594.000 MHz, 600 mm x 340 mm
>                3840 4016 4104 4400 hborder 0
>                2160 2168 2178 2250 vborder 0
>                +hsync +vsync 
>                VertFreq: 60 Hz, HorFreq: 135000 Hz
> ```
> 
> Which is exactly CEA 97 mode.

As I'd responded:
> And this is why hijacking issues is a bad idea - I'd actually picked up MartenBE's EDID with the non-standard 4k60 mode which is where this thread started. Last night's post updated to remove the erroneous information.

Your EDID does indeed have describe CEA mode 97.

> > What you're doing with your tvservice and xrandr commands is just reselecting effectively the same mode.
> 
> Okay, so it seems that `tvservice -s` will always show `CUSTOM` when it was programmed by the linux driver, even if current mode settings are exactly the same as `CEA (97)`. Am I right?

Correct, and that is the expected behaviour.

> So since I am in both cases setting exactly the same mode, proposed solution to "force" CEA would have absolutely no effect in my case. Also using manual `Modeline` would not help either.
> 
> In the end it looks like some timing problem, where firmware setting the mode gets it right, but fKMS not. Not to mention KMS does not either with my monitor.

We've just had a discovery on one monitor that it isn't registering the request to enable scrambling and the high tmds clock ratio which is required for HDMI pixel clocks over 340MHz. Miss that and the monitor won't be able to decode it. It's down to how tolerant the monitor of the HDMI sequencing.
I'm creating a patch that will read back the status, and retry it if necessary.