raspberrypi/firmware #1393: Raspberry pi 4 model B will not boot after rpi-update

PhysicsX, 2020-05-21:
**Describe the bug**
After rpi-update raspberry pi 4 model B is not boot. It waits in black screen. 

**To reproduce**
sudo apt update
sudo apt upgrade
reboot 
sudo rpi-update
-> no boot

**Expected behaviour**
After rpi-update boot should be done properly.

**Actual behaviour**
After rpi-update raspberry pi can not boot properly.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry pi 4 model B
* Which OS and version (`cat /etc/rpi-issue`)?
2020-02-13-raspbian-buster.img
* Which firmware version (`vcgencmd version`)?

* Which kernel version (`uname -a`)?

**Logs**
There is no log

**Additional context**
Before a one year ago, I install libGLESv2.so in /opt/vc/lib using rpi-update. I need to install them to the raspberry pi 4 model B. I can install with rpi-update to version 4 but after reboot I could not run the device. Device is not boot.

popcornmix, 2020-05-21:
Any custom options in config.txt or cmdline.txt?

Do you have a spare sdcard?
Can you install raspbian and check your pi still boots.
If it does, can you run rpi-update and check if that boots.

I'd like to determine if the updated firmware just doesn't work on your Pi in a clean setup, or if you have some custom changes that don't work with the update.

PhysicsX, 2020-05-21:
> Any custom options in config.txt or cmdline.txt?

There is no custom options.
>What do LEDs do when you power on?

When problem occur, after reboot I saw onlye colorful screen, and green led is blinking a few times then it stops. 

>Does it boot if connected to HDMI display rather than touch screen.
I do not have hdmi screen right that is why I can not check it sorry.

>Do you have a spare sdcard?
>Can you install raspbian and check your pi still boots.
>If it does, can you run rpi-update and check if that boots.
I do not have spare sd card
After problem, I install again raspian to that sd card and it boots with fresh os.

>I'd like to determine if the updated firmware just doesn't work on your Pi in a clean setup, or if you have some custom changes that don't work with the update.

Thank you, I understand you concerns, but before some time ago I could do that very much. Right now it is not working.

popcornmix, 2020-05-21:
Is this the display you are using? https://www.raspberrypi.org/products/raspberry-pi-touch-display/

PhysicsX, 2020-05-21:
> Is this the display you are using? https://www.raspberrypi.org/products/raspberry-pi-touch-display/

Exactly.

popcornmix, 2020-05-21:
@6by9 have you booted rpi-update kernel/firmware on Pi4 with LCD recently?

popcornmix, 2020-05-21:
Just tried it - latest rpi-update firmware/kernel has booted up for me with console showing on LCD.

PhysicsX, 2020-05-21:
Did you run the fresh image ? 

Which sd card you are using ?

There are issues like this ->https://www.raspberrypi.org/forums/viewtopic.php?t=264750

popcornmix, 2020-05-21:
I'm using an up to date rasbpian (apt full-upgrade earlier today) followed by rpi-update.
The thread you linked to was from a few months ago and didn't seem to be an issue that affected many.

You said after a fresh install the card booted again. Are the log messages from this fresh install?

PhysicsX, 2020-05-21:
> I'm using an up to date rasbpian (apt full-upgrade earlier today) followed by rpi-update.
> The thread you linked to was from a few months ago and didn't seem to be an issue that affected many.
> 
> You said after a fresh install the card booted again. Are the log messages from this fresh install?

This logs from reboot after rpi-update. With fresh image I can run it with my sd cards whatever size or brand is.

popcornmix, 2020-05-21:
Without being able to reproduce or seeing the log messages (e.g. from uart or a hdmi display where more information is visible) it's going to be hard to proceed.

PhysicsX, 2020-05-21:
I will remove the screen and try over ssh whole scenario. Also I will print uart output after that if issue will occur.

Thank you.

6by9, 2020-05-21:
> have you booted rpi-update kernel/firmware on Pi4 with LCD recently?

Just done so. Works fine.

> When problem occur, after reboot I saw onlye colorful screen, and green led is blinking a few times then it stops.

That would imply an issue with the contents of the SD card. start.elf has initialised the display, but booting Linux is failing for some reason.
How many flashes?
https://www.raspberrypi.org/documentation/configuration/led_blink_warnings.md

PhysicsX, 2020-05-21:
If both of you can not see the issue, there can be another thing. I saw only one time green led is blinking but as I see this is not related with "partition not fat". I will make double check everything. 

Thank you.

closed 2020-05-21 by PhysicsX

PhysicsX, 2020-05-21:
When I remove the touch screen I can connect to rasp from ssh. It means device is booting. But when I connect run with touch screen device is not boot after rpi-update. Touch screen is supplied by another power. There is no issue about power.

What can be here ?

PhysicsX, 2020-05-21:
I think I found the issue. It is related with contact FFC  cable. But I do not have no idea why this is happening after rpi-update. 
There should be some frquency change after rpi-update for touch screen. I remove and plug the cable again and now everthing is working. Interesting.