raspberrypi/firmware #1395: initramfs image is loaded into RAM with extra (DWORD) padding

kirgene, 2020-05-25:
initramfs image is loaded into ram with additional (DWORD) padding

**To reproduce**
create initramfs image (`initrd.gz`) which size is not multiple of 4 and boot system with the following line in config:

**OR**

create a file which will be used as an initramfs image:
dd if=/dev/urandom of=initrd.gz bs=9957521 count=1
```
initramfs initrd.gz
```

Add the following line to [initramfs.c:660](https://github.com/raspberrypi/linux/blob/79495a5ecdfba69de51e88701a69c42d09806d84/init/initramfs.c#L660) to debug this error:

```
printk(KERN_INFO "start = %x %x, end = %x %x %x %x\n", ((unsigned char *)initrd_start)[0], ((unsigned char *)initrd_start)[1], ((unsigned char *)(initrd_end - 4))[0], ((unsigned char *)(initrd_end - 3))[0], ((unsigned char *)(initrd_end - 2))[0], ((unsigned char *)(initrd_end - 1))[0]);
```

**Expected behaviour**
system boots with initramfs image

**Actual behaviour**
kernel panic with the following messages:
```
Trying to unpack rootfs image as initramfs...
rootfs image is not initramfs (Decoding failed); looks like an initrd
Freeing initrd memory: 19452K
```

**System**
Raspberry Pi 1 Model B Rev. 2
Firmware rev: a6c9b6b48ce86ef2527586a50760d52f1b33f642
Kernel: 5.4.42

pelwell, 2020-06-01:
Alignment can't be the issue here, but it's possible the decompression code is checking the length. This is the first report of a problem with the file length padding so your choice of compression might be unusual, but it is clearly safer to preserve the lengths.

Are you using the default loader (start.elf) or one of the other variants? I can get you a test build of a patched version to confirm it works.

pelwell, 2020-06-01:
You can download a complete firmware build here: https://drive.google.com/file/d/1GQFhQPt0uNGi2iYvnc20VpjjtG9Ugo65/view?usp=sharing
Feel free to only install the relevant version.

pelwell, 2020-06-11:
The current rpi-update firmware release includes this fix.

kirgene, 2020-06-14:
It works, thanks @pelwell

closed 2020-06-14 by kirgene