raspberrypi/firmware #1398: "omxplayer --orientation 90" is flickering after recent "sudo apt upgrade"

jamesee, 2020-05-28:
**Describe the bug**
I have been using omxplayer for my project for quite some time. 

I noticed that the video started to flicker after a recent "sudo apt upgrade" (4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux)  when I execute the following command:-

$ omxplayer --orientation 90 video1.mp4

There is no flickering without the --orientation flag:-
$ omxplayer video1.mp4

When I roll back to my previous image backup (4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux) and execute the same command above, the video does not flicker.

I am quite sure that the flickering occurred due to the recent "sudo apt upgrade".

**Expected behaviour** / **Actual behaviour**
please see attached video.zip on actual behaviour and expected behaviour
[video.zip](https://github.com/raspberrypi/firmware/files/4692539/video.zip)

**Reproduce the Actual behaviour**
I took a backup image (4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux) and "sudo apt update && sudo apt upgrade". Execute the following command:-

$ omxplayer --orientation 90 video1.mp4

**System**
 
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry4B-4G model
 
* Which OS and version (`cat /etc/rpi-issue`)?
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

* Which firmware version (`vcgencmd version`)?
$ vcgencmd version
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

* Which kernel version (`uname -a`)?
$ uname -a
Linux jjAIA-PLB 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

popcornmix, 2020-05-28:
Can you try using rpi-update firmware? I think there have been fixes in this area recently.
(backup first is recommended before using rpi-update)

jamesee, 2020-05-28:
@popcornmix 

I tried using "sudo rpi-update" but had the following error messages:

The boot partition has only 44MB, but rpi-update for raspi4B requires 256MB. I tried using gparted to extend the boot partition, but raspi could not boot up.

I was advised by 6by9 to download the firmware from https://github.com/Hexxeh/rpi-firmware and installing the files manually, but if I copied everything, the boot partition still out of space.

Can you guide me on which files I should copy? I believe I do not need all of the files.

James
=======================

$ sudo rpi-update
*** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
*** Performing self-update
*** Relaunching after update
*** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
Partition size 43M may not be sufficient for new Pi4 files
This could result in a system that will not boot.
256M FAT partition is recommended. Ensure you have a backup if continuing.
Would you like to proceed? (y/N)
#############################################################
WARNING: This update bumps to rpi-5.4.y linux tree
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769

'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.

DO NOT use 'rpi-update' as part of a regular update process.

##############################################################
Would you like to proceed? (y/N)
*** Downloading specific firmware revision (this will take a few minutes)
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 168 100 168 0 0 5419 0 --:--:-- --:--:-- --:--:-- 5419
100 115M 100 115M 0 0 7481k 0 0:00:15 0:00:15 --:--:-- 5399k
*** Updating firmware
cp: error writing '/boot/kernel7.img': No space left on device
cp: error writing '/boot/kernel7l.img': No space left on device
cp: error writing '/boot/kernel8.img': No space left on device
cp: error writing '/boot/kernel.img': No space left on device

popcornmix, 2020-05-28:
If your partition size is 44MB you are not using a Buster image (possibly an wheezy/stretch image that has been upgraded, but that's not generally recommended for reasons like this). 

For Pi4 you just need to update start4.elf/fixup4.dat, assuming you are using the default start file (i.e. no start_file= or start_x=1 or gpu_mem=16).

jamesee, 2020-05-28:
but i built the image up from buster lite...
thanks for the advise on start4.elf/fixup4.dat. I am using default start file.

closed 2020-05-28 by jamesee