raspberrypi/firmware #1402: Pi 4 change from 4GB to 8GB model breaks usb-xhci on Plan 9 operating system

9pi, 2020-05-30:
I'm trying to boot the Plan 9 OS on a new 8GB Raspberry Pi 4.  It's not able to initialise usb xhci. The controller responds but it doesn't enumerate the usual two onboard hubs (all four ports of the root hub have status=0x2A0, ie no 'present' bit).  The same kernel (and same firmware, config.txt etc) works fine on 2GB and 4GB Pi 4 boards.

I tried booting an old Raspbian kernel (4.19.75-v7l) which works on the other two Pi 4 boards, and that seems to have a related problem: the first thing I see on the hdmi screen is `xhci_hcd: can't setup: -110`, followed by a kernel null pointer dereference.

After upgrading to current Raspberry Pi OS kernel (4.19.118-v7l+), the 8GB board boots successfully and USB keyboard works.

I surmise that something in the USB3 setup has changed between 4GB and 8GB Pi 4 models, or in the boot EEPROM, and that the linux kernel has been changed to accommodate this.  I hope someone can tell me what the change is, so I can do the same for Plan 9.  Thanks!

ptesarik, 2020-05-31:
This has most likely nothing to do with the available RAM, rather with the fact that the SPI flash with firmware for the VIA VL805 chip is missing on this revision of the board. In turn, the firmware must be provided by the firmware (or OS) whenever it's missing, e.g. after each PCIe bus reset.

ptesarik, 2020-05-31:
See also https://lkml.org/lkml/2020/5/5/998.

9pi, 2020-05-31:
Thanks, that's very enlightening!  Is there any documentation for this, or is it a case of looking to see how linux does it?

ptesarik, 2020-05-31:
I'm sorry, I don't have more details about the firmware API, but @vianpl may know a bit more.

JamesH65, 2020-05-31:
@timg236 Anything we can say about this yet?

9pi, 2020-05-31:
> is it a case of looking to see how linux does it?

Did that, and it's just a one-word message via the mailbox to remind the GPU to load the VL805 firmware. I sort of wonder why the GPU couldn't just do that at power-up without being asked, but that's OK, it's easy enough to do.

8GB Raspberry Pi 4 is now working perfectly on Plan 9. Thank you, @ptesarik , and issue closed.

closed 2020-05-31 by 9pi

timg236, 2020-06-01:
The firmware resets PCI before starting the host OS because without SDRAM the VL805 makes inbound requests to page-in firmware from SDRAM. If the OS has a different PCI mapping that would all break. There's no way to pause and move the mappings and it's not really desirable for the firmware to dictate the PCI configuration. Consequently, the mailbox is required so that the OS can reset the VLI after asserting PCI chip reset.

9pi, 2020-06-01:
> The firmware resets PCI before starting the host OS because without SDRAM the VL805 makes inbound requests to page-in firmware from SDRAM.

Let me make sure I understand this. The vl805 doesn't have its own dedicated memory to execute firmware from (or not enough memory to hold all of it), so after initialisation it will continue to fetch blocks of instructions from the 8GB RAM which is shared by the arm cores and gpu. So, even if the firmware was loaded at power up time (as I guess it must be, to allow booting from USB drives), it has to be loaded again (or pointers to it have to be relocated?) after the OS has set up its own arrangement of PCI bus addressing. Yes?

timg236, 2020-06-01:
Yes, the host OS is free to change where SDRAM appears for inbound accesses so if that's different to the address chosen by the bootloader it will fail. I think disabling bus mastering whilst it's running (e.g. OS PCI enumeration) might potentially be harmful but I don't know the internals of the FW. So it's best to start with a clean setup chosen by the OS.

9pi, 2020-06-01:
Thanks, that's clear.