raspberrypi/firmware #1403: rcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted

beta-tester, 2020-06-01:
**Describe the bug**
hello, i have:

- RPi4B (4GB)
- official Raspberry Pi power supply (no undervoltage messages in syslog)
- Raspbian (Raspberry Pi OS) updated & upgraded
- proper cooling (no above 60°C).

on that RPi i run an access point (hostapd + dnsmasq) on its onboard wifi interface wlan0.
to that access point (wlan0) i connected wifi sensor devices, those are sending its data to my RPi.
so far, so good...
it was working more than 6 month 24h per day ,  7 days a week - only few reboots after bigger updates,

but recently in some situations the wifi interface wlan0 stops working
and only a reboot of the RPi will bring back the wifi adapter back to life.

i think the main issue is the folowing message:
`brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted`

before it breaks, one of the most distant wifi sensor device  with the worst signal was reporting to see the access point with only -85dB...
```
# a log i mage to see the signal strength how strong the sensor can see the access point
2020-06-01T18:35:50.0;-84dB
2020-06-01T18:36:20.0;-85dB
2020-06-01T18:36:50.0;-80dB
2020-06-01T18:37:35.1;-81dB
2020-06-01T18:37:50.0;-85dB
2020-06-01T18:38:50.1;-84dB
```

and on the RPi4 in the /var/log/syslog there is the folowing to see...
```
Jun  1 18:42:29 pxe-server hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Jun  1 18:42:29 pxe-server hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
...
Jun  1 18:44:02 pxe-server kernel: [68600.849888] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
Jun  1 18:44:10 pxe-server kernel: [68609.019622] brcmfmac: brcmf_escan_timeout: timer expired
Jun  1 18:44:13 pxe-server kernel: [68611.579693] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jun  1 18:44:13 pxe-server kernel: [68611.580127] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Jun  1 18:44:13 pxe-server kernel: [68611.580141] brcmfmac: brcmf_notify_escan_complete: Scan abort failed
Jun  1 18:45:03 pxe-server kernel: [68661.430542] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jun  1 18:45:03 pxe-server kernel: [68661.430992] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Jun  1 18:45:03 pxe-server kernel: [68661.431010] brcmfmac: brcmf_run_escan: error (-110)
Jun  1 18:45:03 pxe-server kernel: [68661.431025] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
Jun  1 18:46:03 pxe-server kernel: [68721.421553] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jun  1 18:46:03 pxe-server kernel: [68721.422013] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Jun  1 18:46:03 pxe-server kernel: [68721.422032] brcmfmac: brcmf_run_escan: error (-110)
Jun  1 18:46:03 pxe-server kernel: [68721.422047] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
Jun  1 18:46:51 pxe-server kernel: [68769.822379] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jun  1 18:46:51 pxe-server kernel: [68769.822741] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Jun  1 18:46:54 pxe-server kernel: [68772.382417] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jun  1 18:46:54 pxe-server kernel: [68772.382774] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Jun  1 18:46:54 pxe-server kernel: [68772.382793] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
...
```

this is already the third extra long weekend (because of holiday), that i can observe such an behavoir...

- it starts with a waek signal strength on one wifi client.
the client reports -85dB rx signal strength for the wifi access point.

- then there is a bunch of messages related to the client with the weak signal:
`hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated`
`hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated`

- but the communication completely breaks as soon the folowing message appears in the log
`brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted`

- from this point, the log file will be filled with endless messages of:
```
brcmfmac: brcmf_escan_timeout: timer expired
brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
brcmfmac: brcmf_notify_escan_complete: Scan abort failed
brcmfmac: brcmf_run_escan: error (-110)
brcmfmac: brcmf_cfg80211_scan: scan error (-110)
```
- after that the wifi adapter is not recoverable anymore until i reboot the RPi4.

- a regular normal re-boot in this situation takes for ever, because the stopoing the jobs of wlan0 / hostapd / dhcpsd  run into timeouts at least for 3 minutes...

does anybody know, what is going on here?

that is happens mostely on extra long weekends with holidays.
i guess is because there are more wlan activity in the neighborhood speacially at those times.

but why seems the firmware to break in those situation so badly that only a reboot can bring the wifi adapter back to life?

beta-tester, 2020-06-01:
```
$ sudo rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
BOOTLOADER: up-to-date
CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
```

pelwell, 2020-06-02:
Since you seem to be able to get failures fairly regularly, can you try enabling some kernel debug output?
Add `brcmfmac.debug=0x100000` to /boot/cmdline.txt, keeping it in a single, long line, then reboot. Running `dmesg | grep brcmfmac` should result in output like this:
```
[    7.650239] brcmfmac: CONSOLE: d 0
[    7.650256] brcmfmac: CONSOLE: 000000.063 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.202 (r724630 CY)
[    7.650270] brcmfmac: CONSOLE: 000000.064 TCAM: 256 used: 252 exceed:0
[    7.650284] brcmfmac: CONSOLE: 000000.065 reclaim section 1: Returned 122844 bytes to the heap
[    7.650297] brcmfmac: CONSOLE: 000000.065 reclaim section 4: Returned 44 bytes to the heap
[    7.650310] brcmfmac: CONSOLE: 000000.065 sdpcmd_dpc: Enable
...
```
Then just carry on as normal. When the brcmfmac firmware dies, capture the output of `dmesg` to a file and attach it (or a link to pastebin, etc.) here.

Since the failure triggers other kernel messages, there is a danger that the useful output is lost before you have chance to capture it. A way to avoid this is to leave a shell constantly saving the kernel messages to a file:
```
$ dmesg -w > kernel_log.txt &
```

beta-tester, 2020-06-03:
here is a syslog (with enabled debug information) of that issue from boot to shutdown/reboot...
hope you can see something usefully in there...
i kept also dnsmasq logs in there to see possible relationships...
xx:xx:xx:xx:xx:xx with the name sensor_tx_2 is the wifi sensor with the weak rx.
zz:zz:zz:zz:zz:zz is the onboard wlan0

at 11:34:30 the firmware is halted
[syslog_log-1.txt.zip](https://github.com/raspberrypi/firmware/files/4723042/syslog_log-1.txt.zip)

PS.: i can not force the issue by slowly covering the sensor with a metal case.
the result is then simply no signal and will recover as soon the signal strength is good again.

pelwell, 2020-06-03:
Thanks - I've forwarded your log and background information to Cypress.

pelwell, 2020-06-04:
Cypress have responded very quickly with a new firmware to try - there's no indication yet what might have changed.
1. Download the new firmware from here: https://drive.google.com/file/d/1vz30u4UFWWA8Uh5ivo78ppYkUK3rh8fC/view?usp=sharing
2. Make a backup - `sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.bin{,.orig}`
3. Install the new version - `sudo cp brcmfmac43455-sdio.bin /lib/firmware/brcm/`
4. Reboot.

If you do get another crash, please attach the kernel log as before.

beta-tester, 2020-06-04:
wow, that is really quick!
thank you very much, i will try the new firmware and report if i still have issues.

but maybe it will take quite some time because i haven't figure out how to force the issue manually... so the only thing i can do is waiting...

pelwell, 2020-06-04:
Of course - when you are waiting for a failure, no news is good news.

beta-tester, 2020-06-08:
intermediat report:
til now no issues. the wlan is working now as it should.

BTW: the firmware reports:
`BCM4345/6 wl0: May 20 2020 00:36:51 version manifest (de496ab CY) FWID 01-12aa6092`
the date 2020-05-20 is already some days back in the past.
does it mean my RPi missed to pull this firmware update automatically (apt upgrade)
or is that firmware not released yet and will come later?

pelwell, 2020-06-08:
Thanks for the update - I'll pass on the message. Cypress didn't state the history of that firmware or what might be different, but it is more recent than anything we have been given to release - you didn't miss it.

pintify, 2020-06-16:
I'm currently experiencing this issue when I connect and eighth device to the WiFi AP. I'm on a Raspberry Pi 4 using hostapd to manage the AP.

This solution did not solve the issue:
> Cypress have responded very quickly with a new firmware to try - there's no indication yet what might have changed.
>     1. Download the new firmware from here: [...]

I attach the [kernel log](https://github.com/raspberrypi/firmware/files/4785418/kernel.log), but in a glance:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1311 SMP Mon Apr 27 14:26:42 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
...
[    0.218661] brcm-pcie fd500000.pcie: could not get clock
[    0.218736] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.218787] brcm-pcie fd500000.pcie:   MEM 0x600000000..0x603ffffff -> 0xf8000000
[    0.277680] brcm-pcie fd500000.pcie: link up, 5.0 Gbps x1 (!SSC)
[    0.277945] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
...
[    3.531376] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.540269] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.540607] usbcore: registered new interface driver brcmfmac
[    3.779925] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.793594] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    4.761048] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    4.761065] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    4.932415] bcmgenet: Skipping UMAC reset
[    5.028318] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
...
[   36.197929] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<<--- The first seven devices are connected
[  154.775410] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted <<--- As soon as I connect the eighth
```

Is there any alternative?

pelwell, 2020-06-16:
It's a shame you didn't enable debug output from the firmware as I asked the issue creator to do - there is no useful information in the log.

I think you've probably reached the limits of the device+firmware combination, but I've asked Cypress how many clients should be able to connect.

pintify, 2020-06-16:
That would be very interesting information. I will run again the issue with debug log activated.

JamesH65, 2020-06-16:
Although certainly the firmware should not crash under these circumstances, looks like a buffer overflow or bad error handling somewhere.

pintify, 2020-06-16:
I attach a log with debug log activated. The procedure is the same as in my [previous comment](https://github.com/raspberrypi/firmware/issues/1403#issuecomment-644622250) [kernel.log](https://github.com/raspberrypi/firmware/files/4785713/kernel.log)

pelwell, 2020-06-16:
I see that's with the original firmware. Could you try again with the test firmware?

pintify, 2020-06-16:
Sorry, I changed it back to test a different approach. This is a test applying the change you suggest:

[kernel.log](https://github.com/raspberrypi/firmware/files/4785983/kernel.log)

pelwell, 2020-06-16:
Thanks - I've passed that (and the other log) on.

pelwell, 2020-06-17:
That second log that posted is using the original firmware again - it looks like you've failed to install the updated version.

After a successful installation you should get the following result:
```
pi@raspberrypi:~$ md5sum /lib/firmware/brcm/brcmfmac43455-sdio.bin
0a45a7a828e49e20fee2fe0438149523  /lib/firmware/brcm/brcmfmac43455-sdio.bin
```
If the md5sum doesn't match then you have the wrong file. After booting the kernel log will include:
```
[    7.111945] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May 20 2020 00:36:51 version manifest (de496ab CY) FWID 01-12aa6092
```
Go through the installation instructions again, making sure that the download file is in the current directory before you run the `cp`. You can verify that you have the right file with:
```
pi@raspberrypi:~$ md5sum brcmfmac43455-sdio.bin
0a45a7a828e49e20fee2fe0438149523  /lib/firmware/brcm/brcmfmac43455-sdio.bin
```

stelian42, 2020-06-17:
FYI, I have been experiencing the same kind of issues, quite reproductible after a few hours of operation.

I have updated to the test firmware and enabled the verbose driver logs, will report back if the crash happens again.

pelwell, 2020-06-17:
Thanks - testing and feedback like that are invaluable.

pintify, 2020-06-17:
I'm deeply sorry. Somehow I got confused with [this similar answer](https://github.com/raspberrypi/linux/issues/2453#issuecomment-641168612), with a completely different version. The patch you suggest works better. The eighth device is still unable to connect but at least the driver doesn't crash. Now it reports this when the eighth device tries to connect and then it is rejected:

```
[ 4229.323079] brcmfmac: CONSOLE: 004205.908 wl0: wlc_userscb_alloc: no SCBs available to reclaim
[ 4229.323096] brcmfmac: CONSOLE: 004205.908 wl0: wlc_ap_authresp: out of scbs for 24:62:ab:f2:f6:70
[ 4229.323110] brcmfmac: CONSOLE: 004208.008 wl0: wlc_userscb_alloc: no SCBs available to reclaim
[ 4229.323123] brcmfmac: CONSOLE: 004208.008 wl0: wlc_ap_authresp: out of scbs for 24:62:ab:f2:f6:70
[ 4232.513075] brcmfmac: CONSOLE: 004209.300 wl0: wlc_userscb_alloc: no SCBs available to reclaim
[ 4232.513092] brcmfmac: CONSOLE: 004209.300 wl0: wlc_ap_authresp: out of scbs for 24:62:ab:f2:f6:70
[ 4232.513106] brcmfmac: CONSOLE: 004210.597 wl0: wlc_userscb_alloc: no SCBs available to reclaim
[ 4232.513120] brcmfmac: CONSOLE: 004210.597 wl0: wlc_ap_authresp: out of scbs for 24:62:ab:f2:f6:70
[ 4238.053053] brcmfmac: CONSOLE: 004211.894 wl0: wlc_userscb_alloc: no SCBs available to reclaim
[ 4238.053070] brcmfmac: CONSOLE: 004211.894 wl0: wlc_ap_authresp: out of scbs for 24:62:ab:f2:f6:70
[ 4238.053083] brcmfmac: CONSOLE: 004213.190 wl0: wlc_userscb_alloc: no SCBs available to reclaim
[ 4238.053097] brcmfmac: CONSOLE: 004213.190 wl0: wlc_ap_authresp: out of scbs for 24:62:ab:f2:f6:70
```

Is it caused by the limit of devices you suggested previously?

Thanks in advance

pelwell, 2020-06-17:
I think it has to be because of resource limits, but I've not had confirmation of what the limit is supposed to be. I've seen a datasheet suggesting a maximum of 4 AP clients, so 7 isn't bad considering. Out of curiosity, how many devices does your phone's hotspot support?

pintify, 2020-06-17:
In the past I've reached three devices but never explored the limit. I'll try to perform a test later if possible.

pelwell, 2020-06-17:
Cypress have confirmed the 7-client limit.

pintify, 2020-06-18:
Ok, that's a pity, but good news anyway. Thank you!

pelwell, 2020-06-18:
There's another test firmware - the date is the same as the recent public release but the options list looks different (I have no other information): https://drive.google.com/file/d/10ivocg5PrOwVxAYFKOzJEdv_gdCd-IUF/view?usp=sharing

Be sure to rename to brcmfmac43455-sdio.bin when installing it.

stelian42, 2020-06-18:
I didn't reproduced my hang issues with the updated firmware. Maybe I didn't wait enough.

Anyway, I have updated to the latest firmware and will continue testing:

```
$ md5sum /lib/firmware/brcm/brcmfmac43455-sdio.bin
c4d098fdac7b453408341d5709481b10  /lib/firmware/brcm/brcmfmac43455-sdio.bin

$ dmesg |grep "brcmfmac.*Firmware"
[    5.532513] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea
```

beta-tester, 2020-06-18:
i am wondering, why the new update goes back in time...?
the new one:
c4d098fdac7b453408341d5709481b10
Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea

the prevous new one:
0a45a7a828e49e20fee2fe0438149523 
May 20 2020 00:36:51 version manifest (de496ab CY) FWID 01-12aa6092

the original one:
9d5d7f1953769b5ded866217c9afc437
Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2

beta-tester, 2020-06-18:
i have a question to the 7 clients limit.
i never was thinking about that there a limit may exists...

how long does the "firmware" markes a client as active before it will release the resources of a client?

(in case the client does not show any response - because it is out of range or is in sleep mode or so without actively properly disconnect itself from the access point)

pelwell, 2020-06-18:
This was the suggested firmware to try. I don't care what the date is if it works better.

As for the client count and how long a "lease" lasts, I suggest you experiment.

stelian42, 2020-06-21:
After more than 3 days of uptime, I can confirm that the updated firmware (any one of two) fixes my deconnection issue.

holta, 2020-06-23:
> <<--- The first seven devices are connected
[  154.775410] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted <<--- As soon as I connect the eighth

Is there reason to believe 7-maximum-WiFi-client-devices is a new hard limit?
And if so, where this new restriction is arising from?

Certainly schools around the world using Raspberry Pi 3-and-higher as "learning hotspots" depend on Raspbian (now RaspiOS) firmware allowing 16-to-32 simultaneous WiFi client devices: https://github.com/iiab/iiab/issues/823

(Using grassroots community products like [Internet-in-a-Box](http://internet-in-a-box.org) which crucially depend on this!)

JamesH65, 2020-06-23:
> > <<--- The first seven devices are connected
> > [  154.775410] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted <<--- As soon as I connect the eighth
> 
> Is there reason to believe 7-maximum-WiFi-client-devices is a new hard limit?
> And if so, where this new restriction is arising from?
> 
> Certainly schools around the world using Raspberry Pi 3-and-higher as "learning hotspots" depend on Raspbian (now RaspiOS) firmware allowing 16-to-32 simultaneous WiFi client devices: [iiab/iiab#823](https://github.com/iiab/iiab/issues/823)
> 
> (Using grassroots community products like [Internet-in-a-Box](http://internet-in-a-box.org) which crucially depend on this!)

So are you saying that previous versions of the firmware have allowed more clients?

holta, 2020-06-23:
> So are you saying that previous versions of the firmware have allowed more clients?

Yes.  As documented here: iiab/iiab#823

pelwell, 2020-06-24:
@holta Raspberry Pi have no intention of reducing the number of wireless clients on a Pi AP - we want Pis to be as capable as possible. However, we do want to update the wireless firmware to fix bugs and support newer features such as WPA3 where possible, and it seems that newer releases come with restrictions.
 
Adding features requires larger firmware patches and that may leave less RAM for client support, or it could be that Cypress have reduced the number for other reasons - that is a discussion we need to have - but the change is not at our request, and neither were we informed about it in advance.

Given the limited resources available to the WLAN parts it may be that we end up having to support two firmwares, with some kind of configuration option (perhaps in raspi-config, or perhaps just a file rename) to select between the two. Given that hostapd isn't enabled out of the box, it would seem sensible that a firmware better suited to AP work was available but not selected by default.

holta, 2020-06-24:
@pelwell thanks for the update!

Please keep in touch on this vital question immediately affecting many thousands of children across the developing world.

FYI medical knowledge hotspots (distributed by the Wikipedia community e.g. including 30+ languages for countries like India) also use Raspberry Pi hotspots in villages/clinics/regions having no other option — so let's preserve this up-to-32 WiFi client devices capability that so many around the world are depending on:

https://meta.wikimedia.org/wiki/Internet-in-a-Box

cc: @mitra42 @jvonau @tim-moody @georgejhunt @shanti-bhardwa @avni @WikiDocJames

pelwell, 2020-06-24:
We are not going to (and I'm not even sure we could) invalidate older firmwares, so you always have the option of downloading one from the firmware repo or rewinding the firmware-linux-nonfree package.

holta, 2020-06-24:
> We are not going to (and I'm not even sure we could) invalidate older firmwares, so you always have the option of downloading one from the firmware repo or rewinding the firmware-linux-nonfree package.

Good news.  Thanks for that important reminder.

Please at the same time also communicate this central need to system architects e.g. Cypress?

As nobody (the Raspberry Pi Foundation included) wants to be in a position of distributing buggy older software/firmware to (typically darker skinned, global south) developing countries — who cannot afford a Raspberry Pi for each student — and so very much need to use hostapd as a communal/classroom/clinic "learning hotspots" !

holta, 2020-06-24:
> [Mod edit} removed paragraph that really didn't need to be here.

Are we not allowed to explain the Raspberry Pi design needs of typically darker skinned, Global South, developing (generally extremely poor) countries where not every student can afford a Raspberry Pi?  Was the above censorship an innocent mistake by a Raspberry Pi Foundation employee possibly not experienced in the centrality of narrative (Gherkin stories, cultural contextualization) needed for user-centered design of practical solutions being worked on by deeply disenfranchised communities?

Do Black Lives Matter (yes, #BlackLivesMatter) within the Raspberry Pi Foundation, truly, given the leadership position (James Hughes is Principal Software Engineer) of the above employee, removing the voice of such Raspberry Pi user communities?

Or, where should we have an honest, open and respectful conversation (by voice/phone if that's preferred?) on developing world Raspberry Pi designs to advance RaspiOS "learning hotspots" (and similar Raspberry Pi painful needs of low-income communities) in that position through little-to-no fault of their own &mdash; if that is no longer welcome here?

Either way, we at [Internet-in-a-Box](http://internet-in-a-box.org/) will continue to do scalability/field testing of Raspberry Pi "learning hotspot" and "knowledge hotspot" (i.e. hostapd WiFi hotspot) solutions wherever we can for the [Internet Archive](https://archive.org), for the [Wikipedia community](https://meta.wikimedia.org/wiki/Internet-in-a-Box) (and many others worldwide, in support of all such efforts) as we have done at iiab/iiab#823 and similar places.  Everyone is _greatly appreciative_ of the many employees of all orgs and their volunteers (Raspberry Pi Foundation, Cypress, Broadcom, no doubt others!) for their time working on #1403 here, and many associated infra issues, making all this quite amazing humanitarian progress possible.

Together **yes we can** all truly help millions globally with this effort &mdash; many of whom today living in Nearly Impossible Situations :/

_And as a result, urgently needing up-to-32 simultaneous WiFi client devices wherever possible (typically used by very low-end/older Android phones in humble schools and clinics and community spaces, though we strongly encourage DIY Raspberry Pi client machines too, whenever that can in fact be made to happen!)  Resounding Thanks to Absolutely Everyone Involved in this Raspberry Pi Ecosystem and Community, At Every Level.  Let's keep this a very welcoming place, but make it more inclusive of the rest of the planet as we go forward, with forgiveness during trials and tribulations.  Here's hoping we can All Together now listen to and respect the voices-of-the-unheard going forward as we do so..._

pelwell, 2020-06-24:
You had us at educational.

pelwell, 2020-06-24:
I'm trying to find out (not for the first time) the relative costs (in terms of resources) of the firmware features. With that information we can make informed decisions about the path for future firmware(s).

tim-moody, 2020-06-24:
>I'm trying to find out (not for the first time) the relative costs (in terms of resources) of the firmware features. With that information we can make informed decisions about the path for future firmware(s).

Sounds great. Certainly the new wifi hardware is more capable in terms of frequencies and protocols so it makes sense to use the newest firmware available. But it is a mystery why the number of connections allowed would decrease. @pelwell Perhaps you can verify if this is by design or just due to some limitation. As an example, we have seen that on a quad core i7 NUC the maximum client connections is 12. This seems more a design decision than a hardware limitation. But it is hard to imagine why such a design decision would be made in the case of a Raspberry Pi.

JamesH65, 2020-06-24:
> Sounds great. Certainly the new wifi hardware is more capable in terms of frequencies and protocols so it makes sense to use the newest firmware available. But it is a mystery why the number of [...]

We don't know; the firmware comes from Cypress, we simply ask them to fix bugs, and this is what we get back. It could easily be a memory problem - the newer firmware has more features perhaps, meaning less memory available to handle connections as the memory in the wireless chip is limited (it cannot use main memory AFAIK). Dealing with third party suppliers can be difficult, especially since we are pretty low volume, and the AP case is not a major use case, so not a lot of clout available.

JamesH65, 2020-06-24:
Might be worth explaining a bit more about the memory. These chips have a certain amount of memory on board, as they cannot access the main memory (i.e. any of the 1,2,4,8 GB of RAM on the Pi itself). But you don't want lots of it because it's expensive in die area - more die area = more cost to the end user, and this is a particularly cutthroat area of the market. So you build in as much as you need, but no more. But as features sets increase over time the demands on this memory increase, and since they are built right  on the edge, something has to give. Usually its features (like concurrent connections), hardly ever a re-spin of the chip to add more memory - that's REALLY expensive and given the cost pressures, unlikely to recover dev costs. I should add, this is a generic comment, it possible that something else has caused the connection limit to drop. We won't know until we hear back from Cypress.

mitra42, 2020-06-24:
Thanks for the explanation @JamesH65  - I was wondering about this as well. 
Sorry about being a newbie to the firmware, but is the firmware for the wireless chip separate from that for the rest of the Pi, i..e could this be a firmware update (for just the wireless chip) that was installed as part of the installation process for a box that is going to get used as an wireless AP, while still running current (hopefully less buggy/more feature) versions of the RPI firmware?

tim-moody, 2020-06-24:
To be clear, in our experience the number of connections possible is completely dependent on the driver. If I run ubuntu, which has an old driver, on an rpi4 I get 32 client connections in hostapd. If I run raspbian from Feb on the same rpi4 I get 16 such connections. And I am now told, but haven't tried it, that with the lastest RaspiOs I only get 7.

tim-moody, 2020-06-24:
Is this relevant? https://github.com/seemoo-lab/nexmon

JamesH65, 2020-06-24:
@mitra42 Yes, the wireless chip firmware is separate from the Videocore firmware, and all other firmware's used. 
@tim-moody I suspect that in upgrading the Linux side driver using apt (and maybe rpi-update) you also get the latest firmware. 

So there are two things here, the firmware running on the chip and the Linux side driver - very different bits of software (closed and open respectively), but the Linux side does talk to the firmware via some sort of mailbox interface. It's quite possible the firmware tells the driver how many links it can support during initialisation since that number is most likely dependent on the firmware, not the Linux side driver.

JamesH65, 2020-06-24:
> Is this relevant? https://github.com/seemoo-lab/nexmon

I have no idea what that is for!

tim-moody, 2020-06-24:
>It's quite possible the firmware tells the driver how many links it can support during initialisation since that number is most likely dependent on the firmware, not the Linux side driver.

I'm not sure we are communicating. I write ubuntu to an sd card and Feb raspbian to another sd card. When I put the first card in an rpi 4 I get 32 connections. When I put the raspbian sd card in the same rpi 4 I get 16. I can switch sd cards back and forth at will with the same results. One of our group observed that the linux driver, which we refer to as firmware, was an older release on ubuntu than on raspbian. So if the on chip firmware is telling the driver the number of connections, older drivers seem to get a different number than newer ones from the same (on chip) firmware.

JamesH65, 2020-06-25:
Ah, I see the disconnect,  Note that the firmware in the wireless chip is not in EEPROM, it's run from the internal chip RAM - so it has to be loaded up every time the device is booted. In our case, the Linux driver detects the chip then loads the appropriate firmware from /lib/firmware/bcrm/*

So what firmware you get depends on what is stored in that folder. And that varies between OS's. You could for example, take the firmware from the Ubuntu and try it on the latest PiOS. See if that makes a difference. 

IIRC, it's the brcmfmac43455-sdio.bin you need to change, and the clm blob as well. remember to backup!

pelwell, 2020-06-25:
The clm_blob file is independent of the firmware used. In general it is better to keep your vendor's supplied clm_blob when trying different firmwares.

When dealing with multiple firmwares it can be useful to extract the version, which you can do thus:
```
$ strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv-gtkoe-roamprof-txbf-ve-sae-sr-okc-bpd Version: 7.45.206 (r725000 CY) CRC: bf8ae803 Date: Mon 2020-03-23 02:22:16 PDT Ucode Ver: 1043.2139 FWID 01-febaba43
```

holta, 2020-06-25:
There have been a lot of Linux kernel commits around brcmfmac coming from Cypress in the last while:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/?qt=grep&q=brcmfmac

Are there any that appear particularly relevant e.g. that the Raspberry Pi community should be monitoring and/or advance-testing here?

pelwell, 2020-06-25:
Not that I've seen so far - did you have any particular commits  in mind?

holta, 2020-06-25:
> Not that I've seen so far - did you have any particular commits in mind?

I was just hoping Cypress's brcmfmac kernel patches might shed light / for all.

In any case, thanks much for speaking with Cypress, to try to understand their firmware's "regression" from 32 to ~16 WiFi clients, that seems to have occurred by/around Q2 2018.

As an aside, here's a potential small fix for kernel 4.19 that might help others?

> brcmfmac: abort and release host after error
> https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-4.19.y&id=46eafd0f598787494efb1ea61627ab44105a01eb

jvonau, 2020-06-25:
Think you might want to add https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-4.19.y&id=e12bc9cb438d67ddd64c596a907391d008248f7a if you don't want to make the jump to 4.19.129 or .130 from .118.
Thanks in advance.

JamesH65, 2020-06-25:
> There have been a lot of Linux kernel commits around brcmfmac coming from Cypress in the last while:
> https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/?qt=grep&q=brcmfmac
>
> Are [...]

I monitor all netdev and linux-wireless posts for anything relevant to both wired and wireless ethernet on the Pi, and pass any that might be of interest along. In the recent batch, we do not think there is anything vital , and  I saw nothing about the number of available connections.

dzett, 2020-06-28:
> i am wondering, why the new update goes back in time...?
> the new one:
> c4d098fdac7b453408341d5709481b10
> Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea
>
> the prevous new one: [...]

Just fyi, I stumbled upon this ticket while experiencing problems with Wifi disconnecting after longer periods of time (1 - 5 days) on a headless Raspberry PI 4B (up to date Raspbian installed), so I thought I'd give it a try as this is a quite recent conversation.

I tried out the Mar 23 firmware (01-88ee44ea) and my Wifi would only connect to the AP on the 5GHz band with unusable low signal strength (and also does not fall back to 2.4GHz). 
I have now switched to the May 20 version and it does not show this connectivity issue, with a stable 5GHz connection. 

So the Mar 23 one does not seem to be advisable. 
If you want I can feed back whether my long-term connectivity issue also disappeared now that I am using the May 20 one.

jvonau, 2020-06-28:
FYI June 25th https://community.cypress.com/docs/DOC-20044
Just extracted and renamed .bin and .clm_blob from cypress-firmware-v5.4.18-2020_0625.tar.gz and placed in /lib/firmware/brcm/ 
>pi@box:~ $ dmesg | grep brcmfmac
[    4.238330] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.248776] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.249158] usbcore: registered new interface driver brcmfmac
[    4.491536] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.498227] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May 22 2020 21:24:34 version 7.45.214 (9c83742 CY) FWID 01-59feefd4
[    9.475492] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
pi@box:~ $ strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv-gtkoe-roamprof-txbf-ve-sae-dpp-sr-okc-bpd Version: 7.45.214 (9c83742 CY) CRC: 1baae67b Date: Fri 2020-05-22 21:26:31 PDT Ucode Ver: 1043.2147 FWID 01-59feefd4

note .214 vs .206 or .202, not saying this will correct any known issue, but it works for me with wlan0 used for upstream and ap0 for hostapd.

jvonau, 2020-06-29:
I added debugging with `echo options brcmfmac debug=0x100000 > /etc/modprobe.d/brcmfmac.conf`
Booted with wlan0 connected to upstream AP while hostapd is using ap0 bridged to br0 with 2 clients attached. [dmesg-214](https://github.com/raspberrypi/firmware/files/4847256/dmesg-214.txt) I'll note once setup correctly running on 'lite' I don't suffer from any of the firmware halted errors, but have noted errors when there is a desktop installed or when wlan0 and ap0 are not on the same channel. Hope this helps.

[17 more replies]

closed 2021-11-05 by beta-tester