raspberrypi/firmware #1404: Gamma mailbox message doesn't work correctly on Pi 4

Phlamethrower, 2020-06-01:
**Describe the bug**
The gamma mailbox message that was added for #971 doesn't work correctly on Pi 4. Or at least it doesn't behave the same way that it does on earlier Pi models.

**To reproduce**
Use mailbox property 0x00008012 to set three gamma tables, as per https://github.com/JamesH65/setgamma

**Expected behaviour**
The video output should be adjusted to use the new gamma tables

**Actual behaviour**
Black stays black, while everything else becomes brighter and washed out. The result appears to be unrelated to the content of the supplied gamma tables.

**System**

* Which model of Raspberry Pi?
Pi 4B 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage5
```
* Which firmware version (`vcgencmd version`)?
```
May 11 2020 18:59:41 
Copyright (c) 2012 Broadcom
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

**Additional context**
The discussion in #1274 reveals that the Pi 4 hardware uses a different method for gamma than earlier models. So maybe the mailbox message has been updated to use this new method, and we're just lacking proper documentation on how to use it?

JamesH65, 2020-06-01:
The way gamma is done on the 2711 is unfortunately NOT by tables - it uses a PWL. I've not had time to translate the table approach to a PWL so at the moment this does not work as it did on earlier models. It's possible @timg236 may have added something but I don't think so.

6by9, 2020-06-02:
Duplicate of #1267 
Gamma is not implemented full stop on Pi4. You're probably managing to push your 3 tables into the new structures and getting odd results.

closed 2020-06-02 by pelwell