raspberrypi/firmware #1405: the "enable_gic=0" cannot work on PI4

figozhang, 2020-06-03:
hi guys,
i am writing a Bare Metal app on PI4, and i want to use the legacy interrupt controller, so i add "enable_gic=0" in the config.txt, but it seem the interrupt cannot route to legacy interrupt controller.

here is my config.txt:
[pi4]
kernel=myos4.bin
enable_gic=0

[pi3]
kernel=myos3.bin

[all]
arm_64bit=1  #enable 64bit image
enable_uart=1  #enable pl uart
uart_2ndstage=1   #enable FW debug info

what is the problem?

pelwell, 2020-06-03:
The Device Tree and the firmware configuration have to agree on interrupt usage. If you are using a .dtb file that includes a top-level `interrupt-parent` property referring to a node with a compatible string of `"arm,gic-400"` then it will force `enable_gic` to be 1.

figozhang, 2020-06-03:
if i delete the dtb file in the SD card, it canot boot to my app. 
so any suggestion that if i want to run my bare metal app with legacy interrupt controller?

pelwell, 2020-06-03:
I think you have two options:
1. Create a version of the DTB that uses the legacy interrupt controller. If you aren't going to use Device Tree at all then you could use sed or a hex editor to change "arm,gic-400" to something else the same length, e.g. "not,gic-400".
2. Embrace the GIC - as I understand it, it's not significantly harder to use than the legacy controller, and the performance is going to be better. @dp111 might be able to comment on this.

MartinDupont, 2022-09-13:
I just lost 3 days of work because of this issue. 

When you read the [documentation](https://www.raspberrypi.com/documentation/computers/config_txt.html#enable_gic-raspberry-pi-4-only) for `enable_gic`, it says:

> On the Raspberry Pi 4B, if this value is set to 0 then the interrupts will be routed to the ARM cores using the legacy interrupt controller, rather than via the GIC-400. The default value is 1.

It makes absolutely no mention of the device tree. In fact, it implies that setting `enable_gic=0` is enough to enable the legacy controller. Which is why I spent 3 days messing around with different registers trying to figure out why my IRQ's weren't firing.

At the very least, the documentation needs to be updated.

I had a different workaround to the one proposed by pelwell though. I found the [armstub firmware](https://github.com/raspberrypi/tools/blob/b0c869bc929587a7e1d20a98e2dc828a24ca396a/armstubs/armstub8.S) and simply removed all the lines that deal with the GIC and then compiled it myself. Worked like a charm.