raspberrypi/firmware #1406: vcgencmd get_config doesn't work for ":0", ":1", etc. settings

Phlamethrower, 2020-06-05:
**Describe the bug**
`vcgencmd get_config hdmi_pixel_freq_limit:0` returns the error `hdmi_pixel_freq_limit:0 is unknown`, despite the fact that `vcgencmd get_config int` lists a value for it.
The other settings with ":0" or ":1" suffixes appear to be affected in the same way.
Behaviour seems to be the same regardless of whether a value for the setting is specified in config.txt (and regardless of whether the old-style `hdmi_pixel_freq_limit` or new-style `hdmi_pixel_freq_limit:0` name is used).

**To reproduce**
```
$ vcgencmd get_config hdmi_pixel_freq_limit:0
```

**Expected behaviour**
```
hdmi_pixel_freq_limit:0=0x11ea1a300
```
(or similar), as per `vcgencmd get_config int`

**Actual behaviour**
```
hdmi_pixel_freq_limit:0 is unknown
```

**System**
* Which model of Raspberry Pi?
Pi 4B 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage5
```
* Which firmware version (`vcgencmd version`)?
```
May 11 2020 18:59:41 
Copyright (c) 2012 Broadcom
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

Also seen under RISC OS with May 27th firmware on Pi 4B & 3B

JamesH65, 2020-06-05:
So it does. I'll take a look next week.

bhtooefr, 2020-10-28:
This issue still seems to be present, FWIW.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 1B 256MB
* Which OS and version (`cat /etc/rpi-issue`)?
RISC OS 5.28
* Which firmware version (`vcgencmd version`)?
Both of the following versions of the firmware:
```*vcgencmd version
Jun 10 2020 17:52:12
Copyright (c) 2012 Broadcom
version e46bba1638cca2708b31b9daf4636770ef981735 (clean) (release) (start)```
(version that shipped with the RISC OS 5.28 ROM images)

```*vcgencmd version
Oct 22 2020 14:06:02
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)```
(latest)
* Which kernel version (`uname -a`)?
```*FX 0
RISC OS 5.28 (19 Oct 2020)```