raspberrypi/firmware #1407: A General Question re Raspberry Pi firmware

seamusdemora, 2020-06-09:
This is only a question - not a bug report: 

If I run the command:  

```
sudo /opt/vc/bin/vcgencmd version
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
```
If I examine the [release notes](https://downloads.raspberrypi.org/raspbian/release_notes.txt), I see:
```
2020-02-13:
  * Raspberry Pi Configuration - screen blanking setting disabled if Xscreensaver is installed
  * Bug fix - switch to turn off VNC server in Raspberry Pi Configuration has no effect
  * Bug fix - fix %20 characters in file names
  * Linux kernel 4.19.97
  * Raspberry Pi firmware 9a34efbf2fc6a27231607ce91a7cb6bf3bdbc0c5
    - gencmd: Fix measure_clock name for CLOCK_OUTPUT_108
    - mmal isp: Remote alignment requirements for RGB24 formats
    - Add missing flags for VC_IMAGE_PROP_YUVUV_4K_CHROMA_ALIGN
    - platform: Compromise on gpu overclock settings 
```

The version signatures are different. Is this because the `videocore` firmware (as reported by `vcgencmd` is a subset of the `Raspberry Pi firmware` reported in the release notes? 

If so, how many firmware packages comprise the `Raspberry Pi firmware`?

pelwell, 2020-06-09:
This isn't the right forum to ask questions - it's for reports of bugs and problems - but an answer here might be useful to others.

The hash/signature in the `vcgencmd version` refers to the (unpublished) source tree. In your case it corresponds to a commit titled "platform: Compromise on gpu overclock settings". The commit hash in the release note is from this repo, which holds binaries of compiled firmwares: https://github.com/raspberrypi/firmware/commit/9a34efbf2fc6a27231607ce91a7cb6bf3bdbc0c5

jacobrogers256, 2023-08-10:
Why is the source tree closed? Will it ever be released in the future?

6by9, 2023-08-10:
It contains IP from Broadcom that we are not at liberty to release. There is near-zero chance that will ever change.

closed 2023-08-10 by pelwell