raspberrypi/firmware #1408: SD Card boot failure with ESCPOS printer connected via USB

kolucciy, 2020-06-09:
**Is this the right place for my bug report?**
I believe so.

**Describe the bug**
Pi4B won't boot with ESCPOS printer connected via USB. Printer works fine if connected straight after the boot.

**To reproduce**
Plug the ESCPOS printer to one of the USBs and start Pi4B.

**Expected behaviour**
Should boot normally.

**Actual behaviour**
Hung with black screen. Red light on. No green light activity.

**System**
* Which model of Raspberry Pi? e.g. 
Pi4B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4

* Which firmware version (`vcgencmd version`)?
Jun  1 2020 13:23:40 
Copyright (c) 2012 Broadcom
version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux terminalrpi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

**Logs**
None.

**Additional context**
If printer is connected straight after the boot then all is good.

timg236, 2020-06-09:
Which bootloader version are you using 'vcgencmd bootloader_version' and is there anything on the HDMI screen. 
It might be worth attaching the printer via a powered USB hub.

kolucciy, 2020-06-09:
@timg236 

```
vcgencmd bootloader_version
Apr 16 2020 18:11:26
version a5e1b95f320810c69441557c5f5f0a7f2460dfb8 (release)
timestamp 1587057086
```

I am using 5.1V 3.5A power supply but can certainly try a powered USB hub.

kolucciy, 2020-06-09:
Just to confirm I'm booting of my SD card. Yeah, the failure occur if I run "sudo reboot". It just won't start again until i disconnect the printer.

kolucciy, 2020-06-09:
It feels like its trying to USB boot of the printer.

kolucciy, 2020-06-09:
Found similar issue https://github.com/raspberrypi/firmware/issues/746#issuecomment-516158740
and it looks like I need to use powered hub :( https://github.com/raspberrypi/firmware/issues/746#issuecomment-516365596

pelwell, 2020-06-09:
> Thanks, that bootloader version does support USB boot 

 Does _not_ ...?

timg236, 2020-06-09:
> Found similar issue [#746 (comment)](https://github.com/raspberrypi/firmware/issues/746#issuecomment-516158740)
> and it looks like I need to use powered hub :( [#746 (comment)](https://github.com/raspberrypi/firmware/issues/746#issuecomment-516365596)

That bug was related to USB descriptor parsing, and as @pelwell says that version of the Pi 4 bootloader does **not** use USB. If there's nothing on the HDMI display then that would also suggest a power issue, which might be fixable with a powered hub.

kolucciy, 2020-06-09:
I will order one and try but its far from ideal having a USB HUB to power another USB device during boot considering the fact that it works fine after boot.

pelwell, 2020-06-09:
Boot time tends to be when the power draw is highest, partly because all the devices are being switched on and initialised and partly because systemd is so parallel, so if there were to be a power problem that is when you might expect to see it.

pelwell, 2020-06-09:
But if the green light never even flickers then we can rule out systemd...