raspberrypi/firmware #1409: Raspberry pi 4B (4GB RAM) with HDMI not connected fails to boot using newer release

thedjnK, 2020-06-11:
**Describe the bug**

Using Arch Linux Arm (64-bit) with a 4.19.71-v8+ kernel boots fine when using the firmware package 20200412 (this is commit dd8cbec5a6d27090e5eb080e13d83c35fdd759f7) with nothing connected to the HDMI ports, but when using package 20200528 (commit 62fc8c01165a80021054a430182b504f7b877c2d), the board will only boot up successfully if there is a monitor connected to one of the HDMI ports.

What happens in this state is also odd, the board has an ethernet cable connected and when it boots successfully, the green and orange lights come on and the board is reachable over the network, however when it boots without the monitor connected, the green ethernet light remains off but the orange light comes on and the board is dead to the network.

**To reproduce**

I only have 1 board and 1 SD card so haven't tried reproducing it freshly. How it happens on my board is I have arch linux arm (64-bit) installed, I also have a custom built 4.19.71-v8+ kernel (I upgraded to 5.4 and was seeing issues so went back to 4.9). I upgraded all packages and rebooted (just power and ethernet connected), at this point the board is basically dead.

In my config.txt is:

    gpu_mem=64
    kernel=kernel8.img
    dtparam=i2c_arm=on
    arm_64bit=1

In my cmdline.txt is:

    console=ttyS0,115200 console=tty0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=bfq rw rootwait dwc_otg.speed=1 dwc_otg.host_support_fs_ls_low_power=1 dwc_otg.host_ls_low_power_phy_clk=1 dwc_otg.fiq_fix_enable=1

**Expected behaviour**

Board should boot up fine regardless if HDMI is connected or not, and ethernet should also work.

**Actual behaviour**

Board does not appear to boot up, orange light on ethernet is on whilst yellow is off, board is unresponsive over network, if HDMI is plugged in after this point then nothing appears on the screen, if a USB keyboard is plugged in and caps lock is pressed, the caps lock light never illuminates.

**System**

    Model: Raspberry Pi 4B (4GB)
    OS: Arch Linux Arm 64-bit
    Firmware: Apr  7 2020 15:56:30 version 6ab498ee82102a9669a1909ae90341e3ecc2128f (clean) (release) (start)
    Kernel: Linux BRPi 4.19.71-v8+ #1 SMP Sun Sep 8 14:10:58 BST 2019 aarch64 GNU/Linux

timg236, 2020-06-12:
Please can you try to reproduce using Raspbian Pi OS. Booting without HDMI monitors connected is very common so this is unlikely to be a firmware issue.

thedjnK, 2020-06-12:
@timg236 Where can I download an image with 64-bit kernel/userland of raspbian?

JamesH65, 2020-06-12:
The 64 bit version of Raspberry PI OS is still in beta but available from here https://www.raspberrypi.org/forums/viewtopic.php?t=275370

However, you are asking for support for a third party OS, running 64bits which is still in beta on our own OS! If it cannot be replicated on either the 32bit or beta 64bit Raspberry PI OS distro's then you would need to talk to the Arch developers.

thedjnK, 2020-06-13:
I have replicated the issue with that 64-bit image. Loaded the image to a 64GB sandisk ultra micro SD card, booted it, configured it, disabled wifi from the menu, enabled ssh, set eth0 IP to static, rebooted and checked I could connect via ssh - all working fine.

Powered board off, removed HDMI adapter with monitor, removed USB mouse and keyboard, powered board back up, only orange ethernet light comes on (vs both when booted with HDMI attached), board is dead over ethernet, plugging the HDMI adapter in at this point yields nothing on the display. Unplugged module, put HDMI adapter in, powered board on and it boots fine, yellow LED on ethernet, board is reachable over ethernet. (I have to remove the power cable to shut the board off as using `poweroff` or `reboot` seems to initiate a shutdown but it never seems to complete, ethernet lights stay on)

    Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux
    Raspberry Pi reference 2020-05-27 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 30e2dd32ba47cc3bec15ab1413c16a17e5797775, stage4
    May 27 2020 12:05:49 Copyright (c) 2012 Broadcom version 369ed4e44cb5a080a2dfa7f854ae4ff46b7c9ef9 (clean) (release) (start)

Interestingly though the board does not seem to lock up when using `tvservice off` as it does in https://github.com/raspberrypi/firmware/issues/1410 with the other image, which is one difference, although I'm unable to re-enable the display or query anything about it:

    root@raspberrypi:/home/pi# tvservice -l
    vchi_msg_dequeue -> -1(90)
    1 attached device(s), display ID's are :
    Display Number 2, type HDMI 0
    root@raspberrypi:/home/pi# tvservice --modes=CEA
    vchi_msg_dequeue -> -1(90)
    tvservice-client: Failed to query supported modes for group CEA in [vc_tv_hdmi_get_supported_modes_new_id]
    Group CEA has 0 modes:
    root@raspberrypi:/home/pi# tvservice -v 2 --modes=CEA
    vchi_msg_dequeue -> -1(90)
    tvservice-client: Failed to query supported modes for group CEA in [vc_tv_hdmi_get_supported_modes_new_id]
    Group CEA has 0 modes:
    root@raspberrypi:/home/pi# tvservice -v 2 --modes=DMT
    vchi_msg_dequeue -> -1(90)
    tvservice-client: Failed to query supported modes for group DMT in [vc_tv_hdmi_get_supported_modes_new_id]
    Group DMT has 0 modes:

pelwell, 2020-06-13:
When you say it fails to boot, have you watched what happens to the green ACT LED after you turn on the power? Do a normal boot with HDMI attached (you could video it for comparison), noting the times at which the desktop/console prompt appears and the main ACT flashing stops, then repeat without HDMI. I'm wondering if the device has actually booted but with no network.

There are few other things you could try:
1. Add `hdmi_force_hotplug=1` to config.txt, power up without HDMI, then plug it in after a suitable wait (e.g. 1 minute).
2. With HDMI, attach the Pi to a WLAN AP, note its IP address, then reboot without HDMI and ping/ssh.

thedjnK, 2020-06-13:
I've connected up a serial device and monitored the output (if the device is set to boot to a graphical interface then there is no output on the debug UART at all, not sure if that's a bug or by design, so swapped over to console login).

Log when HDMI is not connected: https://pastebin.com/P2ksMLU9 and log when HDMI is not connected: https://pastebin.com/wswwSQNC the parts that stand out to me in the log with HDMI not connected:

    [    0.953491] simple-framebuffer: probe of 0.framebuffer failed with error -12
    [   23.654781] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
    [   23.660982] rcu:     2-...0: (2 ticks this GP) idle=2a2/1/0x4000000000000000 softirq=7/7 fqs=2626
    [   23.669821]  (detected by 0, t=5254 jiffies, g=-1087, q=2)
    [   23.675391] Task dump for CPU 2:
    [   23.678668] swapper/0       R  running task        0     1      0 0x0000002a
    [   23.685832] Call trace:
    [   23.688326]  __switch_to+0xc4/0x248
    [   23.691875]  _raw_spin_unlock_irqrestore+0x5c/0x2ac

The one without HDMI connected never gets past this point, it doesn't boot the system or give me a login prompt.

pelwell, 2020-06-15:
I can't reproduce the failure, but your kernel log looks interestingly different with respect to the framebuffer handling.

Using your cmdline.txt and config.txt (OK - I added `enable_uart=1` for convenience), it uses the old bcm2708_fb driver:
```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:91:36:7B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=bfq rw rootwait dwc_otg.speed=1 dwc_otg.host_support_fs_ls_low_power=1 dwc_otg.host_ls_low_power_phy_clk=1 dwc_otg.fiq_fix_enable=1
...
[    1.058439] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    1.058483] bcm2708_fb: probe of soc:fb failed with error -2
```
whereas you get simple-framebuffer:
```
[    0.975055] simple-framebuffer 3e503000.framebuffer: fb0: simplefb registered!
```
or
```
[    0.953491] simple-framebuffer: probe of 0.framebuffer failed with error -12
```
The firmware only enables and configures simple-framebuffer if there is no DT alias called "display0" and there is no enabled device node at "/soc/fb".

I think you must have an alternative DTB file on your system. With HDMI connected, boot the Pi and upload the output of `dtc -I fs /proc/device-tree`.

thedjnK, 2020-06-28:
Sorry for the delay in responding.

> Using your cmdline.txt and config.txt 

With the test 64-bit image that James linked earlier I downloaded that I also got the issue on, I flashed to an empty micro SD card and did not change any of the config.txt or cmdline.txt parameters and encountered the issue, I only made the following changes:

    dtparam=audio=off
    gpu_mem=64

Output from the dtc command: https://pastebin.com/6SgsFpEf there were the following too:

    <stdout>: Warning (ranges_format): /chosen:ranges: empty "ranges" property but its #address-cells (1) differs from / (2)
    <stdout>: Warning (unit_address_vs_reg): /scb: node has a reg or ranges property, but no unit name
    <stdout>: Warning (unit_address_vs_reg): /soc: node has a reg or ranges property, but no unit name
    <stdout>: Warning (unit_address_vs_reg): /soc/gpiomem: node has a reg or ranges property, but no unit name
    <stdout>: Warning (unit_address_vs_reg): /soc/axiperf: node has a reg or ranges property, but no unit name
    <stdout>: Warning (unit_address_vs_reg): /emmc2bus: node has a reg or ranges property, but no unit name
    <stdout>: Warning (unit_address_vs_reg): /axi/vc_mem: node has a reg or ranges property, but no unit name
    <stdout>: Warning (unit_address_vs_reg): /v3dbus: node has a reg or ranges property, but no unit name
    <stdout>: Warning (avoid_default_addr_size): /axi/vc_mem: Relying on default #address-cells value
    <stdout>: Warning (avoid_default_addr_size): /axi/vc_mem: Relying on default #size-cells value
    <stdout>: Warning (avoid_unnecessary_addr_size): Failed prerequisite 'avoid_default_addr_size'
    <stdout>: Warning (unique_unit_address): Failed prerequisite 'avoid_default_addr_size'
    <stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 2 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 4 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 6 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/spi@7e204600:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/i2c@7e804000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/hdmi@7ef05700:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/aux@7e215000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/pwm@7e20c800:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/csi@7e801000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/mmcnr@7e300000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/i2c@7e205a00:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/clock@7ef00000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/i2c@7e205800:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e201a00:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e201a00:clocks: cell 2 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/mmc@7e202000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e201800:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e201800:clocks: cell 2 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/spi@7e204a00:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/hvs@7e400000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/thermal@7d5d2200:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/spi@7e204800:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/csi@7e800000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/pwm@7e20c000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/spi@7e215080:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e201400:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e201400:clocks: cell 2 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/serial@7e215040:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/i2c@7e205c00:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/smi@7e600000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 0 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 1 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 3 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 5 is not a phandle reference
    <stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 7 is not a phandle reference
[... cut ...]
    <stdout>: Warning (gpios_property): /leds/pwr:gpios: cell 0 is not a phandle reference
    <stdout>: Warning (gpios_property): /aliases:gpio: property size (19) is invalid, expected multiple of 4
    <stdout>: Warning (gpios_property): /__symbols__:gpio: property size (19) is invalid, expected multiple of 4
    <stdout>: Warning (alias_paths): /aliases: aliases property name must include only lowercase and '-'
    <stdout>: Warning (alias_paths): /aliases: aliases property name must include only lowercase and '-'

kellertk, 2020-07-26:
I've found this too on the latest 64-bit release. The `tvservice` command will fail unless:
* `dtoverlay=vc4-fkms-v3d` is enabled
* Audio is enabled: `dtparam=audio=on`
And even then it isn't able to query the display modes, and can't power the display back on after powering it off.