raspberrypi/firmware #1410: Raspberry pi 4B (4GB RAM) turning HDMI off locks board up

thedjnK, 2020-06-11:
**Describe the bug**

This seems to be similar to this bug https://github.com/raspberrypi/firmware/issues/1409 but has different action. It was tested with a 5.4.45-v8+ kernel, 62fc8c01165a80021054a430182b504f7b877c2d commit of this repository and with the HDMI cable connected. The board powers up, I login and then turn HDMI off using `tvservice --off`, at this point the board is dead, it stops processing network packets and I am unable to get the screen back on by pressing up, backspace 3 times, typing on and pressing enter.

**To reproduce**

I only have 1 board and 1 SD card so haven't tried reproducing it freshly. How it happens on my board is I have arch linux arm (64-bit) installed, I had a custom built 5.4.45-v8+ kernel. I upgraded all packages and rebooted (power, HDMI and ethernet connected), logged in to the system and then issued the command `tvservice --off`.

In my config.txt is:

    gpu_mem=64
    kernel=kernel8.img
    dtparam=i2c_arm=on
    arm_64bit=1

In my cmdline.txt is:

    console=ttyS0,115200 console=tty0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=bfq rw rootwait dwc_otg.speed=1 dwc_otg.host_support_fs_ls_low_power=1 dwc_otg.host_ls_low_power_phy_clk=1 dwc_otg.fiq_fix_enable=1

**Expected behaviour**

The HDMI display to turn off and the board to continue running.

**Actual behaviour**

HDMI turns off but the board is dead.

**System**

    Model: Raspberry Pi 4B (4GB)
    OS: Arch Linux Arm 64-bit
    Firmware: I've since gone back to an older version, but the commit of the firmware package where the problem is apparent is 62fc8c01165a80021054a430182b504f7b877c2d
    Kernel: 5.4.45-v8+, commit from kernel checkout is c9c0a30fbf19cfdc3abdf8ef2ff6427fb5772be1

**Logs**

Cannot get a log as the board is dead after running the command.

alanbork, 2020-09-16:
can't reproduce on Linux raspberrypi 5.4.61-v7l+ #1339 SMP Tue Sep 1 18:51:27 BST 2020 armv7l GNU/Linux

root@raspberrypi:~# tvservice --off
Powering off HDMI
root@raspberrypi:~# tvservice -s
state 0x2 [TV is off]

timg236, 2022-11-28:
It sounds like this is no longer a problem.

closed 2025-02-14 by popcornmix