raspberrypi/firmware #1411: Is it possible to undo dtoverlay?

juergh, 2020-06-13:
I'm using a gpio conditional to boot into a recovery kernel. I would also like to undo any previous dtoverlay settings so that the recovery kernel boots without any overlays. Is that possible?

The current snippet is:

[all]
[gpio4=0]
os_prefix=recovery/
kernel=vmlinuz
initramfs initrd.img
cmdline=cmdline.txt
enable_uart=1

pelwell, 2020-06-13:
The answer is no, but yes. There is currently no way to undo overlays - the action of applying an overlay is destructive - but your configuration makes it easy: just make sure that all overlays are contained within a `[gpio4=1]` section.

pelwell, 2020-06-13:
Wait - we can do better than that. If you have an `overlays` directory inside your `recovery` directory which contains nothing but a file called README then all attempts to load an overlay will fail but the boot will continue.

juergh, 2020-06-13:
Why are you working on a Saturday? ;-)

Your first suggestion wouldn't work as I'm injecting this recovery kernel into a running system and have no control over previous config settings. I want to pretty much undo everything to make sure the recovery kernel always boots.

Adding recovery/overlays/README seems to do the trick though!

Which makes me wonder.. Ubuntu currently doesn't have a README file in the regular overlays directory but things seem to work just fine. From your comment in #1411 it seems that that file is only needed if os_prefix and/or overlay_prefix are set to non-default values. Is that a true statement?

pelwell, 2020-06-13:
Yes, that's correct. The behaviour is documented (gasp) in our config.txt documentation somewhere.

juergh, 2020-06-13:
Thanks for the clarification. Is there anything else you can think of to 'undo' non-default settings that I should add to my gpio conditional section?

juergh, 2020-06-13:
I've also added:

arm_64bit=0
overlay_prefix=overlays/

pelwell, 2020-06-13:
They're good ones. I would just go through the documentation looking for the critical settings like that.

juergh, 2020-06-13:
Thanks Phil.

closed 2020-06-13 by juergh