raspberrypi/firmware #1412: RPI doesn't shutdown when connected to PoE hat

weshouman, 2020-06-14:
**Describe the bug**
RPI restarts whenever it gets shutdown when powered through a PoE hat. 

**To reproduce**
Connect PoE hat to the RPI.
Power the RPI.
Shutdown the RPI.

**Expected behaviour**
RPI shuts down.

**Actual behaviour**
RPI restarts.

**Additional context**
A previous solution is found on the forum but not publicly accessible [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1506137#p1506137)

popcornmix, 2020-06-15:
What model of Pi is it?

weshouman, 2020-06-16:
That's for the PI 4
Note: PI 3B+ doesn't have this issue, _based on multiple users on the forum_

popcornmix, 2020-06-16:
Is bootloader up to date? https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

mitchind, 2020-06-29:
That is fixed now for me running latest stable firmware
sudo rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Mon 15 Jun 2020 01:36:19 PM UTC (1592228179)
 LATEST: Mon 15 Jun 2020 01:36:19 PM UTC (1592228179)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad

weshouman, 2020-06-30:
I confirm that it now got fixed for me too.

### Further details
#### Before Bootloader Update
The previous output of ```vgencmd bootloader_version``` was
```
May 10 2019 19:40:36
version d2402c53cdeb0f072ff05d52987b1b6b6d474691 (release)
timestamp 0
```
#### After Bootloader Update
The current output of  ```vgencmd bootloader_version``` is
```
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
```
#### How-to
I gathered the steps for updating the bootloader in this [gist](https://gist.github.com/weshouman/f39e63512fd563ea295502a876a4a1cf)

closed 2020-06-30 by weshouman