raspberrypi/firmware #1414: Pops in audio with Kernel 4.19.118-v7+ #1311 and Roon Bridge

dsnyder0pc, 2020-06-16:
**Describe the bug**
I'm running the Roon Bridge application (build 171) on DietPi. A freshly flashed DietPi image "ships" with kernel 4.19.118, and when using that kernel, I'm hearing pops and ticks in audio playback. However, when I switch to a different microSD card that was updated to DietPi 6.30.0 but still has kernel 4.19.97 (did not do `apt full-grade`, everything sounds fine.

**To reproduce**
Flash a microSD card with the latest DietPi image. Install Roon Bridge. Connect a USB DAC to the RPi and play any music.

**Expected behaviour**
Assuming a reasonably good DAC, music should sound nice.

**Actual behaviour**
Pops and ticks in the audio. You can hear what I'm hearing by playing this file: https://www.dsnyder.ws-e.com/ts/linux_4.19.97_vs_4.19.118.mp3

**System**
* Which model of Raspberry Pi?  **RPi CM 3+ (armv7l)**
* Which OS and version?  **Raspberry Pi reference 2020-05-27**
* Which firmware version:
```
Apr 15 2020 11:43:28
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start_cd)
```
* Which kernel version?  **Linux usbridge 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux**

**Logs**
Again, please listen to this file to hear what I'm hearing:  https://www.dsnyder.ws-e.com/ts/linux_4.19.97_vs_4.19.118.mp3

**Additional context**
- I can only reproduce this problem while streaming the file from Roon and with kernel 4.19.118. When I play the file locally (eg., download it from here to the local microSD card and use the SoX play command), it sounds fine with both kernels.
- Others in the Roon community can reproduce this problem and say the workaround is to go back to the kernel that came with the January 26th DietPi image.
- More context in https://github.com/MichaIng/DietPi/issues/3608

dsnyder0pc, 2020-06-17:
I've created a workaround that rolls back to the last good kernel. Works for me, but no guarantee that it will work for everyone. :)

```
# curl -L ws-e.com/to-97.sh | bash
```

MichaIng, 2020-06-17:
Little not about the workaround, before running the curl command:
- Works OOTB on **RPi2 and RPi3(+) only**.
- On other RPi models, adjust the line `target="4.19.97-v7+"`:
  - **RPi1/Zero**: `target="4.19.97+"`
  - **RPi4**: `target="4.19.97-v7l+"`
  - With **64-bit** kernel enabled: `target="4.19.97-v8+"`

Hopefully the kernel version gives a hint which commit is the culprit.