raspberrypi/firmware #1415: arm64: kernel image misaligned at boot

vianpl, 2020-06-16:
Linux v5.8-rc1 showed up with a new warning in arm64:

`[    0.000000] [Firmware Bug]: Kernel image misaligned at boot, please fix your bootloader!`

As per ./Documentation/arm64/booting.rst:

> The Image must be placed text_offset bytes from a 2MB aligned base address anywhere in usable system RAM and called there

I found out that text_offset changed in v5.8-rc1[1], it was previously set to 0x80000 but now is 0x0. Hence the message above.

I suggest we change the load address to 0x0 or if not possible to 0x200000 (2MB). I tested the later successfully.

[1] https://www.spinics.net/lists/arm-kernel/msg798878.html

pelwell, 2020-06-24:
For the curious, the commit zeroing TEXT_OFFSET is https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=cfa7ede20f133cc81cef01dc3a516dda3a9721ee.

This is going to be awkward or inefficient for the firmware because the API is currently "load <this> file <here>", and in the general case (e.g. from a TFTP server) we can't/don't want to seek a file. However, `kernel_address=0x200000` ought to work and silence the warning.

pelwell, 2020-06-25:
The prediction was needlessly pessimistic - there's a patch in the internal repo which should do the right thing:
* If the 'ARMd' ('ARM\x64') magic value is detected in the header, the text offset is read and honoured. If an offset would cause the kernel to overlap with the stub (e.g. `TEXT_OFFSET=0`) then an extra 2MB is added to the load address. As a result, 5.8 aarch64 kernels should load to 0x200000.
* Otherwise it behaves as before.

vianpl, 2020-06-25:
Great! So it'll be available on the next firmware release?

popcornmix, 2020-06-25:
It's in our source repo - will go to raspberry/firmware (master branch) on next build (there will be one today or tomorrow).

popcornmix, 2020-06-27:
@vianpl this is in latest firmware.

pelwell, 2020-06-27:
Note that this change appears to have introduced a problem with network booting.

pelwell, 2020-07-03:
The network booting problem has been resolved in the latest firmware release (https://github.com/raspberrypi/firmware/commit/66fc5eaac3d0af1d5a7ffa616086cbfaefd72e98).

closed 2020-07-03 by pelwell