raspberrypi/firmware #1417: Raspberry Pi 4 hangs at reboot, needs to be unplugged to start up again

user897943, 2020-06-17:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

After updating the bootloader to use USB mass storage boot and changing the boot settings, my raspberry pi stopped rebooting cleanly, it now hangs at some stage, but it appears that this happens only when hdmi is disconnected, I tried to reboot with the pi 4 connected to my monitor to spot any messages but the behavior was different, the pi 4 rebooted normally. With hdmi disconnected, it blinks the green led randomly, the red led stays lit, it does not reboot so I have to pull the power and plug it back in, then the pi 4 starts up as normal. I am using Samsung 860 EVO 250GB with a ASMedia adaptor (it's in the compatibility list)

**To reproduce**
List the steps required to reproduce the issue.

Updated bootloader to 03.06.2020 version with USB mass storage boot
changed bootloader settings to the following, then updated the bootloader again:
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=5000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=10000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf214
MAX_RESTARTS=1
SD_BOOT_MAX_RETRIES=1
NET_BOOT_MAX_RETRIES=1
USB_MSD_DISCOVER_TIMEOUT=60000
USB_MSD_LUN_TIMEOUT=10000

added quirks to make sure UAS is not enabled

the SSD contains a mirror copy of my SD card which used to reboot without issues before I updated the bootloader

I did not generate new UUIDs when cloning to the SSD so it mirrors the SD card

Environment:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I use this Pi 4 to run several containers including my Home Assistant for home automation so when the reboot fails almost everything stops working in my house since all the light bulbs are Philips Hue and the light switches are covered so there is no way to turn lights on/off manually if I'm away from the house, here's the docker data in case it's relevant:

 Version:           19.03.11
 API version:       1.40
 Go version:        go1.13.10
 Git commit:        42e35e6
 Built:             Mon Jun  1 09:20:15 2020
 OS/Arch:           linux/arm
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.11
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.10
  Git commit:       42e35e6
  Built:            Mon Jun  1 09:14:09 2020
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.2.13
  GitCommit:        7ad184331fa3e55e52b890ea95e65ba581ae3429
 runc:
  Version:          1.0.0-rc10
  GitCommit:        dc9208a3303feef5b3839f4323d9beb36df0a9dd
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683

I am also running ISC-DHCP-SERVER on this Pi 4 so when it fails to reboot, I lose access to many devices when their leases expire (set to max lease 600).

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

Clean reboot, as it was when running from SD card and prior to beta bootloader installation

**Actual behaviour**
Add a clear and concise description of what actually happened.

Clear description is all above

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW  - Pi 4
* Which OS and version (`cat /etc/rpi-issue`)?  - Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4

* Which firmware version (`vcgencmd version`)? - Jun 10 2020 17:47:19
Copyright (c) 2012 Broadcom
version e46bba1638cca2708b31b9daf4636770ef981735 (clean) (release) (start)

* Which kernel version (`uname -a`)? - Linux raspberrypi4 5.4.44-v7l+ #1320 SMP Wed Jun 3 16:13:10 BST 2020 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

not sure what logs would be relevant here, I can obtain and provide later if I'm told where to get it from and what I should look out for

**Additional context**
Add any other relevant context for the problem.

no further info

timg236, 2020-06-17:
Please can you retest using the latest released bootloader 
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=277413&start=25

user897943, 2020-06-17:
Before I do this, can I confirm that this is the latest released with USB-MSD boot?

/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-06-15.bin

From my reading on the post you've linked it appears that it's been made stable?

The compatibility list is the unofficial one, you're right.

And the issue isn't when booting, it's when shutting down to reboot, so would that still be the right place?

Thanks in advance

timg236, 2020-06-17:
Yes, this is now the stable release. The easiest way to install is to download the binary zip and create an sd-card using the Raspberry Pi Imager

https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.06.15-137ad

user897943, 2020-06-17:
Thanks.

sudo rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
BOOTLOADER: up-to-date
CURRENT: Mon 15 Jun 2020 01:36:19 PM UTC (1592228179)
 LATEST: Mon 15 Jun 2020 01:36:19 PM UTC (1592228179)
 FW DIR: /lib/firmware/raspberrypi/bootloader/beta
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad

Seems like I was already running the latest...

user897943, 2020-06-17:
I will perform a reboot now. Can you please tell me what logs I should copy from and their location/commands to use to open the logs so I can give you everything straight after booting - I will probably have to unplug and reattach power. I'm using the official USB C power supply.

timg236, 2020-06-17:
If you have a USB serial then please add enable_uart=1 and uart_2ndstage=1 in config.txt

user897943, 2020-06-17:
Can you please elaborate? I have a USB > Serial cable for flashing ESP8266 chips, do I need to connect this to the pi to pull logs? If you're talking about the USB to SATA bridge, yes I am using one so I can connect the SSD to the Pi via USB 3. I'm sorry for being a bit thick but I will need more detail in your responses, I am not asking you to give it to me in a platter but when you say stuff like config.txt it may be obvious to you however I have no idea which config.txt you're referring to, there could be more than one for a different application.

user897943, 2020-06-17:
Here is my /boot/config.txt if that is what you're referring to:

pi@raspberrypi4:/boot $ cat config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=82

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
#hdmi_enable_4kp60=1
pi@raspberrypi4:/boot $

user897943, 2020-06-17:
I added the following to the file:

```
## attempt to avoid reboot crash
enable_uart=1
uart_2ndstage=1
```

user897943, 2020-06-17:
I also added boot_delay=10
as I noticed that was an option I could add, just in case the SSD needs to "initialise"

user897943, 2020-06-17:
I have now issued sudo reboot and as expected, I can no longer ssh into the machine, this is how I know it's frozen, from here I have to pull the plug if I want to have access again. Please let me know what to do and I'll go pull the plug and restart, then I can provide you whatever you need, but please tell me exactly where to fetch the info rather than using "config.txt" so there is no ambiguity.

user897943, 2020-06-17:
```
pi@raspberrypi4:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.44-v7l+ (dom@buildbot) (gcc version 4.9.3 (cros                                                                                                                                                             stool-NG crosstool-ng-1.22.0-88-g8460611)) #1320 SMP Wed Jun 3 16:13:10 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, s                                                                                                                                                             ize 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id share                                                                                                                                                             d-dma-pool
[    0.000000] On node 0 totalpages: 1012480
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 815872 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49856 r8192 d23872 u81920
[    0.000000] pcpu-alloc: s49856 r8192 d23872 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010176
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835                                                                                                                                                             .enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1                                                                                                                                                              video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:22:BF:F9 vc_mem.mem_base=                                                                                                                                                             0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=ttyS0,115200                                                                                                                                                              console=tty1 root=PARTUUID=43ae0dac-02 rootfstype=ext4 elevator=deadline fsck.r                                                                                                                                                             epair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes,                                                                                                                                                              linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, li                                                                                                                                                             near)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x17cc0000-0x1bcc0000] (64MB)
[    0.000000] Memory: 3654596K/4049920K available (10240K kernel code, 724K rwd                                                                                                                                                             ata, 2736K rodata, 2048K init, 853K bss, 133180K reserved, 262144K cma-reserved,                                                                                                                                                              3263488K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 29995 entries in 59 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jif                                                                                                                                                             fies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x518 wit                                                                                                                                                             h crng_init=0
[    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2                                                                                                                                                             147483647500ns
[    0.000031] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_                                                                                                                                                             idle_ns: 1911260446275 ns
[    0.000109] bcm2835: system timer (irq = 17)
[... cut ...]
[   23.355237] eth0: renamed from vethd6e9c4c
[   23.425772] docker0: port 6(vethd3cde60) entered blocking state
[   23.425781] docker0: port 6(vethd3cde60) entered forwarding state
[   25.662928] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net                                                                                                                                                             .ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead

```

user897943, 2020-06-17:
journalctl output after pulling the plug:

https://pastebin.com/ktmjhP4w

user897943, 2020-06-18:
Issue appears to be fixed now.
I erased my SSD then installed a fresh copy of the latest release 27/05, I also did a eeprom restore and set my parameters to turn off the usb reset. Seems to be working now, I can reboot as normal and get to my SSH prompt.

closed 2020-06-18 by user897943