raspberrypi/firmware #1419: What kernel parameters can be passed in cmdline.txt?

gotouchgo, 2020-06-22:
Hi there, do you know what to put in cmdline.txt to specify kernel and initrd ramdisk image?

In old SysLinux style it is:
linux /boot/vmlinuz-linux root=/dev/sda3 initrd=/boot/initramfs-linux.img

I hit issue finding or mounting boot partition when try boot from USB SSD with BerryBoot v2 files. Guess if I can specify the image in cmdline.txt it could help.

lurch, 2020-07-05:
https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md

https://www.raspberrypi.org/documentation/configuration/cmdline-txt.md

gotouchgo, 2020-07-06:
Thanks for the pointers. I figured out how to boot BerryBoot from USB. This is just the first step though, as the non-microSD single USB drive case is not handled well by berryboot for OS installation.

closed 2020-07-06 by gotouchgo

lurch, 2020-07-06:
You might want to have a look at [PINN](https://github.com/procount/pinn), if you haven't already?

gotouchgo, 2020-07-06:
> You might want to have a look at [PINN](https://github.com/procount/pinn), if you haven't already?

Looked at it but PINN doesn't boot from USB SSD for me. I am fine not using them though.