raspberrypi/firmware #1420: Commit 2b76cfc start4.elf and fixup4.dat files doesn't boot

KrisztianKende, 2020-06-27:
I tried to boot with the latest (2b76cfc) start4.elf and fixup4.dat files on Pi4 (8GB), but instead I got seven ACT LED flashes (kernel.img not found). The boot is OK with the previous commit 9d4f640.

Here is the log:
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c302dea0 DATE: Jun 15 2020 TIME: 14:36:22 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1592228179 0x289c4502 0x00d03114
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
Boot mode: USB-MSD (04) order f1
VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x150d8
VL805 0xfff00000 0xfff20000
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3 rport 2 status 00000000 -> 000002a0
USB3 rport 3 status 00000000 -> 000002a0
USB3 rport 4 status 00000000 -> 000002a0
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 400002f1 -> 40200e03
root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3 rport 3 status 000002a0 -> 00021203
root HUB port 3 init
DEV [02:00] 3.16 000000:03 class 0 VID 174c PID 55aa
MSD device [02:00] 3.16 000000:03 conf 0 iface 0 ep 81#1024 02#1024
MSD [02:00] 3.16 000000:03 register MSD
PM_RSTS: 0x00001000
MSD [02:00] 3.16 000000:03 LUN 0
MSD [02:00] 3.16 000000:03 lun 0 block-count 3907029168 block-size 512
MBR: 0x00000001,3907029167 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT dcc3cf174503604e9ac54251ae442ebd 000000001 0e8e088af num-partitions 128 entry-size 128
GPT partition: 0 a2a0d0ebe5b9334487c068b6b72699c7 000000800..0000807ff
lba: 2048 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
PM_RSTS: 0x00001000
Partition: 0
GPT dcc3cf174503604e9ac54251ae442ebd 000000001 0e8e088af num-partitions 128 entry-size 128
GPT partition: 0 a2a0d0ebe5b9334487c068b6b72699c7 000000800..0000807ff
lba: 2048 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
Read config.txt bytes      475 hnd 0x000529f0 hash '57de04aefcb7f6b1'
recover4.elf not found (6)
recovery.elf not found (6)
Read start4.elf bytes  2277216 hnd 0x0005188e hash 'c4f805aae1aba767'
Read fixup4.dat bytes     5409 hnd 0x00051883 hash 'd9b8d1561115150d'
0x00d03114 0x00000000 0x0000001f
MEM GPU: 76 ARM: 947 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0

MESS:00:00:07.723545:0: bfs_xhci_retrieve 3ee76220
MESS:00:00:08.252170:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:08.259419:0: *** Restart logging
MESS:00:00:08.264301:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:08.273599:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:08.279532:0: HDMI0: hdmi_pixel_encoding: 300000000

MESS:00:00:08.285086:0: HDMI1: hdmi_pixel_encoding: 300000000

MESS:00:00:08.295353:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:08.297414:0: Failed to load Device Tree file '?'
MESS:00:00:08.302707:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:08.309589:0: No compatible kernel found

KrisztianKende, 2020-06-27:
With working commit 9d4f640:
MESS:00:00:07.725474:0: bfs_xhci_retrieve 3ee76060
MESS:00:00:08.111168:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:08.114121:0: brfs: File read: 475 bytes
MESS:00:00:08.140931:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:08.149898:0: brfs: File read: 475 bytes
MESS:00:00:08.645237:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:08.652478:0: *** Restart logging
MESS:00:00:08.665190:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:08.671130:0: HDMI0: hdmi_pixel_encoding: 300000000

MESS:00:00:08.676685:0: HDMI1: hdmi_pixel_encoding: 300000000

MESS:00:00:08.687172:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:08.781405:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:08.784646:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xb767
MESS:00:00:08.867451:0: brfs: File read: 46951 bytes
MESS:00:00:08.910982:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:09.349204:0: brfs: File read: 1371 bytes
MESS:00:00:09.351661:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:09.356590:0: dtparam: pwr_led_trigger=none
MESS:00:00:09.454488:0: dtparam: pwr_led_activelow=off
MESS:00:00:09.547250:0: dtparam: act_led_trigger=none
MESS:00:00:09.642323:0: dtparam: act_led_activelow=off
MESS:00:00:09.734982:0: dtparam: eth_led0=4
MESS:00:00:09.824454:0: dtparam: eth_led1=4
MESS:00:00:09.914355:0: dtwarn: overlay 'pi3-disable-wifi' has been renamed 'disable-wifi'
MESS:00:00:09.920277:0: brfs: File read: 475 bytes
MESS:00:00:09.926449:0: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
MESS:00:00:10.006809:0: Loaded overlay 'pi3-disable-wifi'
MESS:00:00:10.126675:0: dtwarn: overlay 'pi3-disable-bt' has been renamed 'disable-bt'
MESS:00:00:10.132277:0: brfs: File read: 387 bytes
MESS:00:00:10.137348:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:10.244636:0: Loaded overlay 'pi3-disable-bt'
MESS:00:00:10.528395:0: brfs: File read: 954 bytes
MESS:00:00:10.530494:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:10.535538:0: Read command line from file 'cmdline.txt':
MESS:00:00:10.541400:0: 'console=serial0,115200 console=tty1 root=PARTUUID=107db29d-fe2f-4bdc-9615-bb43a292042d rootfstype=ext4 fsck.repair=yes rootwait'
MESS:00:00:12.416643:0: brfs: File read: 128 bytes
MESS:00:00:13.537027:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:13.539617:0: Loading 'kernel8.img' to 0x80000 size 0xfb5200
MESS:00:00:13.545878:0: Device tree loaded to 0x2eff4400 (size 0xbb8b)
MESS:00:00:13.552175:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:13.561236:0: uart: Baud rate change done...
MESS:00:00:13.563290:0: uart: Baud rate change done...
MESS:00:00:13.568887:0: bfs_xhci_stop
MESS:00:00:13.571532:0: XHCI-STOP
MESS:00:00:13.574722:0: xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb

pelwell, 2020-06-28:
How many filesystems, on which devices and of which types, do you have on this system? e.g. boot on SD card, kernel root on SSS via USB.

KrisztianKende, 2020-06-28:
A 2 TB SATA HDD on ASM1351 adapter cable ([StarTech](https://www.startech.com/HDD/Adapters/usb-3-1-gen-2-sata-adapter-cable~USB312SAT3CB)), which is connected via USB 3.0, and the SD card slot is empty.
GPT partition table, and there are two filesystems; FAT32 (boot) and EXT4 (system root):
```
> gdisk /dev/sda
Disk /dev/sda: 3907029168 sectors, 1.8 TiB
Model: SPZX-75UA7T0    
Sector size (logical/physical): 512/4096 bytes
Disk identifier (GUID): 17CFC3DC-0345-4E60-9AC5-4251AE442EBD
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 3907029134
Partitions will be aligned on 2048-sector boundaries
Total free space is 2014 sectors (1007.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048          526335   256.0 MiB   0700  boot
   2          526336      3907029134   1.8 TiB     8300  system_data

> parted /dev/sda
Number  Start   End     Size    File system  Name         Flags
 1      1049kB  269MB   268MB   fat32        boot         msftdata
 2      269MB   2000GB  2000GB  ext4         system_data

```
Bootloader EEPROM version is _2020-06-15_:
```
[all]
BOOT_UART=1
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=0
DISABLE_HDMI=0
BOOT_ORDER=0xf14
USB_MSD_PWR_OFF_TIME=0
```
config.txt:
```
disable_overscan=1
framebuffer_width=1920
framebuffer_height=1080
hdmi_force_hotplug=1
hdmi_ignore_cec_init=1
hdmi_group=2
hdmi_mode=82
hdmi_drive=2
max_framebuffers=2

arm_freq=1600
arm_64bit=1
uart_2ndstage=1
disable_splash=1
avoid_warnings=2
hdmi_pixel_encoding=2
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
dtparam=eth_led0=4
dtparam=eth_led1=4
dtoverlay=pi3-disable-wifi
dtoverlay=pi3-disable-bt
```
Custom configured 64-bit Linux kernel:
```
> uname -a
Linux RPI4 5.7.6-v8 #1 SMP PREEMPT Fri Jun 26 15:58:01 CEST 2020 aarch64 aarch64 aarch64 GNU/Linux
```
[... 187 lines cut ...]
-rwxr-xr-x 1 root root  1968 jún   26 15:57 superaudioboard.dtbo
-rwxr-xr-x 1 root root 35570 jún   26 15:57 sx150x.dtbo
-rwxr-xr-x 1 root root  1529 jún   26 15:57 tc358743-audio.dtbo
-rwxr-xr-x 1 root root  2127 jún   26 15:57 tc358743.dtbo
-rwxr-xr-x 1 root root  4822 jún   26 15:57 tinylcd35.dtbo
-rwxr-xr-x 1 root root   779 jún   26 15:57 tpm-slb9670.dtbo
-rwxr-xr-x 1 root root   928 jún   26 15:57 uart0.dtbo
-rwxr-xr-x 1 root root   980 jún   26 15:57 uart1.dtbo
-rwxr-xr-x 1 root root   589 jún   26 15:57 uart2.dtbo
-rwxr-xr-x 1 root root   589 jún   26 15:57 uart3.dtbo
-rwxr-xr-x 1 root root   589 jún   26 15:57 uart4.dtbo
-rwxr-xr-x 1 root root   589 jún   26 15:57 uart5.dtbo
-rwxr-xr-x 1 root root  3532 jún   26 15:57 udrc.dtbo
-rwxr-xr-x 1 root root  2267 jún   26 15:57 upstream.dtbo
-rwxr-xr-x 1 root root  2843 jún   26 15:57 upstream-pi4.dtbo
-rwxr-xr-x 1 root root  1238 jún   26 15:57 vc4-fkms-v3d.dtbo
-rwxr-xr-x 1 root root  1112 jún   26 15:57 vc4-kms-kippah-7inch.dtbo
-rwxr-xr-x 1 root root  2481 jún   26 15:57 vc4-kms-v3d.dtbo
-rwxr-xr-x 1 root root  3084 jún   26 15:57 vc4-kms-v3d-pi4.dtbo
-rwxr-xr-x 1 root root   758 jún   26 15:57 vga666.dtbo
-rwxr-xr-x 1 root root  1036 jún   26 15:57 w1-gpio.dtbo
-rwxr-xr-x 1 root root  1171 jún   26 15:57 w1-gpio-pullup.dtbo
-rwxr-xr-x 1 root root  1558 jún   26 15:57 w5500.dtbo
-rwxr-xr-x 1 root root  1056 jún   26 15:57 wittypi.dtbo
```

satmandu, 2020-06-28:
Seeing the same issue, but with stock kernel, and also using USB boot on the 8Gb RPi4, and with no sd card installed.

This is a working setup:
```
/proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.4

Debian GNU/Linux 10 \n \l

Linux goat 5.4.49-v8+ #1323 SMP PREEMPT Fri Jun 26 14:47:11 BST 2020 aarch64 GNU/Linux
vcgencmd version
Jun 17 2020 21:17:03 
Copyright (c) 2012 Broadcom
version 73bf56edb302498ebd92415077229e743d36065a (clean) (release) (start_x)
vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
vcgencmd bootloader_config
[all]
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_IP=192.168.0.20
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf214
USB_MSD_PWR_OFF_TIME=0
dmesg:
[Jun27 20:51] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[  +0.000000] Linux version 5.4.49-v8+ (dom@buildbot) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1323 SMP PREEMPT Fri Jun 26 14:47:11 BST 2020
[  +0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[  +0.000000] efi: Getting EFI parameters from FDT:
[  +0.000000] efi: UEFI not found.
[  +0.000000] Reserved memory: created CMA memory pool at 0x000000001dc00000, size 256 MiB
[  +0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[  +0.000000] On node 0 totalpages: 2047744
[  +0.000000]   DMA zone: 3580 pages used for memmap
[  +0.000000]   DMA zone: 0 pages reserved
[  +0.000000]   DMA zone: 229120 pages, LIFO batch:63
[  +0.000000]   DMA32 zone: 12288 pages used for memmap
[  +0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[  +0.000000]   Normal zone: 16384 pages used for memmap
[  +0.000000]   Normal zone: 1048576 pages, LIFO batch:63
[  +0.000000] percpu: Embedded 31 pages/cpu s89240 r8192 d29544 u126976
[  +0.000000] pcpu-alloc: s89240 r8192 d29544 u126976 alloc=31*4096
[  +0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[  +0.000000] Detected PIPT I-cache on CPU0
[  +0.000000] CPU features: detected: EL2 vector hardening
[  +0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[  +0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2015492
[  +0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:B0:18:56 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyS0,115200 console=tty1 root=ZFS=rpool/ROOT/debian rootfstype=zfs elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles bootvol=MSD
[  +0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[  +0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[  +0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[  +0.000000] software IO TLB: mapped [mem 0x33f00000-0x37f00000] (64MB)
[  +0.000000] Memory: 7669624K/8190976K available (9532K kernel code, 1104K rwdata, 3348K rodata, 1088K init, 1201K bss, 259208K reserved, 262144K cma-reserved)
[  +0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[  +0.000000] ftrace: allocating 31708 entries in 124 pages
[  +0.000000] rcu: Preemptible hierarchical RCU implementation.
[  +0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[  +0.000000] 	Tasks RCU enabled.
[  +0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[  +0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[  +0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[  +0.000000] GIC: Using split EOI/Deactivate mode
[  +0.000000] random: get_random_bytes called from start_kernel+0x32c/0x4c4 with crng_init=0
[  +0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[... 375 lines cut ...]
max_framebuffers=2
arm_64bit=1
start_x=1
gpu_mem=128
#bootloader_update=1

# Disable Ethernet LEDs
dtparam=eth_led0=4
dtparam=eth_led1=4

# Disable the PWR LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
# Disable the Activity LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
initramfs initrd.img-5.4.49-v8+ followkernel
kernel_address=0x80000
cmdline.txt:
console=serial0,115200 console=tty1 root=ZFS=rpool/ROOT/debian rootfstype=zfs elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles bootvol=MSD
```

(Only difference with non-working config was having the most recent elf and dat files on the boot partition.)

The kernel_address line is changed as per https://github.com/raspberrypi/firmware/issues/1421#issuecomment-650565425 because I was tired and was just trying to throw fixes at the wall.

pelwell, 2020-06-28:
Thanks - by forcing `kernel_address=0x80000` and seeing it not boot you've confirmed that this is not the same issue as #1421. I have a hunch this will also fail with a 32-bit kernel.

Which is your boot partition?

satmandu, 2020-06-28:
boot partition is #1 ```/dev/sda1```

KrisztianKende, 2020-06-29:
I did some tests, and I found that this bug is definitely related to the GUID partition table. So the boot is OK on MBR, but fails with GPT.

timg236, 2020-06-29:
The fix for SD GPT broke the USB GPT change, there's a fix in the pipeline.

KrisztianKende, 2020-07-02:
Thanks, commit 66fc5ea has successfully resolved this issue.

closed 2020-07-02 by KrisztianKende