raspberrypi/firmware #1423: USB boot not working(rpi4) even after firmware update

lohriialo, 2020-06-30:
I've updated to the latest firmware but usb boot not working

**Board**: rpi4 4GB
**OS**: Raspberry Pi OS
```sudo apt update``` 👍 
```sudo apt full-upgrade``` 👍 

- ```vcgencmd bootloader_version```
```
Jun 15 2020 14:36:19
Version c302d.........d781 (release)
```

- ```vcgencmd bootloader_config```
```
BOOT_ORDER=0xf41
```

**Scenario :**

- sdcard boot just fine 👍 
- usb boot not working (took out the sdcard, inserted into a USB micro sd card reader, tried too boot from both USB 2.0 and 3.0 port)

**Result :** https://www.dropbox.com/s/xcsm2klq9knmvpn/IMG_20200630_132446__01.jpg?dl=0

Any pointers is appreciated!

JamesH65, 2020-06-30:
Please use the forum for technical support questions. There is a thread on there specifically for USB boot on the Pi4.

lohriialo, 2020-06-30:
@JamesH65 thanks I'll ask for further help there on the forum

lohriialo, 2020-06-30:
**Update :**
Here's my workaround for anyone else having the same problem
1. Flash Raspberry Pi OS on another USB thumb drive
2. Boot up from sdcard
3. Plugin your USB thumb drive
4. Copy all boot files from sdcard and replace it on USB thumb drive
```sudo cp /boot/*.elf /media/pi/boot```
```sudo cp /boot/*.dat /media/pi/boot```
5. Shut down rpi
6. Remove sdcard
7. Boot from USB should work now

closed 2020-06-30 by lohriialo