raspberrypi/firmware #1424: VL805 xHCI controller: don't report HCCPARAMS1.AC64 as 1.

andreiw, 2020-07-08:
Taking in xHCI + Pi's PCIe RC as a whole, it has a rather important quirk: accesses to 64-bit xHCI registers *cannot* be done using 64-bit MMIO, but must be done as two 32-bit transfers. 64-bit MMIO gives wrong results.

It just so happens that most OSes don't bother with 64-bit transfers, OR hack around in their PCIe CAM to fake out 64-bit accesses (presumably).

U-boot hit this problem: http://u-boot.10912.n7.nabble.com/PATCH-v4-0-9-USB-host-support-for-Raspberry-Pi-4-board-64-bit-td413625.html#a413628

Also, this issue prevents the stock Microsoft Windows 10 xHCI driver from operating correctly with the xHCI device. In this configuration, the xHCI device is exposed using the ACPI device tree instead of looking like a PCIe device (because of non-standard Pi PCIe, which is not ECAM, so no OS would support it anyway in ACPI mode). It's been validated that if the driver is forced to use 32-bit accesses, USB works correctly in Windows 10. We now know that the USB driver bases its decision to use 32-bit or 64-bit transfers on the hCCPARAMS1.AC64 field.

If AC64 is 0, the xHC shall ignore the high order 32 bits of 64 bit data structure pointer fields, and system software shall ignore the high order 32 bits of 64 bit xHC registers.

VL805 reports AC64 as 1.

Would it be possible to change the VL805 microcode to report AC64 as 0? This will have __no impact__ on any current OS working the controller (as there's a 3GB DMA limit anyway), but it will mean Windows 10 USB support without extra OS-level hacks.

andreiw, 2020-07-08:
@pbatard

pelwell, 2020-07-08:
> 64-bit MMIO gives wrong results.

Can you expand on that a little?

> Would it be possible to change the VL805 microcode to report AC64 as 0? This will have no impact on any current OS working the controller (as there's a 3GB DMA limit anyway)

Later versions of the silicon remove the 3GB RAM limit, and if you look at the driver you'll see that for 4GB and 8GB RAM the memory space is positioned at 4GB and 8GB respectively, with the device I/O space and MSI peripheral just below the 4GB boundary. Therefore I think that setting AC64 to 0 will not work.

timg236, 2020-07-08:
This wouldn't help on a CM4 because there is no on-board VL805.  Instead, the PCIe bus is exposed allowing arbitrary XHCI controllers to be connected.

pelwell, 2020-07-08:
XHCI on Linux doesn't even attempt to do QWORD accesses, suggesting this quirk isn't uncommon; Microsoft are going to have to support Lo/Hi DWORD accesses, at least optionally.

pbatard, 2020-07-08:
> Can you expand on that a little?

Sure. This is something that has now been confirmed by multiple developers, and that's probably better illustrated by what the U-Boot folks reported [here](http://u-boot.10912.n7.nabble.com/RFC-PATCH-0-9-USB-host-support-for-Raspberry-Pi-4-board-tp408021.html) as well as [here](http://u-boot.10912.n7.nabble.com/PATCH-v1-00-10-USB-host-support-for-Raspberry-Pi-4-board-tp408481p408484.html):

> Even though the architecture is 64-bit the PCIe BAR is 32-bit and likely the 64-bit wide register accesses initiated by the CPU are not properly translated to a sequence of 32-bit PCIe accesses.
> (64-bit reads), for example, always returns same value in upper and lower 32-bits, e.g. `0xabcd1234abcd1234` instead of `0x00000000abcd1234`.

In other words, when issuing a qword read, the high dword contains erroneous data, and it is also likely that qword writes aren't properly translated either.

Forcing qword accesses as double-dword, in both the U-Boot and the Windows xHCI drivers, fully resolve this problem.

pbatard, 2020-07-08:
@pelwell

> Microsoft are going to have to support Lo/Hi DWORD accesses, at least optionally.

They already do. This is decided by whether the `AC64` flag is set or not, which is the reason we are requesting this flag to be `0` if possible, as this will instantly fix the Microsoft xHCI driver.

pelwell, 2020-07-08:
But my understanding is that the AC64 bit declares whether the controller supports 64-bit _addresses_, not 64-bit _accesses_. If AC64=0 then there is no need for 64-bit accesses, so they should never write to the high words at all.

P33M, 2020-07-08:
> @pelwell
> 
> > Microsoft are going to have to support Lo/Hi DWORD accesses, at least optionally.
> 
> They already do. This is decided by whether the `AC64` flag is set or not, which is the reason we are requesting this flag to be hardcoded to `0` if possible, as this would instantly fix the Microsoft xHCI driver.

Doing this conflates a property of the RC (whether it supports outbound Qword reads/writes in PCIe window space) with properties of the host controller (whether the high dword of addresses are implemented or not, and whether Qword accesses are supported to 64-bit registers).

Nobbling the VL805 firmware to de-feature the xhci controller won't solve the problem if the RC can't physically support Qword accesses - and doesn't help if a future Pi-like product uses a non-VLI xhci device.

pbatard, 2020-07-08:
> But my understanding is that the AC64 bit declares whether the controller supports 64-bit addresses, not 64-bit accesses.

Yes, that was my original interpretation as well, but we are getting some information that Microsoft are using this flag as a means to decide between double 32-bit and 64-bit access.

However, looking back at the Ghidra output for the relevant section of the Windows xHCI driver (`usbxhci.sys`), I am starting to wonder if Microsoft are explicitly using the `AC64` flag in the manner we expect them to do, as, if they did, then the test for `((param_1 + 8) + 0x150) & 1)` below should be for `1` when deciding to read as 64-bit, and not `0` as it currently is:

```C
undefined8 XilRegister_ReadUlong64(longlong param_1,undefined8 *param_2)

{
  undefined8 local_30 [6];
  
  local_30[0] = 0;
  if (*(char *)(*(longlong *)(param_1 + 8) + 0x219) == '\0') {
    DataSynchronizationBarrier(3,3);
    if ((*(ulonglong *)(*(longlong *)(param_1 + 8) + 0x150) & 1) == 0) {
      local_30[0] = *param_2;
    } else {
      DataSynchronizationBarrier(3,3);
      local_30[0] = CONCAT44(*(undefined4 *)((longlong)param_2 + 4),*(undefined4 *)param_2);
    }
  } else {
    Register_ReadSecureMmio(param_1,param_2,3,1,local_30);
  }
  return local_30[0];
}
```

I think there might still be something we need to clarify with regards to whether `AC64` is actually the flag being used to control the Windows driver, so you may want to put this topic on hold until we do.

pelwell, 2020-07-08:
Although I'm curious about where your dialog with Microsoft gets, this topic is effectively closed because we won't be changing the VL805 firmware in the way you have requested.

pbatard, 2020-07-08:
Fair enough. We'll look into this further and try to come up with alternative options.

At any rate, we do appreciate your investigation into this.

andreiw, 2020-11-15:
Closing this off as we'd reached a conclusion that the workaround would be elsewhere.

closed 2020-11-15 by andreiw