raspberrypi/firmware #1430: USB3 and 2.4 GHz Wifi

vandlol, 2020-07-20:
Not sure if this is a firmware or even a hardware related issue.
Whenever i plug in ANY device to any of my USB3 Ports my wifi will no longer connect to any given 2.4 GHz Wifi Networks, while 5.0 GHz works just fine. I tinkered around for a bit and i found what even my Logitech MK300 dongle will be interfered when anything is connected to USB3, as soon as i move the dongle away from the pi (by using a f to m usb cable) it just works again.
Seems like there are troubles just like the HDMI Resolution Problem with USB3.
BTW I'm not the only one having this problem - i found this: https://www.indilib.org/forum/general/6576-pi4-usb3-and-wireless-2-4ghz-interference.html

Not sure if this is even fixable but wanted to report anyway.

Cheers

pelwell, 2020-07-20:
A Logitech MK300 dongle is not _any_ USB3 device - it's a wireless communications device. How about a thumb drive, SD card reader, wired mouse or keyboard, etc.

asavah, 2020-07-21:
This has been known for ages.
Not a pi specific issue.

https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html

vandlol, 2020-07-21:
> A Logitech MK300 dongle is not _any_ USB3 device - it's a wireless communications device. How about a thumb drive, SD card reader, wired mouse or keyboard, etc.

I did specify that the dongle is connected via USB3, actually my ssd is connected via USB3 and my mk300 dongle is connected via USB2

pelwell, 2020-07-21:
Can you briefly try different combinations of USB devices - iust SSD, just MK300, etc. -  to see which breaks WiFi for you. You may need to power it up without the MK300, see if WiFi connects, then add it to use the Pi.

vandlol, 2020-07-21:
> Can you briefly try different combinations of USB devices - iust SSD, just MK300, etc. - to see which breaks WiFi for you. You may need to power it up without the MK300, see if WiFi connects, then add it to use the Pi.

Tried all of that, as soon as i connect anything, even a tiny thumbdrive via USB3 the WiFi disconnects.

pelwell, 2020-07-21:
OK - thanks for clarifying. What is the strength of the 2.4GHz WiFi signal with no USB devices connected? You can run "watch iwconfig" in a terminal window, then disconnect any USB devices, wait a while then read the Link Quality and Signal level. Then plug them back in and re-read the values after a while.

njheb, 2020-07-24:
In case this is useful I had problems when my pi4 was mounted in a heatsink case. I came across the problem when I was beta tesing MSD USB Booting. I started investigating after my wifi would not associate when the USB3 was in use in certain cases.

It seems my pi4 can be very sensitive to its immediate environment where 2.4GHz wifi and USB3 are concerned YMMV. Removing just the bottom plate of the heatsink case or the whole case restored the ability to use 2.4GHz wifi. But the quality of the wifi link varied depending on what was connected and how it was connected with the case in place.

Please see forum posts for detail:
https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=266353&p=1693114#p1693026
https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=266353&p=1693114#p1693114

Typical stats using wavemon without heatsink case: LQ = c.96% SL = c.-43dBm
I did not know how to get the SNR.

vandlol, 2020-07-26:
> OK - thanks for clarifying. What is the strength of the 2.4GHz WiFi signal with no USB devices connected? You can run "watch iwconfig" in a terminal window, then disconnect any USB devices, wait a [...]

Will do that and report back when i get around to my pi

vandlol, 2020-07-26:
> In case this is useful I had problems when my pi4 was mounted in a heatsink case. I came across the problem when I was beta tesing MSD USB Booting. I started investigating after my wifi would not [...]

using an acrylic case with small headsink ribs and an active fan

danywheeler, 2020-09-24:
I have the same problem, but a bit different.
[image]
[image]
The signal is ok, but unable to connect to it.

pelwell, 2020-09-24:
That's not the same problem at all - -23 dBm is a _very_ strong signal. Please start a new issue.

danywheeler, 2020-09-24:
well, if no usb3.0 device is pluged in, it's working fine, so I though might be related.

pelwell, 2020-09-24:
Is the screenshot of the scan results taken with or without the USB3 device attached?

danywheeler, 2020-09-24:
with, but can connect, without everything is working.

pelwell, 2020-09-24:
What is the USB3 device? What does `lsusb` report?

danywheeler, 2020-09-24:
[image]
[image]
by the way the hub has external power, which cause reboot not working, remove the power then it reboots.
also it attached two hard disk. but I think that's not really matter.

danywheeler, 2020-09-24:
[image]

https://www.lulian.cn/product/208-cn.html

pelwell, 2020-09-24:
Try with the hub attached to a USB2 port - performance will be bad, but we're just trying to see if it makes any difference.

danywheeler, 2020-09-24:
[image]

It's working, and scan shows more ESSID, that's odd even in usb3.0 the signal level is not so bad.

pelwell, 2020-09-24:
In the success case it connects to "Dany", but it fails connecting to "DanyHome". How do the two APs/SSIDs differ?

Without the hub attached, can you show all the scan results for Dany and DanyHome?

danywheeler, 2020-09-24:
Nope all connect, DanyHome is a wifi6, I though of compatibility, so use my phone opened a access point.

pelwell, 2020-09-24:
Please disable the credentials for the DanyHome network - it's just confusing things. You can edit /etc/wpa_supplicant/wpa_supplicant.conf and rename the DanyHome to (e.g.) xDanyHome. Then at least we'll be able to see if it can still connect to Dany or now.

danywheeler, 2020-09-24:
without usb3.0 is working, with it just can't connect anymore. you mean set DanyHome as open and try it with usb 3.0 ?

pelwell, 2020-09-24:
You've shown it connecting to Dany _without_ USB3 and not connecting to DanyHome _with_ USB3. I want to see it trying and failing to connect to Dany _with_ USB3, so to do that you need to stop it trying to connect to DanyHome.

danywheeler, 2020-09-24:
wlan0: Trying to associate with SSID 'Dany'
wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
wlan0: Trying to associate with SSID 'Dany'
wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
wlan0: Trying to associate with SSID 'Dany'

This is with usb3.0 and connect to Dany it's the same no matter which access point.

danywheeler, 2020-09-24:
It's working if connect to a 5Ghz access point, I bet it's interference, but hey I'm no expert.

pelwell, 2020-09-24:
Can you post the scan results for Dany with _and_ without USB3?

danywheeler, 2020-09-24:
wlan0     Scan completed :
          Cell 01 - Address: A2:9D:7E:1C:2A:E6
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-13 dBm
                    Encryption key:off
                    ESSID:""
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 0000
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030101
                    IE: Unknown: 050400010000
                    IE: Unknown: 0706434E20010D1E
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AEF1903FFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D1601050000000000000000000000000000000000000000
                    IE: Unknown: 7F0804000F0200000040
                    IE: Unknown: BF0C92398933FAFF0000FAFF0020
                    IE: Unknown: C005000300FCFF
                    IE: Unknown: FF1A230D01081A400002600C891FC1839C010800FAFFFAFF191CC771
                    IE: Unknown: FF0724F43F0012FCFF
                    IE: Unknown: FF03270500
                    IE: Unknown: FF0E260008A9FF2FA9FF4575FF6575FF
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD088CFDF00101020100
                    IE: Unknown: DD138CFDF001010201000201010303010100040101
                    IE: Unknown: DD168CFDF0040000494C51030209720100000000FEFF0000
          Cell 02 - Address: 9C:9D:7E:1C:2A:E6
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-14 dBm
                    Encryption key:on
                    ESSID:"DanyHome"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 000844616E79486F6D65
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030101
                    IE: Unknown: 050400010000
                    IE: Unknown: 0706434E20010D1E
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 460573D000000C
                    IE: Unknown: 2D1AEF1903FFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D1601050400000000000000000000000000000000000000
                    IE: Unknown: 7F0804000F0200000040
                    IE: Unknown: BF0C92398933FAFF0000FAFF0020
                    IE: Unknown: C005000300FCFF
                    IE: Unknown: FF1A230D01081A400002600C891FC1839C010800FAFFFAFF191CC771
                    IE: Unknown: FF0724F43F0012FCFF
                    IE: Unknown: FF03270500
                    IE: Unknown: FF0E260103A42827A428427328627228
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD088CFDF00101020100
                    IE: Unknown: DD138CFDF001010201000201010303010100040101
                    IE: Unknown: DD168CFDF0040000494C51030209720100000000FEFF0000
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
[... 40 lines cut ...]
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 9 Mb/s
                              18 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 12 Mb/s; 24 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 00026332
                    IE: Unknown: 010882848B961224486C
                    IE: Unknown: 030106
                    IE: Unknown: 32040C183060
                    IE: Unknown: 0706434E20010D14
                    IE: Unknown: 050400010000
                    IE: WPA Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 2A0104
                    IE: Unknown: 7F080000080000000000
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: 0B05010018127A
                    IE: Unknown: DD07000C4303000000

pelwell, 2020-09-24:
It could well be interference - there is a known problem, not just with Pis, with USB3 clocks and 2.4G WiFi interfering. Channel 1 is the worst case - try moving the AP to channel 11.

danywheeler, 2020-09-24:
without usb3.0

wlan0     Scan completed :
          Cell 01 - Address: A2:9D:7E:1C:2A:E6
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-11 dBm
                    Encryption key:off
                    ESSID:""
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 0000
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030101
                    IE: Unknown: 050400010000
                    IE: Unknown: 0706434E20010D1E
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AEF1903FFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D1601050000000000000000000000000000000000000000
                    IE: Unknown: 7F0804000F0200000040
                    IE: Unknown: BF0C92398933FAFF0000FAFF0020
                    IE: Unknown: C005000300FCFF
                    IE: Unknown: FF1A230D01081A400002600C891FC1839C010800FAFFFAFF191CC771
                    IE: Unknown: FF0724F43F0012FCFF
                    IE: Unknown: FF03270500
                    IE: Unknown: FF0E260008A9FF2FA9FF4575FF6575FF
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD088CFDF00101020100
                    IE: Unknown: DD138CFDF001010201000201010303010100040101
                    IE: Unknown: DD168CFDF0040000494C51030209720100000000FEFF0000
          Cell 02 - Address: 9C:9D:7E:1C:2A:E6
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-11 dBm
                    Encryption key:on
                    ESSID:"DanyHome"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 000844616E79486F6D65
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030101
                    IE: Unknown: 0706434E20010D1E
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 460573D000000C
                    IE: Unknown: 2D1AEF1903FFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D1601050000000000000000000000000000000000000000
                    IE: Unknown: 7F0804000F0200000040
                    IE: Unknown: BF0C92398933FAFF0000FAFF0020
                    IE: Unknown: C005000300FCFF
                    IE: Unknown: FF1A230D01081A400002600C891FC1839C010800FAFFFAFF191CC771
                    IE: Unknown: FF0724F43F0012FCFF
                    IE: Unknown: FF03270500
                    IE: Unknown: FF0E260103A42827A428427328627228
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD168CFDF0040000494C51030209720100000000FEFF0000
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
[... 658 lines cut ...]
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 0000
                    IE: Unknown: 010882848B0C12961824
                    IE: Unknown: 03010B
                    IE: Unknown: 050400010000
                    IE: Unknown: 0706434E20010D1B
                    IE: Unknown: 23021E00
                    IE: Unknown: 46050000000000
                    IE: Unknown: 2A0103
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: 32043048606C
                    IE: Unknown: 0B0500006F0000
                    IE: Unknown: 2D1AED1117FFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D160B001100000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080100000000000000
                    IE: Unknown: DD08AC853D8201000000
                    IE: Unknown: DD230050F204104A0001101044000102100800020780103C0001011049000600372A000120

GithubUser5462, 2020-11-02:
Hello i just wound up here after several hours of troubleshooting.
In my case the USB device is a WD 2TB external HDD (My Passport).

If it is not plugged in or if it is plugged into any USB2 port the Rpi4 connects to the wifi.
If it is plugged into the bottom USB3 port the Rpi4 can still see the wifi networks but won't connect to any of them.
Interestingly enough, if i use the top USB3 port the wifi works.

What i've noticed while troubleshooting is that, when the drive is plugged in before boot the following line is missing from journalctl:
`kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready`

Wpa supplicant prints the following error when trying to join a network:
`wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16`

I have access to 3 wifi networks and i have added them in the wpa supplicant config.
All 3 of them work when the disk is not plugged in the bottom USB3 port.
One wifi is setup on an Asus router 10cm away from the raspberry pi. The raspberry pi has heatsinks but is not in a case.

Plugging in the external drive is enough to disrupt the wifi, it doesn't need to be in use or even mounted.
If i plug it in after boot and after the pi connects to the wifi network, the connection stops working.
ping -I wlan0 1.1.1.1 - went from 16.3 ms to 15524 ms and eventually starts printing `ping: sendmsg: No buffer space available`.
If i disconnect the drive and restart the command, it returns back to normal speed.

pelwell, 2020-11-03:
How are you powering that external HDD? I wouldn't be surprised if your Pi is suffering from undervoltage to some extent - although USB3 interference with WiFi is a common problem (not just with Pi 4), I didn't think it was this bad.

draeath, 2021-04-23:
I can replicate this behavior when connecting a Samsung T5 USB3 SSD drive (idVendor=04e8, idProduct=61f5), rated 5V 0.8A

- vcgencmd get_throttled shows 0x0 (no undervolt there)
- vcgencmd measure_volts shows 0.8700V core, 1.1000V sdram - no difference when the device is disconnected.

pelwell, 2021-04-23:
Well, as the first reply said:
> This has been known for ages.
Not a pi specific issue.
>
> https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html

draeath, 2021-04-23:
Yes, I saw that. I hobby with radio and DSP, so I even understand it.

There is still value in providing detailed reports.

This could be addressed in future hardware revisions if there was will and budget to do so, and meanwhile it could be noted [in the documentation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md), which it currently is not. Other similar issues are. Adding this extra detail supports replication of the problem, which would be useful in such development or documentation.

The Pi board could have been engineered to reduce or prevent this by properly shielding (or coupling with ground) the USB3 signal lines, and the phenomena being apparently undocumented on the Pi end of things, leading anyone who runs into it on an odyssey until they discover an issue report like this one. It could also be partially mitigated by implementation of wifi hardware that supports the 5ghz band channels, instead of a chip limited to the affected channels.

pelwell, 2021-04-23:
> It could also be partially mitigated by implementation of wifi hardware that supports the 5ghz band channels, instead of a chip limited to the affected channels.

Pardon? The only Pis that supports USB3 also support 5GHz WiFi bands, which is why the original issue report says:

> 5.0 GHz works just fine.

draeath, 2021-04-23:
Then I have more research to do, because my 5ghz network doesn't show up.

Thank you, I'll to bark up that tree :)