raspberrypi/firmware #1431: [RPi2] [Kernel5.4] Lowering arm_freq_min leads to system hang/crash

MichaIng, 2020-07-21:
**Describe the bug**
I was upgrading to the newest firmware + kernel packages, which resulted in system hangs and/or crashes. I narrowed down the issue to `arm_freq_min` which I lowered to `150` or `300` (tested both) to allow the system clocking below 600 Mhz. Commenting the setting leads to a stable system, setting/reducing it leads to a quickly hanging or crashing system.

**To reproduce**
1. Upgrade the kernel on `Raspberry Pi 2 Model B Rev 1.1` to current package release `5.4.51-v7+`.
2. Set `arm_freq_min` to `300` (`gpu_mem=16`, if relevant)
3. reboot
4. play around in the file system, which some executables (like htop) until it either hangs or crashes. The last time I triggered it with `vcgencmd measure_clock gpu`.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Setting `arm_freq_min` to `300` should not lead to system crashes.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `Raspberry Pi 2 Model B Rev 1.1`
* Which OS and version (`cat /etc/rpi-issue`)? `Raspbian GNU/Linux bullseye/sid`
* Which firmware version (`vcgencmd version`)? `version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start_cd)`
* Which kernel version (`uname -a`)?
```
Linux micha.gnoedi.org 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux
```

**Logs**
```
[  189.433811] 8<--- cut here ---
[  189.433874] Unable to handle kernel NULL pointer dereference at virtual address 0000000c
[  189.433930] pgd = 89d1d828
[  189.433974] [0000000c] *pgd=36ce9835, *pte=00000000, *ppte=00000000
[  189.434029] Internal error: Oops: 17 [#1] SMP ARM
[  189.434055] Modules linked in:
[  189.434089] CPU: 3 PID: 487 Comm: bash Not tainted 5.4.51-v7+ #1326
[  189.434116] Hardware name: BCM2835
[  189.434151] PC is at filemap_map_pages+0x118/0x448
[  189.434181] LR is at filemap_map_pages+0x43c/0x448
[  189.434207] pc : [<80272f9c>]    lr : [<802732c0>]    psr: 80000113
[  189.434235] sp : b82afe40  ip : b82afe40  fp : b82afe9c
[  189.434261] r10: b8e9d868  r9 : b82afeb4  r8 : 000000c3
[  189.434289] r7 : 80d04f48  r6 : 00000406  r5 : 000000cf  r4 : ba319480
[  189.434318] r3 : 00000004  r2 : 000000c3  r1 : b9406130  r0 : 00000008
[  189.434350] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  189.434382] Control: 10c5387d  Table: 3832006a  DAC: 00000055
[  189.434412] Process bash (pid: 487, stack limit = 0x41237b4b)
[  189.434441] Stack: (0xb82afe40 to 0xb82b0000)
[  189.434475] fe40: 000000c3 00000000 b82afeb4 b832ba80 80d04f48 b8e9d86c 000000c3 00030000
[  189.434518] fe60: b9406130 00000000 00000000 b19558f1 00000001 000000cf 80d04f48 000000c0
[  189.434561] fe80: b82afeb4 000d7000 00000007 00000000 b82aff1c b82afea0 802ab58c 80272e90
[  189.434604] fea0: 8010bbbc 80d04f48 00000000 000d7cfc b82afef4 b6d06360 00000054 00100cca
[  189.434646] fec0: 000000c7 000d3000 b8320000 b8320000 00000000 00000000 00000000 00000000
[  189.434690] fee0: b6ce934c ba309d34 00000000 b19558f1 b82aff1c b82affb0 b6c88000 000d7cfc
[  189.434733] ff00: b8833400 00000017 b8833440 00000000 b82aff74 b82aff20 808d37cc 802aadd4
[  189.434776] ff20: 8012fcb0 8020b210 b82aff54 b82aff38 8010cfdc 8010cf4c 8010bbbc 80d04f48
[  189.434819] ff40: 00000000 00000054 b82aff8c 80d0a7d4 00000017 808d34ac 000d7cfc b82affb0
[  189.434862] ff60: 000d7110 00000000 b82affac b82aff78 80115854 808d34b8 80101d1c 00000001
[  189.434905] ff80: b82affac b82aff90 8020b3fc 00071ae0 20000010 ffffffff 10c5387d 10c5387d
[  189.434948] ffa0: 00000000 b82affb0 80101d24 80115818 00000014 00000000 000d7cfc 00000000
[  189.434991] ffc0: 00000014 000fe744 00103f60 000001e7 00d9b0e8 000d7110 00000000 001052ac
[  189.435035] ffe0: 000fe078 7eb3b408 00054684 00071ae0 20000010 ffffffff 00000000 00000000
[  189.435067] Backtrace:
[  189.435104] [<80272e84>] (filemap_map_pages) from [<802ab58c>] (handle_mm_fault+0x7c4/0xa4c)
[  189.435151]  r10:00000000 r9:00000007 r8:000d7000 r7:b82afeb4 r6:000000c0 r5:80d04f48
[  189.435186]  r4:000000cf
[  189.435220] [<802aadc8>] (handle_mm_fault) from [<808d37cc>] (do_page_fault+0x320/0x3a8)
[  189.435266]  r10:00000000 r9:b8833440 r8:00000017 r7:b8833400 r6:000d7cfc r5:b6c88000
[  189.435301]  r4:b82affb0
[  189.435333] [<808d34ac>] (do_page_fault) from [<80115854>] (do_DataAbort+0x48/0xc4)
[  189.435377]  r10:00000000 r9:000d7110 r8:b82affb0 r7:000d7cfc r6:808d34ac r5:00000017
[... 81 lines cut ...]
[  210.451944]  r8:000000c3 r7:b82afe24 r6:ffffffff r5:80000113 r4:80272f9c
[  210.451984] [<80272e84>] (filemap_map_pages) from [<802ab58c>] (handle_mm_fault+0x7c4/0xa4c)
[  210.452030]  r10:00000000 r9:00000007 r8:000d7000 r7:b82afeb4 r6:000000c0 r5:80d04f48
[  210.452064]  r4:000000cf
[  210.452095] [<802aadc8>] (handle_mm_fault) from [<808d37cc>] (do_page_fault+0x320/0x3a8)
[  210.452140]  r10:00000000 r9:b8833440 r8:00000017 r7:b8833400 r6:000d7cfc r5:b6c88000
[  210.452174]  r4:b82affb0
[  210.452205] [<808d34ac>] (do_page_fault) from [<80115854>] (do_DataAbort+0x48/0xc4)
[  210.452249]  r10:00000000 r9:000d7110 r8:b82affb0 r7:000d7cfc r6:808d34ac r5:00000017
[  210.452283]  r4:80d0a7d4
[  210.452313] [<8011580c>] (do_DataAbort) from [<80101d24>] (__dabt_usr+0x44/0x60)
[  210.452348] Exception stack(0xb82affb0 to 0xb82afff8)
[  210.452378] ffa0:                                     00000014 00000000 000d7cfc 00000000
[  210.452421] ffc0: 00000014 000fe744 00103f60 000001e7 00d9b0e8 000d7110 00000000 001052ac
[  210.452462] ffe0: 000fe078 7eb3b408 00054684 00071ae0 20000010 ffffffff
[  210.452498]  r8:10c5387d r7:10c5387d r6:ffffffff r5:20000010 r4:00071ae0
```

**Additional context**
```
2020-07-21 19:43:53 root@micha:~# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
300000 360000 450000 600000 900000
```
This is new and probably the reason for the crashes when lowering minimum frequency. When leaving at 600, there are only two pstates 600 and 900 and with kernel 4.19 there are always only two.
I was actually hoping for that feature, so great work, however sadly at least my RPi model does not work fine with it.

popcornmix, 2020-07-21:
I can reproduce this. The backtraces in kernel seemed pretty random to me, so probably a clock/voltage issue, rather than a kernel bug.

MichaIng, 2020-07-22:
> I can reproduce this.

You mean you "can" or you "can't" reproduce it? The issue is present with default clocks+voltage in my case, with only the minimum arm clock reduced. We probably just found a second case with RPi Zero.

popcornmix, 2020-07-22:
I can see the crash on a Pi3+. I couldn't provoke it on a Pi4.
I added logging and last thing that occurred was a switch from ~300MHz to 1400MHz.
Everything looked as expected (e.g. the core voltage was raised before the frequency).
Possibly there is an issue with clocks/PLLs switching by large amounts (and perhaps overshooting),
but that is just guessing currently.

Workaround for now is to disable the arm_freq_min. I'll let you know when it's safe to add back in.

MichaIng, 2020-07-22:
Strange only that pre-5.4 the same large clocks jump was never an issue and with only lowest and highest clocks as only two pstates the jump as always the largest possible.

One could make a test with adjusting `arm_freq` and `arm_freq_min` so that only three as close as possible pstages are available. There seem to be fixed closks, 900, 600, 450, 360, 300, ?, 200, 150, so this should be easy. With this the intermediate pstages themselves could be ruled out as issue, which at least this was my very guess.

jeffmd, 2020-07-23:
Had the same problem on a model 1b from 2013 that runs off a battery bank and solar.  Don't remember why I had arm_freq_min set so low since it did not make a huge difference in power draw.  When I updated yesturday to 5.4 and then rebooted everything seemed to be fine.  Ran top and within about 30 seconds the screen froze.  I then used the default config.txt and after a reboot there was no more freezing when cpu load increased.  Was going to continue trouble shooting today but checked the github issues and bingo Michalng saved me some time, thankyou.

MichaIng, 2020-07-25:
May I express the urgency I see in resolving or working around this bug? This has the potential to destroy systems by causing file corruption in unconditionally crashed services, e.g. databases and similar.

E.g. `postinst` could remove/comment `arm_freq_min` at least on affected systems (all but RPi4 it seems). There was a package upgrade yesterday but I don't see a hint that it has been fixed or worked around.

Let me know if there is anything I can test to help getting this resolved quickly.

popcornmix, 2020-08-04:
We have a workaround in latest rpi-update firmware that will disallow arm_freq_min below 600.
A fuller fix is being worked on but will need more work/testing so will wait until after the stable release is settled.

vmachiel, 2020-08-05:
Kind of a noob here: is this fix available now (using apt-update/upgrade)?

JamesH65, 2020-08-05:
No, you will need to use rpi-update. No schedule on apt.

vmachiel, 2020-08-05:
Ok thanks

MichaIng, 2020-08-05:
And remember that this is only a workaround for user which are not yet aware of the issue. In your case you simply `arm_freq_min` to value of 600 or higher to have the exact same result (regarding this specific issue) 😉.

vmachiel, 2020-08-06:
> And remember that this is only a workaround for user which are not yet aware of the issue. In your case you simply `arm_freq_min` to value of 600 or higher to have the exact same result (regarding this specific issue) 😉.

So If i just comment out my arm_freq_min=300, it will go to the default 600 right? And I can change it back once a full update is out that fixes this? (I'm not using rpi-update, I'm waiting for apt, the warning scared me 🙂)

MichaIng, 2020-08-06:
> So If i just comment out my arm_freq_min=300, it will go to the default 600 right?

Yes exactly. On RPi1+Zero it's 700 Mhz but all defaults work fine.

vmachiel, 2020-08-20:
Maybe this is the wrong place to ask, but how do I know when a version is out that fixes this?

MichaIng, 2020-08-20:
Subscribe to this issue, I'm sure we'll get a dev notice once a real fix is merged and I'll anyway keep an eye on it as well and search through release commits when I recognise them and will post here in case.

popcornmix, 2020-08-20:
There is a proper fix for this as in internal PR. I'll let you know when it reached rpi-update.

popcornmix, 2020-08-25:
Latest rpi-update firmware contains a fix for this issue that doesn't involve limiting arm_freq_min. Please update and test.

MichaIng, 2020-08-25:
Jep seems to work fine. Just tested on RPi2 with `arm_freq_min=150` which enables a lowest pstate of 200 MHz and all states are used, no hang or crash until now:
```
2020-08-25 23:05:25 root@micha:~# cat /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state
200000 36921
225000 152
257142 195
300000 225
360000 282
450000 307
600000 1073
900000 8511
```
200 MHz seems to be the lowest supported frequency, right?

vmachiel, 2020-08-26:
That's great! But as I understand, rpi-update is for pre-release stuff right? It will come to apt eventually?

MichaIng, 2020-08-26:
Yes, `rpi-update` by default loads the current [master branch](https://github.com/raspberrypi/firmware/tree/master), compared to the [stable branch](https://github.com/raspberrypi/firmware/tree/stable) that matches the apt packages. It would be great if you could test it as well, but only if you have a full SD card backup that you can recover, just in case any issues appear.

popcornmix, 2020-08-26:
It will come to apt but will take a while as we've just finalised a stable apt version and a commit like this (which affects clocks/plls) could do with some time on the testing branch.

vmachiel, 2020-08-28:
> It will come to apt but will take a while as we've just finalised a stable apt version and a commit like this (which affects clocks/plls) could do with some time on the testing branch.

Ok cool. Totally get that, I’m just happy it’s getting a fix! I’ll keep my eye on this thread for info on the apt release.

vmachiel, 2020-09-20:
So I’m seeing another update round of updates of the firmware in apt. Does this contain the fix? Thanks

MichaIng, 2020-09-20:
No, only a single commit has been merged that has nothing to do with this issue: https://github.com/raspberrypi/firmware/commit/2b41f509710d99758a5b8efa88d95dd0e9169c0a

MichaIng, 2020-09-30:
Okay not sure if it is related, but with newest kernel I get another crash:
```
[  407.680929] rcu: INFO: rcu_sched self-detected stall on CPU
[  407.680961] rcu:     3-....: (2099 ticks this GP) idle=9a2/1/0x40000002 softirq=9183/9183 fqs=1049
[  407.680969]  (t=2100 jiffies g=35521 q=628)
[  407.680977] NMI backtrace for cpu 3
[  407.680989] CPU: 3 PID: 633 Comm: mysqld Not tainted 5.4.68-v7+ #1343
[  407.680993] Hardware name: BCM2835
[  407.680998] Backtrace:
[  407.681022] [<8010d4f4>] (dump_backtrace) from [<8010d7ec>] (show_stack+0x20/0x24)
[  407.681032]  r6:b5f48000 r5:00000000 r4:80e97abc r3:02ff4cf1
[  407.681045] [<8010d7cc>] (show_stack) from [<80908944>] (dump_stack+0xe0/0x124)
[  407.681059] [<80908864>] (dump_stack) from [<80910398>] (nmi_cpu_backtrace+0xc8/0xcc)
[  407.681069]  r8:00000140 r7:80a0202c r6:00000003 r5:00000000 r4:00000003 r3:02ff4cf1
[  407.681083] [<809102d0>] (nmi_cpu_backtrace) from [<80910498>] (nmi_trigger_cpumask_backtrace+0xfc/0x138)
[  407.681090]  r5:80e07d3c r4:8010f3dc
[  407.681104] [<8091039c>] (nmi_trigger_cpumask_backtrace) from [<801105fc>] (arch_trigger_cpumask_backtrace+0x20/0x24)
[  407.681112]  r7:80e05044 r6:80000193 r5:80a0201c r4:00000003
[  407.681125] [<801105dc>] (arch_trigger_cpumask_backtrace) from [<80196744>] (rcu_dump_cpu_stacks+0xb4/0xe4)
[  407.681135] [<80196690>] (rcu_dump_cpu_stacks) from [<80195e18>] (rcu_sched_clock_irq+0x868/0xa80)
[  407.681147]  r10:3d818000 r9:80ed5b18 r8:80e05034 r7:80da6f40 r6:be5bef40 r5:80ea8904
[  407.681153]  r4:80e10d80 r3:00002a0f
[  407.681166] [<801955b0>] (rcu_sched_clock_irq) from [<8019f674>] (update_process_times+0x3c/0xa8)
[  407.681177]  r10:801b2e50 r9:be5b85f0 r8:be5b8540 r7:0000005e r6:eb9f3387 r5:b5f49dd0
[  407.681181]  r4:00000000
[  407.681196] [<8019f638>] (update_process_times) from [<801b25a8>] (tick_sched_handle+0x64/0x70)
[  407.681202]  r4:be5b8870 r3:80000013
[  407.681213] [<801b2544>] (tick_sched_handle) from [<801b2eac>] (tick_sched_timer+0x5c/0xb8)
[  407.681224] [<801b2e50>] (tick_sched_timer) from [<801a0638>] (__hrtimer_run_queues+0x164/0x324)
[  407.681233]  r7:b5f48000 r6:be5b8540 r5:be5b8580 r4:be5b8870
[  407.681243] [<801a04d4>] (__hrtimer_run_queues) from [<801a104c>] (hrtimer_interrupt+0x130/0x2a4)
[  407.681254]  r10:be5b85c8 r9:be5b85f0 r8:be5b8540 r7:ffffffff r6:7fffffff r5:00000003
[  407.681259]  r4:20000193
[  407.681273] [<801a0f1c>] (hrtimer_interrupt) from [<80771744>] (arch_timer_handler_phys+0x40/0x48)
[  407.681284]  r10:80e04f88 r9:b5f48000 r8:80e66b54 r7:b946dd00 r6:000000a2 r5:b9402f40
[  407.681288]  r4:80da62e4
[  407.681301] [<80771704>] (arch_timer_handler_phys) from [<80188500>] (handle_percpu_devid_irq+0x88/0x23c)
[  407.681314] [<80188478>] (handle_percpu_devid_irq) from [<80181a00>] (generic_handle_irq+0x34/0x44)
[  407.681324]  r9:b5f48000 r8:b9460000 r7:00000001 r6:00000000 r5:00000000 r4:80da62e4
[  407.681336] [<801819cc>] (generic_handle_irq) from [<8018214c>] (__handle_domain_irq+0x6c/0xc4)
[  407.681348] [<801820e0>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
[  407.681359]  r8:b5f49e40 r7:b5f49e04 r6:ffffffff r5:80000013 r4:00000003 r3:b5f49dd0
[  407.681370] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[  407.681376] Exception stack(0xb5f49dd0 to 0xb5f49e18)
[  407.681384] 9dc0:                                     b9651e88 00000000 0000b5f4 0000de24
[  407.681395] 9de0: 00000001 80e04f88 b9651e84 b9651e84 b5f49e40 ffffffff 80e04f88 b5f49e2c
[  407.681403] 9e00: b5f49e30 b5f49e20 801b4e58 8092a02c 80000013 ffffffff
[  407.681410]  r4:8092a02c r3:02ff4cf1
[  407.681424] [<80929fec>] (_raw_spin_lock) from [<801b4e58>] (futex_wake+0xc4/0x1c0)
[  407.681436] [<801b4d94>] (futex_wake) from [<801b79dc>] (do_futex+0x160/0xcfc)
[  407.681447]  r10:80e04f88 r9:02648194 r8:00000081 r7:000000f0 r6:7fffffff r5:80e04f88
[  407.681451]  r4:00000001
[  407.681464] [<801b787c>] (do_futex) from [<801b8780>] (sys_futex_time32+0x9c/0x19c)
[  407.681475]  r10:7fffffff r9:02648194 r8:00000081 r7:000000f0 r6:00000000 r5:80e04f88
[  407.681479]  r4:00000001
[  407.681491] [<801b86e4>] (sys_futex_time32) from [<801011a4>] (__sys_trace_return+0x0/0x1c)
[  407.681497] Exception stack(0xb5f49fa8 to 0xb5f49ff0)
[  407.681506] 9fa0:                   00000001 00000000 02648194 00000081 7fffffff 00000000
[  407.681515] 9fc0: 00000001 00000000 00000001 000000f0 000008f0 00000001 00000000 0264816c
[  407.681523] 9fe0: 02648188 58ff9c38 02648168 76d00a14
[  407.681534]  r10:000000f0 r9:b5f48000 r8:801011c4 r7:000000f0 r6:00000001 r5:00000000
[  407.681538]  r4:00000001
```

popcornmix, 2020-10-01:
I don't see any connection with that backtrace and this issue.
You'll need to find a way of provoking this reliably.
Then confirm if issue still occurs without arm_freq_min.
Also confirm if this issue is new to a recent rpi-update kernel/firmware 
(e.g, report first version it occurs with and last version is didn't occur with)

MichaIng, 2020-10-01:
Very latest kernel/firmware, to test the [initial_turbo+performance](https://github.com/raspberrypi/firmware/issues/1469) governor solution. Just reverted min frequencies to defaults, booted with performance governor, which gives 900 MHz (RPi2) now reliable with `initial_turbo`, then switched to `schedutil` governor, which works fine until now:
```
2020-10-01 12:21:26 root@micha:~# cat /sys/devices/system/cpu/cpufreq/policy0/{scaling_governor,/stats/time_in_state}
schedutil
600000 127355
700000 892
800000 831
900000 31905
2020-10-01 12:21:36 root@micha:~# uname -a
Linux micha.gnoedi.org 5.4.68-v7+ #1343 SMP Mon Sep 28 12:38:29 BST 2020 armv7l GNU/Linux
```
I'll leave it for a while, then re-enable `arm_freq_min=300` to see if that triggers the issues.

Definitely fine was `#1336`, which I used with `arm_freq_min=150` and `schedutil` governor for a long time. To run tests with the other issue, I rpi-update'd to `#1341`. Since it was to test performance governor and only very short sessions with over governors, I didn't recognise any issue there, but indeed I had a complete system crash a while later when bringing things back to production, leading to SD card corruptions which is why I flashed a new system from scratch with stable firmware packages. Due to much back and forth and testing I didn't think much about this, nor was I able to see any crash-related system log entries (analysing SD card on external system), but now I think it might be related to what I see just now after rpi-updating to `#1343`.

So if I can replicate, `#1336` - `#1341` are the version to go through, but since the issue often took a while to become apparent, I'll let every version run for a longer time to assure it's working, so may take a while until I can narrow it down.

MichaIng, 2020-10-01:
Okay I was able to successfully replicate a system crash multiple times by running a CPU intense task, `mysqlcheck` worked well in this particular case. I had `dmesg -w` running but sadly this time the crash broke SSH immediately, so I was not able to see any kernel errors. The symptoms are pretty much the same as before. Either tasks start to hang, producing this kind of random kernel errors, like `rcu: INFO: rcu_sched self-detected stall on CPU`, `tick_sched_handle`, `tick_sched_timer` which is found in first post log and new logs as well, and a few other matches around CPU scheduling, or the system crashes completely with no chance of leaving any logs left, even persistent journal is empty.
This was `#1343` + `schedutil` + `arm_freq_min=300` after `arm_freq_min=600` worked very well the whole night and half day and `arm_freq_min=300` + `performance` worked well the other half of the day.

MichaIng, 2020-10-02:
@popcornmix 
As of above testing, it is something between `#1336` and `#1337` that reintroduced the instability which was solved with `#1336`. `#1336` runs rock solid through a bunch of stress tests, including and excluding RAM, SD card and USB HDD disk writes, while `#1337` breaks with pretty much identical symptoms as reported originally with this threads. Probably something related to the fix has been reverted accidentally? Based on limited insights, https://github.com/raspberrypi/linux/pull/3815 and the related commits (not directly part of the PR) around CPUfreq seem to be the only changes that could have affected it.

popcornmix, 2020-10-02:
@MichaIng is all your testing with a Pi2?
https://github.com/raspberrypi/linux/pull/3815 does enable additional arm frequency points on all Pi models.
I suspect prior to 3815  300MHz wasn't being used when idle (check with `vcgencmd measure_clock arm`)

Is the only non-default setting in config.txt arm_freq_min=300 ?

MichaIng, 2020-10-02:
- All tests were done on `RPi 2`, yes.
- `300 MHz` was used before as well: https://github.com/raspberrypi/firmware/issues/1431#issuecomment-680270601
    ```
    2020-10-02 14:25:06 root@micha:/var/log# echo powersave > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
    2020-10-02 14:28:27 root@micha:/var/log# vcgencmd measure_clock arm
    frequency(48)=300000000
    ```
  - I cannot show it with `schedutil` governor and `vcgencmd measure_clock arm` since the command itself draws enough CPU time to have frequency raised to 600 MHz or 900 MHz already. But no reason to suspect it's being used, when `scaling_available_frequencies` and stats show it being available and used and `powersave` + `vcgencmd` does?
  - The **edit** was after upgrade to newer version to test the initial_turbo-related issue, as can be seen worked well for a while, but after some more testing it finally crashed 😄.
- There are some other non-default configs as well:
```
2020-10-02 14:33:29 root@micha:/var/log# grep -Ev '^[[:blank:]]*(#|$)' /boot/config.txt
hdmi_ignore_hotplug=1
framebuffer_width=16
framebuffer_height=16
max_framebuffer_width=16
max_framebuffer_height=16
framebuffer_depth=8
disable_overscan=1
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16
disable_splash=1
enable_uart=0
temp_soft_limit=50
temp_limit=65
initial_turbo=20
over_voltage=-2
arm_freq=900
gpu_freq=100
core_freq=450
sdram_freq=450
dtparam=sd_overclock=100
arm_freq_min=300
dtoverlay=micha
force_eeprom_read=0
```
- `over_voltage=-2` and `gpu_freq=100` were commented before without preventing the crash on `#1343` and `#1337`.
- All those are set now with `#1336` without causing any issues.
- `dtoverlay=micha` disables `vcsm` which fails to start and causes kernel errors on boot anyway with `gpu_mem=16` on completely headless device.
- But I can remove/comment all settings but `arm_freq_min=300` once and test it again with `#1337` or `#1343`. Any preference?

popcornmix, 2020-10-02:
You should remove all overclock settings for a fair test. over-clocking/under-volting comes with no guarantees.
Test with just `arm_freq_min=300` and nothing else. I think you'll probably still see the issue, but the other settings are muddying the water.

I have been poking around a Pi1 and have seen a hang. It looks like rapidly switching powersave/performance governors (300MHz<->700MHz) is safe. I have found switching from 500MHz to 700MHz can fail (and that can happen with ondemand governor after https://github.com/raspberrypi/linux/pull/3815

arm_freq_min=500 may prove to be a more reliable setting for provoking failure (at least on Pi1).

MichaIng, 2020-10-02:
Okay I'll do some cleaner tests. Interesting that it is only on certain frequency jumps. I btw tried to avoid the required reboot when switching the setting by instead using `scaling_min_freq`, but it has no effect in my case:
```
2020-10-02 16:09:05 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo 600 > scaling_min_freq
2020-10-02 16:09:12 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq
300000
```
Is that expected on RPi? Not a big deal, not really required due to `arm_freq_min` being available, but would have made this testing a bid faster.
Another way to test specific frequency jumps without changing `arm_freq_min`:
```
cd /sys/devices/system/cpu/cpufreq/policy0
echo userspace > scaling_governor
while :; do for i in 500 700; do echo $i > scaling_cur_freq; done; done
```
or put a short sleep inside. I'll play around with some methods.

popcornmix, 2020-10-02:
Looks like you can increase `scaling_min_freq` and decrease `scaling_max_freq`, but it considers the initial settings as the absolute limits.
```
root@domnfs:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq 
600000
root@domnfs:/sys/devices/system/cpu/cpufreq/policy0# echo 700000 > scaling_min_freq
root@domnfs:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq 
700000
root@domnfs:/sys/devices/system/cpu/cpufreq/policy0# echo 300000 > scaling_min_freq
root@domnfs:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq 
600000
```

MichaIng, 2020-10-02:
> Looks like you can increase `scaling_min_freq` and decrease `scaling_max_freq`

Hmm, exactly that is not working in my case:
```
2020-10-02 16:48:40 root@micha:/sys/devices/system/cpu/cpufreq/policy0# vcgencmd get_config arm_freq_min
arm_freq_min=300
2020-10-02 16:48:46 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq
300000
2020-10-02 16:48:58 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo 600 > scaling_min_freq
2020-10-02 16:49:10 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq
300000
2020-10-02 16:49:12 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo 400 > scaling_min_freq
2020-10-02 16:49:19 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_min_freq
300000
```
Okay now funny things happen when I touch the max frequency which was 900 MHz at the start:
```
2020-10-02 16:49:20 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo 700 > scaling_max_freq
2020-10-02 16:51:01 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_max_freq
300000
2020-10-02 16:51:19 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo 900 > scaling_max_freq
2020-10-02 16:51:30 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_max_freq
300000
2020-10-02 16:51:49 root@micha:/sys/devices/system/cpu/cpufreq/policy0# vcgencmd measure_clock arm
frequency(48)=300000000
2020-10-02 16:53:30 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo schedutil > scaling_governor
2020-10-02 16:53:38 root@micha:/sys/devices/system/cpu/cpufreq/policy0# echo performance > scaling_governor
2020-10-02 16:53:42 root@micha:/sys/devices/system/cpu/cpufreq/policy0# vcgencmd get_config arm_freq
arm_freq=900
2020-10-02 16:53:58 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_cur_freq
300000
2020-10-02 16:54:22 root@micha:/sys/devices/system/cpu/cpufreq/policy0# vcgencmd measure_clock arm
frequency(48)=300000000
2020-10-02 16:54:31 root@micha:/sys/devices/system/cpu/cpufreq/policy0# cat scaling_governor
performance
```
Touching max frequency a single time freezes it at 300 MHz regardless of governor 🤔. No dmesg/kernel output related to that.

popcornmix, 2020-10-02:
The numbers in in kHz. You should be using 700000 rather than 700.

MichaIng, 2020-10-02:
How was I able to overlook that 🤣, I removed/commented related parts from my posts to not mix/confuse with the actual issue this is about. `scaling_min_freq` + `scaling_max_freq` are working pretty fine as expected and might serve another way to avoid reboot for changing `arm_freq_min`.

MichaIng, 2020-10-05:
I found some time for testing, reverted to kernel `#1343` and `config.txt` containing `arm_freq_min=300` only, nothing else. Applied `userspace` and ran something like this for a while:
```
while :; do for i in 500000 700000; do echo $i > scaling_setspeed; done; done
```
With different frequency combinations, which and without adding some `sleep` after each change and verified on a different terminal that indeed the frequency is changing (not too fast or such) via `vcgencmd measure_clock arm` and `scaling_cur_frequency`, but so far I couldn't trigger any error 🤔. I as well tried to run some `stress` in parallel.

I'm going through a few more combinations, but probably manual frequency changes can simply not trigger it by CPUFreq induced changes via governor only?

paralin, 2020-10-06:
I'm still seeing this against 5.4.69-v8 (with associated latest rpi-firmware) 

```
[  484.329877] Workqueue: events_freezable mmc_rescan
[  484.331478] Call trace:
[  484.333067]  __switch_to+0x110/0x1d0
[  484.334669]  __schedule+0x2f4/0x750
[  484.336254]  schedule+0x44/0xe0
[  484.337836]  __mmc_claim_host+0xb8/0x210
[  484.339889]  mmc_get_card+0x38/0x50
[  484.343794]  mmc_sd_detect+0x24/0x90
[  484.347691]  mmc_rescan+0xc8/0x390
[  484.351593]  process_one_work+0x1c0/0x470
[  484.355503]  worker_thread+0x50/0x430
[  484.358914]  kthread+0x11c/0x150
[  484.362033]  ret_from_fork+0x10/0x20

```

popcornmix, 2020-10-06:
@paralin does your problem occur with no custom arm_freq_min setting in config.txt?
If it doesn't then removing that setting is recommended.
It it does then your problem is unrelated to this issue.

paralin, 2020-10-06:
@popcornmix even with defaults:

```
[  363.486818] INFO: task kworker/3:1:58 blocked for more than 120 seconds.
[  363.488636]       Tainted: G         C        5.4.69-v8 #11
[  363.490429] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  363.492282] kworker/3:1     D    0    58      2 0x00000028
[  363.494147] Workqueue: events_freezable mmc_rescan
[  363.496002] Call trace:
[  363.497846]  __switch_to+0x110/0x1d0
[  363.499697]  __schedule+0x2f4/0x750
[  363.501536]  schedule+0x44/0xe0
[  363.503386]  __mmc_claim_host+0xb8/0x210
[  363.505241]  mmc_get_card+0x38/0x50
[  363.507091]  mmc_sd_detect+0x24/0x90
[  363.508936]  mmc_rescan+0xc8/0x390
[  363.510759]  process_one_work+0x1c0/0x470
[  363.512584]  worker_thread+0x50/0x430
[  363.514388]  kthread+0x11c/0x150
[  363.516185]  ret_from_fork+0x10/0x20
```

testing on a second host w/ different brand SD card right now.

MichaIng, 2020-10-06:
@paralin 
Looks like a different issue then. Also I don't see any matches (aside of the blocked task, which is an unspecific symptom of the underlying issue) between my kernel stack traces and ours. I'm not good in reading it but mine contains a lot words related to CPU and scheduling and yours seem to be an issue with the SD card.
__________
I not have the "issue" that I cannot replicate the issue until now with schedutil scheduler and 300 MHz minimal frequency. After giving up to break it with the methods that worked reliable before, I now switched to "production" (full SD card backup created 😅) and still nothing for 1.5 days. Will now re-add the `config.txt` settings I used before one-by-one (moreless) and see if/which one re-adds the instability. I am pretty sure it ruled out the lowered voltage (`-2`) before, but maybe I made a mistake during all that back and forth. Would be said of the new scheduling pstates would require a higher voltage for some reason 😢.

MichaIng, 2020-10-25:
Okay I think I found the reason for the recent instability. Until crash, I see:
<details>

```
[14409.431666] mmc0: timeout waiting for hardware interrupt.
[14409.431748] [1a495b09] FCM> a5783dc8 0
[14409.431771] [1a495b0a] CMD  d 0
[14409.431793] [1a495b0c] REQ> a5783dc8 0
[14409.431813] [1a495b4f] TSK< a5783dc8 0
[14409.431832] [1a495b77] TSK> a5783dc8 0
[14409.431853] [5a4bb88a] REQ< a5757dc0 10801
[14409.431872] [5a4bb88d] CMD< d 70000
[14409.431895] [5a4bb89b] FCM< a5757dc0 a5757e28
[14409.431915] [5a4bb8a3] RSP  900 0
[14409.431935] [5a4bb8bf] FCM> a5757dc0 0
[14409.431954] [5a4bb8c1] CMD  d 0
[14409.431974] [5a4bb8c2] REQ> a5757dc0 0
[14409.431993] [5a4bb911] TSK< a5757dc0 0
[14409.432012] [5a4bb937] TSK> a5757dc0 0
[14409.432036] [5a5b9727] REQ< a5757dc0 10801
[14409.432055] [5a5b972f] CMD< d 70000
[14409.432075] [5a5b9740] FCM< a5757dc0 a5757e28
[14409.432094] [5a5b974a] RSP  900 0
[14409.432114] [5a5b9767] FCM> a5757dc0 0
[14409.432133] [5a5b976a] CMD  d 0
[14409.432156] [5a5b976b] REQ> a5757dc0 0
[14409.432176] [5a5b97ae] TSK< a5757dc0 0
[14409.432195] [5a5b97da] TSK> a5757dc0 0
[14409.432215] [5a607d49] PRD< b8f19590 0
[14409.432235] [5a607d4e] PRD1 b9711840 0
[14409.432253] [5a607d6b] PRD2 1 0
[14409.432276] [5a607d94] PRD3 b7c6be04 0
[14409.432296] [5a607d99] PDM> b8f19590 0
[14409.432316] [5a607da2] REQ< b8f194b0 10801
[14409.432336] [5a607da6] CMD< 19 18c6078
[14409.432355] [5a607db5] CMDD 8 200
[14409.432375] [5a607db8] SDMA b8f19590 b9711840
[14409.432398] [5a607dbf] FCM< b8f194b0 b8f19528
[14409.432418] [5a607dc9] RSP  900 0
[14409.432438] [5a607dcb] FCM> b8f194b0 0
[14409.432457] [5a607dce] CMD  19 200
[14409.432477] [5a607dcf] REQ> b8f194b0 0
[14409.432497] [1a608138] DMA< b8f19590 0
[14409.432516] [1a60813b] DMA  99 10801
[14409.432539] [1a608145] FDA< b8f194b0 0
[14409.432559] [1a608149] TCM< b8f19590 0
[14409.432578] [1a60814c] CMD< c 0
[14409.432597] [1a608153] TCM> b8f19590 0
[14409.432618] [1a608154] FDA> b8f194b0 b8f1955c
[14409.432636] [1a608155] DMA> 0 0
[14409.432659] [1a608b99] IRQ< 400 0
[14409.432679] [1a608b9c] IRQB b8f1955c 400
[14409.432699] [1a608b9e] FCM< b8f194b0 b8f1955c
[14409.432719] [1a608ba1] RSP  c00 0
[14409.432739] [1a608ba5] FCM> b8f194b0 b8f1955c
[14409.432758] [1a608ba6] IRQ> 0 0
[14409.432781] [1a608bb8] TSK< b8f194b0 0
[14409.432801] [1a608bfe] TSK> b8f194b0 0
[14409.432820] [1a608c6c] REQ< a5783dc8 10801
[14409.432840] [1a608c6d] CMD< d 70000
[14409.432860] [1a608c79] FCM< a5783dc8 a5783e30
[14409.432879] [1a608c7f] RSP  900 0
[14409.432902] [1a608ca0] FCM> a5783dc8 0
[14409.432922] [1a608ca1] CMD  d 0
[14409.432941] [1a608ca2] REQ> a5783dc8 0
[14409.432961] [1a608cdf] TSK< a5783dc8 0
[14409.432981] [1a608d07] TSK> a5783dc8 0
[14409.433000] [5a686039] PRD< b8f19790 0
[14409.433023] [5a68603f] PRD1 b9711840 0
[14409.433043] [5a686059] PRD2 1 0
[... 198 lines cut ...]
[14409.437041] mmc0: SDRSP0 0x00000900
[14409.437054] mmc0: SDRSP1 0x0000182e
[14409.437067] mmc0: SDRSP2 0xffffffff
[14409.437080] mmc0: SDRSP3 0x0002400f
[14409.437093] mmc0: SDHSTS 0x00000081
[14409.437106] mmc0: SDVDD  0x00000001
[14409.437119] mmc0: SDEDM  0x00010801
[14409.437132] mmc0: SDHCFG 0x0000050e
[14409.437149] mmc0: SDHBCT 0x00000200
[14409.437162] mmc0: SDHBLC 0x00000001
[14409.437173] mmc0: ===========================================
[14409.437454] mmc0: reducing overclock due to errors
[14409.441821] mmc0: overclocking to 75000000Hz
```
</details>
On every error it clocks down MMC until default 50 MHz, but the errors go on, e.g. on larger file writes, and even without SD overclocking it occurs. Extremely strange was the fact that it even happens when:
- Scaling governor `conservative` (or any other non-static)
- `scaling_min_frequency` = `arm_freq` = 900 MHz, hence static frequency, confirmed via time_in_state and vcgencmd.

As fast as scaling governor is switched to powersave or performance, all works stable, as fast as arm_freq_min is not set (default 600 MHz), all works stable as well. I found a way very reliably trigger the above error followed by hanging until crashing session.
I'm a bid out of ideas why only the setting itself has an effect and why a non-static governor causes reproducible issues even stat the frequency does not change at all.

But I finally remembered that I `uio_pdrv_genirq` in an effort to find out what is required for what and that did not cause any issues the last year or what. But since it has something to do with IRQ handling and the errors was an interrupt timeout, I enabled the module, and viola, was unable to trigger the error anymore.

No idea what changed and still a bunch of open questions as of above tests, but I think there is a good reason why the module is enabled by default. I'll continue to to test and verify the results but it looks now.

vmachiel, 2020-10-30:
I saw there was another release. Dit this fix this issue?

MichaIng, 2020-10-30:
@vmachiel 
Yes if you do not lower your `over_voltage` it seems stable now. As I'm customising my system quite much, for testing and lowering load/power usage to an absolut minimum, I'm still facing some instabilities as below:

I have a stable system with even `arm_freq_min=100`, for which `over_voltage` must not be lowered. The same is true for any other lowered `arm_freq_min` on my RPi 2, while this worked well with kernel 4.19 and 4.14, with `arm_freq_min=300` and `over_voltage=-2`. I'm currently testing if `over_voltage_min=-2` works, which anyway makes much more sense when lowering the minimum frequency.
This is basically fine, as lowering voltage is always a trial & failure topic, as each individual SoC behaves slightly different and I assume other hardware states affect it as well. However, that intermediate frequency scaling requires a higher voltage is a bid said in regards to power savings and lowering temperature, as I was hoping for an overall lowered power consumption with intermediate frequencies compared to the 1 step jumps from min to max before. But not sure if it's worth to investigate this further.

Additionally `uio_pdrv_genirq` kernel module needs to be active now (it is loaded by default), else I get the mmc interrupt timeouts which can be triggered easily by saving/writing some larger text/config files. I also never faced this the last year on 4.19 where I had this module blacklisted to test if/where its needed for 😅. Using a static CPU governor solves it as well somehow. I'll again run a test with this isolated with default `arm_freq_min`, voltage etc., as I believe it is also only an issue in combination with lowered voltage.

vmachiel, 2020-11-01:
Thanks, let me know!

MichaIng, 2020-11-13:
@popcornmix 
Do you actually want to further investigate this issue? From my point of view, there is still something not as it _should_ be, but I have limited insights to know what may effect what. I tested a few other combinations, and a few lead to errors or crashes quickly  (within an hour, sometimes easily to trigger quickly with some load) while the others run stable for days, including stress tests, and all of them ran stable for months on Linux 4.19.X. Notably I narrowed down everything to the following cases, all with a dynamic CPUFreq governor, of course:
1. `over_voltage=0` + `arm_freq_min=100` => **stable**
2. `over_voltage=-2` + `arm_freq_min=600` (or commented) => **stable**
3. `over_voltage=-2` + `arm_freq_min=500` => **hangs or crashes quickly**, with different errors, either producing kernel errors + hangs, which can be solved by raising the `scaling_min_frequency` or applying any static CPU governor (`powersave`, `performance`, `userspace`), or crashing the system completely without any kernel error, where I used `dmesg -w` on a dedicated SSH session + persistent `journald` logs.
_I think the `uio_pdrv_genirq` I had in view before does not really have an effect, but loading it might have reloaded something else around kernel/IRQ, just shifting the symptoms of the underlying issue, or so. As far as I understand, that kernel driver is either used actively by another module or service, or not, and in my case it is simply not used._

Since `over_voltage` has no effect on the voltage when the lowest frequency (e.g. 500 MHz in 3. case) is currently applied, it _should_ not have an effect, at least the voltage + frequency combinations in 2. and 3. case above are exactly the same. So probably it has something to do with the way/order/timing in which voltage and frequency are adjusted, e.g. the frequency is raised quickly while the voltage is raised later, leaving a short time frame with a too low voltage for the current frequency. But in above case, since `over_voltage_min` is the same, the 1. case then should have the same issue. So probably it is then about a short voltage peak demand when doing a certain larger step and while 1. case allows that (1.35V), 2. does not required it (peak demand lower) and 3. case frequency switch requires higher voltage for a short time but that is capped to insufficient 1.3V. That is all I can imagine to explain the results 😄.

However, I can understand when this is not followed further since a stable `over_voltage` is something that needs to be found/tested on every system individually anyway and the default value works stable, at least on all my tests. While I'm not happy with that situation, say a word and I'll stop investing time and keeping this issue up here 🙂.

MichaIng, 2020-11-23:
I just recognised https://github.com/raspberrypi/firmware/commit/bff705fffe59ad3eea33999beb29c3f26408de40. Needed to learn what a [VCO](https://en.wikipedia.org/wiki/Voltage-controlled_oscillator#Frequency_synthesizers) is 😄. I'll try it ASAP.

Meanwhile my Pi runs stable with `over_voltage=-1` + `arm_freq_min=100` + `over_voltage_min=-2` + `schedutil` governoe, let's see if `over_voltage=-2` is possible again with the new kernel.

m0nk3h, 2021-03-11:
Is this issue still being worked on?  I have recently purchased a Pi Zero (non-wifi), running DietPi 7.0 with kernel Linux DietPi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux.  Setting arm_freq_min seems to be ignored (tried ondemand & conservative govenor) and is always set to 700-1000mhz as Michalng reports above.

TheyKilledKenny, 2021-03-22:
We are also still waiting for the resolution of this issue. 

We're using rpi2 v1.2 and used to change this in config.txt:
arm_freq_min=350
core_freq_min=150
temp_limit=70

dtparam=i2c_arm=on
dtparam=spi=on

dtoverlay=i2c-rtc,ds3231
dtoverlay=sc16is752_0

enable_uart=1

No over/undervoltage and no other changes in config.txt
Standard governor, never touched this settings.
Normally used in headless mode, boot in console not logged in. No desktop loaded at startup.

Kernel 4.9.35-v7+ # 1014  works as expected, so we know the hardware part is able to manage it.

We started having problems when we upgraded to the kernel 4.19.80-v7+ # 1275 and never resolved also in kernel 5.x
The latest kernel simply ignores the arm_freq_min parameter and this results in **increasing the cpu temperature by about 3 to 5° C** during normal operation with no load on the cpu (vcgencmd measure_temp arm)
Unfortunately this also leads to a much faster rise in CPU temperature as the load rises. 

If can help, on kernel 4.19.80 there is a temperature range that cause more often the issue and the reboot phase, the reboot phase high the cpu temperature some degree more.
To reproduce we usually stress the cpu to a temperature range from 55 to 65°C and then launch a reboot command. Most of the time the shutdown process is ok, but it is not able to boot up again, it stops before the RGB splash screen. Sometime the red and the green led are fixed on, sometime only the red led is on.
I also checked all the possible outputs on uart by enabling with the command sed -i -e" s / BOOT_UART = 0 / BOOT_UART = 1 / "bootcode.bin", but I have never seen a useful message.

It seems that during the boot phase the kernel reads the frequency and temperature setting from the config.txt file, since it is already hot then it start throttling the cpu accordingly, but this break the boot process (cpu, sd card reader or something else).

In conclusion:
We need to lower the idle frequency down to 350Mhz, if possible even less.
Is there an ETA for solving this issue? 

Thank you.

mloebl, 2021-03-22:
For me I've had a Pi Zero W running for weeks with this:
```
arm_freq_max=700
arm_freq=700
```

Any higher and it will fail to reboot.

MichaIng, 2021-03-23:
`arm_freq_max` is ineffective btw, `arm_freq` is the maximum frequency already.

The Raspberry Pi Zero has a default of 1000 MHz, so if that causes issues, e.g. when you simply remove or comment the two lines and it fails to reboot, then there seems to be an issue with the hardware. But that is not related to the `arm_freq_min` topic, this issue is about 😉.

mloebl, 2021-03-23:
> `arm_freq_max` is ineffective expectedly, `arm_freq` is the maximum frequency already.
> 
> The Raspberry Pi Zero has a default of 1000 MHz, so if that causes issues, e.g. when you simply remove or comment the two lines and it fails to reboot, then there seems to be an issue with the hardware. But that is not related to the `arm_freq_min` topic, this issue is about .

Just realized I crossed streams on github issues, deleted it :facepalm:

ForceConstant, 2021-03-31:
I have a Raspi 3B+, and have noticed that I have noticied that changes to arm_freq_min below 600 have no affect, and see that there was a previous workaround forcing this, but I did a recent rpi-update, and still it persists. Should I be able to set min frequency below 600 now?

MichaIng, 2021-03-31:
Then this should be documented to avoid confusion. The setting can still be used to raise the minimum frequency, although that can be done via CPUfreq as well.

JamesH65, 2021-03-31:
Documentation has been updated https://github.com/raspberrypi/documentation/commit/66e749c4ec540381062b36d06052a89cc4920ac2

TheyKilledKenny, 2021-04-09:
> 
> 
> It's not something currently supported.
> In general the clock gating of the arm core is pretty good and the benefits are minor when lowering arm clock below 600MHz

The benefit on the working temperature is enough to make the difference between using or not using a heat sink, in our case.
So, as we know the hardware is able to manage it (Ras Pi2 V1.2), please try to allow us to use lower frequencies. I hope this issue can be resolved asap.

MichaIng, 2021-06-03:
On Raspberry Pi 4, `arm_freq_min` still works to reduce the minimal frequency down to 100 MHz. Interestingly it seems to cause issues there as well: https://github.com/MichaIng/DietPi/issues/4455#issuecomment-853245807
- With `arm_freq_min=100`, the system clock runs extremely slow and with `arm_freq_min=200` it still has a notable shift.
- 300 MHz and more seems to work fine.
- `over_voltage=-2` was set as well, but setting it to zero did not solve the system clock issue.

Just a single case for now, but probably the same underlying issue.

C0D3-M4513R, 2021-06-03:
I am doing some more tests, this time with ahk timing the execution of the date command, to 5 seconds.
If longer tests are required, I will gladly do a minute timing each!
oc is none/normal
```
root@DietPi:~# echo 300000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq
root@DietPi:~# vcgencmd measure_clock arm
frequency(48)=300111328
root@DietPi:~# date +%H:%M:%S:%N
13:50:02:917133289
root@DietPi:~# date +%H:%M:%S:%N
13:50:07:943119323
root@DietPi:~# echo 200000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq
root@DietPi:~# vcgencmd measure_clock arm
frequency(48)=200074224
root@DietPi:~# date +%H:%M:%S:%N
13:50:15:251591411
root@DietPi:~# date +%H:%M:%S:%N
13:50:19:534279556
root@DietPi:~# echo 100000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq
root@DietPi:~# vcgencmd measure_clock arm
frequency(48)=100037112
root@DietPi:~# date +%H:%M:%S:%N
13:50:26:282622460
root@DietPi:~# date +%H:%M:%S:%N
13:50:26:654301830
root@DietPi:~#
```
I used the following ahk script:
```ahk
#NoEnv  ; Recommended for performance and compatibility with future AutoHotkey releases.
; #Warn  ; Enable warnings to assist with detecting common errors.
SendMode Input  ; Recommended for new scripts due to its superior speed and reliability.
SetWorkingDir %A_ScriptDir%  ; Ensures a consistent starting directory.

#MaxThreadsPerHotkey 2
SetBatchLines -1

F17::
	SendInput {Raw}vcgencmd measure_clock arm`n
	;Wait for command output
	Sleep, 500
	SendInput {Raw}date +`%H:`%M:`%S:`%N`n
	Sleep, 5000
	SendInput {Raw}date +`%H:`%M:`%S:`%N`n
```

TheyKilledKenny, 2021-06-30:
> 
> 
> It's not something currently supported.
> In general the clock gating of the arm core is pretty good and the benefits are minor when lowering arm clock below 600MHz

Unfortunately by now we have more than 4K rpi2 v1.2 and v1.1 installed which are having continuous blocking problems and needing more reboots to restart,
we are forced to operate manually on each to bring the kernel back to version 4.9.35 which allows us to manage the frequencies
without any block, but this exposes to risks.
Unfortunately it is impossible to equip the rpi2 with a cooling fan due to the environment in which they work, so in addition to the blocks, we are also having overtemperatures that we never had before. 
What is the problem that prevents from reenabling this very useful feature?

Please, let me know if you need more information than what @MichaIng  has already given you, which seems to me already very complete and exhaustive.

I apologize to the readers, but I must express my disappointment:
The apparent superficiality with which this very big issue is treated and closed gives me a lot to think about...
It doesn't seem serious to me to close a so big problem with a simple: "It is no longer supported" or modify the documentation as @JamesH65 wrote, leaving all in trouble.
Please, show some commitment even for those who buy in batches and not only have to make gaming machines and overclocking, but things like environment monitoring, energy management, etc.
I hope to see some sort of progress very soon on this issue which at the moment seems to be abandoned by you.

[23 more replies]