raspberrypi/firmware #1433: Latest stable kernel upgrade breaks Qt 5.12 with EGLFS on FKMS

pinaraf, 2020-07-22:
**Describe the bug**
After upgrading to the new kernel, our application stopped working. The display remained black, and Qt was showing thousands of lines like:
```
     2.216 unknown:0 critical Could not queue DRM page flip on screen HDMI1 (Invalid argument)
```

When setting Qt to use atomic KMS, the application stops working after some random amount of time, with the following message in kernel:
```
[  701.916447] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:74:crtc-2] flip_done timed out
```

**To reproduce**
Upgrade to latest kernel, using a Qt 5.12.6 app built with EGLFS support and FKMS.

**Expected behaviour**
Qt apps keep working

**Actual behaviour**
Qt apps break

**System**
raspinfo output with atomic KMS:
```
System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

Linux raspberrypi 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux
Revision        : a020d3
Serial          : 0000000035ba5f2d
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jul 17 2020 11:04:27 
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       15022716 1966892  12413488  14% /
devtmpfs          342740       0    342740   0% /dev
tmpfs             474324       0    474324   0% /dev/shm
tmpfs             474324   12364    461960   3% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             474324       0    474324   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   54459    203636  22% /boot
tmpfs              94864       0     94864   0% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
---------------------------
[... 582 lines cut ...]
[   11.107037] usb 1-1.1.2: Product: WS170120
[   11.107043] usb 1-1.1.2: Manufacturer: WaveShare
96663G8477049] usb 1-1.1.2: SerialNumber: V
[   11.109974] input: WaveShare WS170120 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.0/0003:0EEF:0005.0004/input/input4
[   11.357683] hid-multitouch 0003:0EEF:0005.0004: input,hidraw0: USB HID v1.10 Device [WaveShare WS170120] on usb-3f980000.usb-1.1.2/input0
[   11.775296] usb 1-1.3: new high-speed USB device number 14 using dwc2
[   11.906935] usb 1-1.3: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=32.98
[   11.906946] usb 1-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   11.906952] usb 1-1.3: Product: USB2.0 Hub
[   11.907705] hub 1-1.3:1.0: USB hub found
[   11.908058] hub 1-1.3:1.0: 4 ports detected
[   12.225238] usb 1-1.3.1: new low-speed USB device number 15 using dwc2
[   12.380192] usb 1-1.3.1: New USB device found, idVendor=04d9, idProduct=0006, bcdDevice= 1.40
[   12.380203] usb 1-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   12.380210] usb 1-1.3.1: Product: RPI Wired Keyboard 2
[   12.380216] usb 1-1.3.1: Manufacturer:  
[   12.393747] input:   RPI Wired Keyboard 2 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.1/1-1.3.1:1.0/0003:04D9:0006.0005/input/input5
[   12.456961] hid-generic 0003:04D9:0006.0005: input,hidraw1: USB HID v1.11 Keyboard [  RPI Wired Keyboard 2] on usb-3f980000.usb-1.3.1/input0
[   12.470037] input:   RPI Wired Keyboard 2 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.1/1-1.3.1:1.1/0003:04D9:0006.0006/input/input6
[   12.536062] hid-generic 0003:04D9:0006.0006: input,hidraw2: USB HID v1.11 Device [  RPI Wired Keyboard 2] on usb-3f980000.usb-1.3.1/input1
[   13.441274] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   22.260466] tun: Universal TUN/TAP device driver, 1.6
[   37.687846] rtc-pcf8523 1-0068: low voltage detected, time is unreliable
[  701.916447] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:74:crtc-2] flip_done timed out
```

Vliro, 2020-09-12:
I had a similar error and use the env
QT_QPA_EGLFS_ALWAYS_SET_MODE="1
and it works, although for me it complained about hdmi port 0.

Julien-Blanc-tgcm, 2020-12-03:
Confirmed here as well.

The `QT_QPA_EGLFS_ALWAYS_SET_MODE=1` workaround indeed fixes it, thanks for providing it.