raspberrypi/firmware #1434: Raspberry pi 4 is not booting up

hriday-khurana, 2020-07-22:
I recently ran sudo apt-get update and sudo apt-get upgrade using putty while performing the upgrade command it stopped in middle and raspberry started booting up & failing continuously and now it giving me error and i am not able to start my raspberry pi os at all.

i am not linux geek and i am not sure how to fix the issue, can someone please help recover the os. i have added a snapshot as well.
[image]

timg236, 2020-07-22:
You’ve not provided any details of the hardware configuration although it appears you have an out of date beta version of the bootloader without an SD card.

hriday-khurana, 2020-07-23:
I am sorry about that i am using Raspberry Pi 4 1 GB RAM and 64 GB HP SD card. it was running fine until the moment i ran apt-get upgrade. 
So, i just want to fix the existing sd card instead of reinstalling everything if you have any suggestions to fix the boot.

timg236, 2020-07-23:
The Raspberry Pi forums the best place for support questions but here's a couple of suggestions

Unplug the external USB drive and try booting - although the diagnostics show an SD card was found but no valid firmware was found i.e. it's failed over to start.elf instead of start4.elf

Windows chkdsk or similar on a PC may repair corruption on the boot partition, it's not practical to determine why the APT upgrade failed e.g. it was interrupted / the SD card is failing etc

If that doesn't work then a new OS install is probably the quickest thing to do. Perhaps do that on a different SD card if there is data that you want to recover by mounting the current SD card via a USB card reader.

Unless you specifically want to use that early beta version of the bootloader then use the Raspberry Pi Imager to create an EEPROM image with the factory defaults.

hriday-khurana, 2020-07-23:
[Timg236]: Unplug the external USB drive and try booting - although the diagnostics show an SD card was found but no valid firmware was found i.e. it's failed over to start.elf instead of start4.elf
[hriday]: Tried this but did not make any difference and also it was failing for start.elf as well as start4.elf as it was throwing error that "start4.elf not found"

[Timg236]: Windows chkdsk or similar on a PC may repair corruption on the boot partition, it's not practical to determine why the APT upgrade failed e.g. it was interrupted / the SD card is failing etc
[hriday]: tried this today but nothing changed and issue was still there.

[Timg236]: If that doesn't work then a new OS install is probably the quickest thing to do. Perhaps do that on a different SD card if there is data that you want to recover by mounting the current SD card via a USB card reader.
[hriday]: I can perform a new OS but how do I recover all my files from existing sd card using USB card reader, appreciate if you provide detailed information for this.

[Timg236]: Unless you specifically want to use that early beta version of the bootloader then use the Raspberry Pi Imager to create an EEPROM image with the factory defaults.
[Hriday]: i am not really interested to stick with beta version, it ws just unfortunate that i didnt ran update commmand for quite while and when i did it corrupted my bootloader+OS. Now its not recovering, i also tried Pi imager to recover using EEPROM image but still no luck.

timg236, 2020-07-24:
The bootloader isn’t corrupted, otherwise you wouldn’t have seen the HDMI diagnostics screen. 
It’s not clear why you changed the bootloader but the beta is working correctly in this case.
You could try copying the boot files from another a different raspberry is SD card.
However it apt upgrade was interrupted or SD card is failing there may be other errors.
If you have a second SD card and a usb card reader you could try that to recover the files.

This doesn’t appear to be a firmware bug so you might get more responses on the raspberry pi forums

hriday-khurana, 2020-07-24:
Well i am not expert in linux so that i could pin point exact issue where it lies. I am still noob in raspberrypi and it was running fine. Only action i performed is following two commands using putty through my laptop:
sudo apt-get update (ran successfully)
sudo apt-get upgrade (caused issues in middle while performing the changes)

That's all i know. however i did reinstall os in seperate SD card but i would like to recover data/files of installed application from my old sd card. Hope you can provide steps or information of tools which can help recovering the data.

JamesH65, 2020-07-25:
Once you have the second SD card working you can use a SD to USB adaptor on the first card, plug it in and it should appear as another drive, just copy your data from it.

elgreg75, 2021-10-03:
I had the same issue today, I will try to do like you said

sebres, 2022-09-17:
For people having similar issue and want to repair either:

Disclamer: 
*⚠️ Only follow any given step if you completely understand it and its implications for your SD-card or Pi. Depending on what you’re doing, some of these steps could cause data loss.*

0. firstly ensure your sd-card adapter is not locked (write protected);
1. sometimes it is simply a mark on sd-card (e. g. after power failure on write process, or if sd-card removed unexpectedly), which can be fixed this way:
  - on linux - insert card (usb adapter), take a look which dev is assigned (here `sdb` but can be `mmc0` etc), unmount both partitions (if they mounted automatically) and do fsck:
  ```
# need to be root or within sudo -s:
umount /dev/sdb1; umount /dev/sdb2
fsck -v /dev/sdb1
fsck -v /dev/sdb2
  ```
  - on windows - insert card, windows will detect the issue like "removed unexpectedly" and ask you to repair it, click on "yes" (but hereafter on "no", if it would ask you to format the drive).
2. if boot is really corrupted, we can try to copy it from image (linux only, and also as sudoer/root):
  - try to backup firstly, so mount the boot partition (here `/dev/sdb1`) and if wanted the main disk (here `/dev/sdb2`):
  ```
mkdir /tmp/rpi-boot; mount /dev/sdb2 /tmp/rpi-boot
mkdir /tmp/rpi-disk; mount /dev/sdb2 /tmp/rpi-disk
# backup it (here tar/zstd:
tar -C /tmp/rpi-boot/ --zstd -cf /backup/rpi-boot.tar.zst .
tar -C /tmp/rpi-disk/ --zstd -cf /backup/rpi-disk.tar.zst .
  ```
  - download pi-image, unpack (.xz) and mount it also:
  ```
mkdir /tmp/rpi-orgboot; mount -o loop,offset=4194304 /download/2022-09-06-raspios-*.img /tmp/rpi-orgboot
  ```
  * the offset can be calculated as `4194304 = 512 (sector size) * 8192 (start point of img1)`, which can be found using:
    `fdisk -l /download/2022-09-06-raspios-*.img`
  - then copy the content of `/tmp/rpi-orgboot/` to `/tmp/rpi-boot/` (possibly retaining files in root-dir you modified by yourself).
  - remount boot-partition again and check whether the content is still there (some sd-cards may get broken without to set digital write-protection but get internally RO, so you're able to write to it, but the controller didn't realy write at its end, so the files are not overwritten after all).
  - unmount everything of `/tmp/*` and try to boot in RPI.

  - on windows you can use either some linux VM or boot from linux live CD
3. if sd-card got digital write-protection (readonly), you can try to remove it (on linux with `echo "0" > /sys/block/sdb/ro`, etc or on windows with `diskpart` with clearing RO attribute on disk); 
although mostly it happens because the card got corrupted, for instance because it has limited number of write cycles (and the cards hardware lifespan is exceeded due to many writes by logging, swap or journaling on the filesystem) - in that case you may need a new card either.
4. here are few links for those who are interested in how to extend the lifespan of sd-card:
  - [durability - How can I extend the life of my SD card? - Raspberry Pi Stack Exchange](https://raspberrypi.stackexchange.com/questions/169/how-can-i-extend-the-life-of-my-sd-card)
  - [Reducing SD Card Wear on a Raspberry Pi or Armbian Device # Chris Dzombak](https://www.dzombak.com/blog/2021/11/Reducing-SD-Card-Wear-on-a-Raspberry-Pi-or-Armbian-Device.html)
  - [Stopping SD Card Corruption on Raspberry Pi’s Raspbian – IdeaHeap](https://ideaheap.com/2013/07/stopping-sd-card-corruption-on-a-raspberry-pi/)

closed 2024-01-08 by timg236