raspberrypi/firmware #1435: Aux UART breakage with newer firmwares

vianpl, 2020-07-23:
Hi everyone,
I was made aware in the upstream kernel mailing lists that with the latest firmware versions (starting in April) the aux UART isn't working properly. Only outputs noise. I've been looking into it and found out this fixes the issue:

```
diff --git a/drivers/tty/serial/8250/8250_bcm2835aux.c b/drivers/tty/serial/8250/8250_bcm2835aux.c
index 12d03e678295..b0a4cc3cf0e2 100644
--- a/drivers/tty/serial/8250/8250_bcm2835aux.c
+++ b/drivers/tty/serial/8250/8250_bcm2835aux.c
@@ -145,12 +145,7 @@ static int bcm2835aux_serial_probe(struct platform_device *pdev)
                return ret;
        }

-       /* the HW-clock divider for bcm2835aux is 8,
-        * but 8250 expects a divider of 16,
-        * so we have to multiply the actual clock by 2
-        * to get identical baudrates.
-        */
-       up.port.uartclk = clk_get_rate(data->clk) * 2;
+       up.port.uartclk = clk_get_rate(data->clk);

        /* register the port */
        ret = serial8250_register_8250_port(&up);

```
Obviously applying this would break the UART for older firmware users. So, for a more subtle solution, I need some feedback from anyone more knowledgeable. The way I understand the clock setup is the following:

```
    pllc
       pllc_per
       pllc_core2
       pllc_core1
       pllc_core0
          vpu
             fe804000.i2c_div
             fe205000.i2c_div
             aux_spi2
             aux_spi1
             aux_uart
                8/16_divisor << This changed with newer firmwares
                   actual_aux_uart
```

There should be a way to control this divisor, but unfortunately it isn't documented anywhere. Any ideas/comments?

pelwell, 2020-07-23:
I use the AUX UART every day with no issues. Are you specifically talking about upstream builds? Have you tried a Raspberry Pi OS image?

vianpl, 2020-07-23:
Yes, sorry, we've observed that behavior running upstream Linux, upstream device-tree and this config.txt: 

```
enable_uart=1
arm_64bit=1
```
I'll see if I'm able to reproduce this with the new arm64 RPi OS.

pelwell, 2020-07-23:
And when you say the latest firmware, is that from top of tree in the firmware repo?

vianpl, 2020-07-23:
Any firmware released after Feb12 (1.20200212) is affected.

pelwell, 2020-07-23:
So you've tried the Jul17 release?

On some platforms (and Pi 4B isn't one of them) the automatic core frequency clamping when UART1 is enabled wasn't working - that was fixed earlier this month. The other changes in UART clock handling came in early 2019, so it's not clear what has caused this change.

You can halve the problem space by confirming whether or not the core clock is changing with the ARM clock. Starting from a new shell (or at least one with no active jobs):
```
$ sleep 1; vcgencmd measure_clock core; (while true; do true; done) & sleep 1; vcgencmd measure_clock core; kill %1
```

vianpl, 2020-07-27:
Hi @pelwell sorry for the silence, but I got sidetracked. All this is using the Jul 23 release, RPi OS 64bit, stock config.txt.

I managed to reproduce the issue with rpi-5.8.y but I had to revert https://github.com/raspberrypi/linux/commit/b76c8d538204e7d932afa8707070936f53d697f6.

Here's what I'm reading:
 - rpi-5.8.y:  vpu = 500000000Hz (clk-bcm2835.c), 500000992Hz (vcgencmd measure_clock core)
 - rpi-5.8.y + patch reverted: vpu = 999999996Hz (clk-bcm2835.c), 500000992Hz (vcgencmd measure_clock core)

That  1GHz VPU clock makes sense? Note that this runs with Maxime's clock series

vianpl, 2020-07-27:
Sorry for closing the issue, I mis-clicked.

More tests:
 - rpi-5.8.y + Feb12 fw : vpu = 500000000Hz (clk-bcm2835.c), 500000992Hz (vcgencmd measure_clock core)
 - rpi-5.8.y + patch reverted + Feb12 fw : vpu = 500000000Hz (clk-bcm2835.c), 500000992Hz (vcgencmd measure_clock core)

pelwell, 2020-07-27:
500MHz is the correct value for the VPU clock in turbo mode - it would fail at 1GHz. For some reason the upstream clock driver is getting the wrong answer - that shouldn't be possible, whatever the firmware is doing differently.

vianpl, 2020-07-27:
OK, so it turns out that PLLC and PLLA have the "using_prediv" flag set with newer firmwares (either that, or the code that checks it is buggy).

With that enabled this happens:
```
static unsigned long bcm2835_pll_get_rate(struct clk_hw *hw,
					  unsigned long parent_rate)
{
[...]
	if (using_prediv) {
		ndiv *= 2;
		fdiv *= 2;
	}

	return bcm2835_pll_rate_from_divisors(parent_rate, ndiv, fdiv, pdiv);
}
```
Which doubles PLLC's rate and then VPU's.

vianpl, 2020-07-29:
I read the following here: https://elinux.org/The_Undocumented_Pi (see bcm2711 part) 

_There is no prescaler in the feedback loop. The corresponding bit in ANA1 can still be set, but it does not have any effect on the PLL frequency._ 

This would explain everything, and it's pretty simple to fix in the clk driver. Could you confirm?

timg236, 2020-07-30:
On 2711 bits 13:14 of ANA1 (PLLA,PLLB,PLLC,PLLD) have been repurposed for to for VCO RANGE bits which tweak the analog settings according to the VCO frequency. There was a firmware fix around the time that you mentioned which resolve and issue where this was not always set which could cause problems on some chips. 

Since the VCO frequencies are fixed when start.elf is initialised I think the simplest fix would be for clk-bcm2835.c to skip the prediv path on 2711.

vianpl, 2020-07-30:
@timg236  Thanks! I'll take care of it.

closed 2020-07-30 by vianpl

pelwell, 2020-07-30:
And/or perhaps to use the new firmware clocks driver instead, at least for the core clock.

vianpl, 2020-07-30:
@pelwell That's a more drastic move. I'd rather keep it simple for the time being.

That said, in the long run, IMO I'd be nice to look at moving everything to the FW driver. :slightly_smiling_face:

vianpl, 2020-08-04:
FYI https://lkml.kernel.org/lkml/159649004236.1360974.8603728575729464532@swboyd.mtv.corp.google.com/T/